meerk40t/meerk40t #896: RC2

tatarize, 2022-03-17:
### Summary Description

RC8 page discussion

### MeerK40t Version

0.8.0 RC-2

### MeerK40t Type

Executable

### Your Operating System

Windows

tatarize, 2022-03-17:
Galvo flip might be fixed... unsure.

jpirnay, 2022-03-17:
The interpretation of lengths without units (i.e. mils) has changed. Try:
```
rect 0 0 1000 1000
circle 1000 1000 1000
```
[image]

tiger12506, 2022-03-17:
> The interpretation of lengths without units (i.e. mils) has changed. Try:

This isn't unexpected. I tried to get Sophist to change the default keymaps for circle and rect to account for that change, but I think it fell through the cracks. I think the defaults should now be:
rect 0 0 1in 1in stroke red classify
circle 1in 1in 1in stroke red classify

Or something of that nature. IF we think 'fill black' would be better than 'stroke red', I'm fine with that.

Sophist-UK, 2022-03-17:
A few minor bugs as I started to use MK 0.8 for preparing files for later burning on 0.7.5 executable...

a. Right click change raster op speed, was not reflected in the Op label. A few more tries changing with right click / speed and through Op properties and it seems to be a bit hit and miss,.

Sophist-UK, 2022-03-17:
Just tried to load the file I saved in 0.8.0RC2 into MK 0.7.5 and instead of two raster ops I got 1 raster op and two unknown ops.
[Circular Nikki Smith - Cici Daisy LASER7.txt](https://github.com/meerk40t/meerk40t/files/8288295/Circular.Nikki.Smith.-.Cici.Daisy.LASER7.txt)

joerlane, 2022-03-17:
Galvo is still upside down.

tatarize, 2022-03-18:
> The interpretation of lengths without units (i.e. mils) has changed. Try:

It has. They used to be 1:1 with mils/steps in the M2 Nano. But, they got switched to nano meters and then switched again to Pixels. Specifically 1/96th of an inch pixels. This should be somewhat configurable but was part of a larger units upgrade that was needed to correctly process different distances with different devices. GRBL for example runs gcode and can run mm or inch mode though the stepper motors for those typically runs at about 1 mil per step. Ruida runs in micrometers. Galvo lasers have distances that differ based on the lens being used and distance from the material.

jpirnay, 2022-03-18:
> This isn't unexpected. I tried to get Sophist to change the default keymaps for circle and rect to account for that change, but I think it fell through the cracks. I think the defaults should now [...]

I've provided [PR#902](https://github.com/meerk40t/meerk40t/pull/902) for this.

jpirnay, 2022-03-18:
I had still quite a lot of old "control+" keybinds from previous versions (therefore [PR#900](https://github.com/meerk40t/meerk40t/pull/900)), We should either mention to do a 'bind default' or provide a migration routine.

Sophist-UK, 2022-03-18:
no gui command line options -z and -Z do not seem to work. -c works, but e.g. a crash with -cZ still shows a GUI crash box showing that wx has still been loaded.

Sophist-UK, 2022-03-19:
Ran from source with -cZ. When I typed quit at the console prompt, the GUI opened. When I quit I received this crash:
```
MeerK40t crash log. Version: 0.8.0003 RC2 git on Windows:AMD64 - 4.1.1 msw (phoenix) wxWidgets 3.1.5
Traceback (most recent call last):
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t.py", line 10, in <module>
    sys.exit(main.run())
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\main.py", line 228, in run
    kernel()
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\kernel\kernel.py", line 1031, in __call__
    self.set_kernel_lifecycle(self, LIFECYCLE_KERNEL_MAINLOOP)
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\kernel\kernel.py", line 744, in set_kernel_lifecycle
    plugin(kernel, "mainloop")
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\gui\plugin.py", line 184, in plugin
    meerk40tgui = kernel_root.open("module/wxMeerK40t")
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\kernel\context.py", line 389, in open
    return self.open_as(registered_path, registered_path, *args, **kwargs)
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\kernel\context.py", line 428, in open_as
    self.kernel.set_module_lifecycle(instance, LIFECYCLE_MODULE_OPENED)
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\kernel\kernel.py", line 955, in set_module_lifecycle
    m.module_open(*args, **kwargs)
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\gui\wxmeerk40t.py", line 555, in module_open
    context.setting(int, "language", None)
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\kernel\context.py", line 141, in setting
    load_value = self._kernel.read_persistent(
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\kernel\settings.py", line 78, in read_persistent
    value = self._config_dict[section][key]
AttributeError: 'Kernel' object has no attribute '_config_dict'
```

tatarize, 2022-03-19:
> Ran from source with -cZ. When I typed quit at the console prompt, the GUI opened. When I quit I received this crash:

Ah, yes. When you type quit the gui still opens but the kernel has shutdown and thus `_config_dict` is deleted by that time. It should probably avoid launching the gui in that case or abort the mainloop processing in that case. I'm not quite sure what the correct answer is but I have seen that before and it certainly does need fixing.

tatarize, 2022-03-19:
Also, for the old style keybinds. These aren't actually needed. The code integrates in a `Settings` code that does the persistent settings itself rather than depending on wx to perform that with `wx.Control` this is so that CLI versions can preserve persistent settings etc. This creates the settings in the relevant save-directory which depends on the OS type but also it means that the settings from 0.7.x and before are *NOT* compatible and thus will never have older carried over settings since that's impossible.

tatarize, 2022-03-19:
* Properties windows should be changed to a lookup based system with pages that are defined rather than type based different window categories.
* Properties windows should be changed when the selection changes (not double clicking on something) (I think jpirnay has a stab at that and I thought it was charming but also ties into #879 where the properties weren't reflected the currently selected object.
* Images and the screen sizes for correct raster sizes may be in error.
* RasterWizard scripts should defer to the planning stage.
* Images should have properties to set the DPI of the image. This can often be exactly a raster-step amount on the image but all DPIs are functionally plausible.

Sophist-UK, 2022-03-19:
Agree with all of the above - we could then have a properties pane.

Sophist-UK, 2022-03-19:
Discord user @bax had trouble in 0.7.5 getting camera working - initial settings for all 5 cameras are to point at the same default camera rather than having enumerated the cameras on the system at startup and add any newly found ones to the end of the existing list. Of course, you may only be able to enumerate camera devices that are physically attached to the computer rather than streaming devices - streaming devices may need to be manually added.

jpirnay, 2022-03-19:
> * Properties windows should be changed to a lookup based system with pages that are defined rather than type based different window categories.
> * Properties windows should be changed when the selection changes (not double clicking on something) (I think jpirnay has a stab at that and I thought it was charming but also ties into [rPi and general issues list (Tat's

I am working on this indeed, but my focus is right now to get my code for item manipulation ported to the new Widget methodology

Sophist-UK, 2022-03-19:
1. Collapse All Children on Operations and Elements nodes actually collapse these nodes, whereas for these specific nodes, they should only collapse the children nodes. (Bug in 0.7 and 0.8)
2. Some elements in my standard test file are not being classified - fewer operations are being created and the elements that should be classified into them are not being classified. These are clearly linked, but without looking the code I am not sure which is cause and which effect. (I will create a PR to fix this.)

Sophist-UK, 2022-03-19:
> I am working on this indeed, but my focus is right now to get my code for item manipulation ported to the new Widget methodology

@jpirnay Jens - I certainly didn't intend any criticism of your progress, and knowing @tatarize I am reasonably sure he didn't either. This is a team effort and we are all doing what we can.

Sophist-UK, 2022-03-19:
Emphasised elements / bbox is not working correctly.

Select an operation, children have grey background, but corresponding elements in elements part of tree are not emphasised and bbox is not shown. Select the operation children, and the children elements are emphasised. However elements groups where all (not sure which) children are emphasised should also be emphasised.

Sophist-UK, 2022-03-19:
Simulate is starting bottom left (even though I have not selected Home Bottom in Device Config.

tatarize, 2022-03-20:
* Balor/GRBL settings do not save or load correctly from the crc.
* Beep triggers on data send complete but not on wait-finish at the laser.

tatarize, 2022-03-20:
* user_scale_x/y amounts do not seem to take effect correctly.

tatarize, 2022-03-20:
> Just tried to load the file I saved in 0.8.0RC2 into MK 0.7.5 and instead of two raster ops I got 1 raster op and two unknown 
> I suspect that the Raster op was one generated by Classification and the two unknowns are from the file.

I tested this and it produced no unknown operations.

Sophist-UK, 2022-03-20:
Python crashes on MK 0.8 exit on Windows - not an exception style crash, python itself crashes.

tatarize, 2022-03-20:
Only thing I know that causes a windows crash there is a GraphicPath bug. You can sometimes set one and fail to delete it and it'll crash after the program closes. So if you open the program and don't get a crash if you close immediately it might be one of those.

If you open it draw a rect with tools and it crashes it's likely graphicpath with tools or whatever. So try it a few things.

Sophist-UK, 2022-03-20:
I get it without drawing anything. But perhaps only when I have loaded a file.

closed 2022-03-20 by tatarize