meerk40t/meerk40t #909: RC3

tatarize, 2022-03-20:
### Summary Description

RC3 test code.

### MeerK40t Version

0.8.0 RC-3

### MeerK40t Type

Executable

### Your Operating System

Windows

tatarize, 2022-03-20:
@jpirnay
> I am working on this indeed, but my focus is right now to get my code for item manipulation ported to the new Widget methodology

I mostly finished the properties window to display properties of items in the tree. While it's not perfect (do add things here) it addresses a few issues that cropped up with the RCs and the usability stuff my brother checked with what seems weird and out of place.

jpirnay, 2022-03-21:
The navigation panel needs to be revisited, the size panel therein looks strange, will look at it tomorrow night

tatarize, 2022-03-21:
My brother also was of the opinion, a fairly well informed opinion that the transformation panel is not a navigation widget. That while it might be useful in various ways it does not do anything to navigate the laser. -- There is also the #876 idea and corner lock sub-idea within that. Where it would be entirely reasonable to lock the corner and the selected corner together in some mode.

Eg. You select a different item and if the emphasized object's corner exists within the bed then the laser moves directly to that corner. (or center). When you move the laser, the emphasized selection is moved to the known position of the laserhead.  This would/could be extremely useful. You often are tasked with getting material to line up and it's a major part of the workflow and moving the project or the laser and having it reflected in the other is easy to code but also is what the navigation panel does.

If you're giving it a rethink, can you add that to the thought mix? I mean we might well only need one move things around panel if our options are, move the laser, move the material, or move both. Since the idea of corner locking basically also would accomplish that. And the ability to move the laser by dragging the project around, move the project by using the WASD keys to jog the laser, or issuing a move command with the move_the_laser dragging tool ( #789 ), and having that directly reflected in the other might accomplish those better.

Codewise all moved objects issue a `modified` signal and all fundamentally changed objects issue an `altered` signal. And a change is selection of the currently selected object issues a `emphasized` signal and `driver;position` signals are issued when the driver's position has changed, usually context.device.current_x and context.device.current_y will tell you where in the scene the object is located. But, all of this is pretty each to watch for and adjust when the other things adjust.

The panels are mostly stand-alone so a better solution could be coded up basically from scratch and not mess with the other older things since deprecating such things is doable. But, maybe there's some rethink that would make the whole thing pretty much perfect.

jpirnay, 2022-03-21:
Hmm, I have some ideas… Will give it a try tonight

tatarize, 2022-03-23:
With console open using WASD keys steals focus and causes an infinite loop (likely due to changed console code), need to check 0.7.x for similar bug.

closed 2022-03-27 by tatarize