meerk40t/meerk40t #929: 0.8.0 RC-4 (Request for Comments)

tatarize, 2022-03-27:
The latest Release Candidate is 4.  THIS IS THE LATEST RC.
This issue is for reporting issues with this specific version of MeerK40t.

Please do all further testing on this release. Please do not report any further bugs with previous beta releases. Do give any other types of feedback, questions, concerns or comments.

----
Previous Issue: https://github.com/meerk40t/meerk40t/issues/909
----

Carry-overs from RC-1
- [ ] Continues to try USB on quit; when there is no USB driver selected. 
- [ ] Polygon and polyline ignore quite a lot of mouseclicks
- [ ] Adding elements via the Tools section add those only to the Operations part of the tree, not to the Elements-part (Drawn shapes require a refresh to be seen)
- [ ] Galvo is still up-side down.
- [ ] GRBL still no pre-post gcode.

Carry-overs from RC-2
- [ ] Scale does not apply to Jog moves with pane or navigation. 
- [ ] Right click change raster op speed, was not reflected in the Op label.
- [ ] A few more tries changing with right click / speed and through Op properties and it seems to be a bit hit and miss,.
- [ ] Images and the screen sizes for correct raster sizes may be in error.
- [ ] RasterWizard scripts should defer to the planning stage.
- [ ] Images should have properties to set the DPI of the image. This can often be exactly a raster-step amount on the image but all DPIs are functionally plausible.
- [ ] Simulate is starting bottom left (even though I have not selected Home Bottom in Device Config.
- [ ] Balor/GRBL settings do not save or load correctly from the crc.
- [ ] user_scale_x/y amounts do not seem to take effect correctly.

RC-3 Issues:
- [ ] With console open using WASD keys steals focus and causes an infinite loop (likely due to changed console code), need to check 0.7.x for similar bug.
- [ ] ANSI console wrongly cannot process bytearray. If text is bytearray it channel fails. Error likely in legacy7
- [ ] Extend saving to load and save custom element types.
- [ ] configuration files do no persist if the program crashed during execution.
- [ ] Regmarks do not show up without a refresh.
- [ ] CAG functions have bad icons.
- [ ] CAG objects do not inherit fill and other properties from their parents.
- [ ] GAG-Union can cause long infinite looped pause.
- [ ] Launching with -c causes gui not to launch. Quitting while -c is called causes post-console gui to fail due to kernel shutdown.
- [ ] Launching with -c lets you start the gui with `gui` but then hangs the console.
- [ ] CAG tools only work on one item at a time.
- [ ] People want to fiddle with the write buffer thinking it will fix buffer underruns. This might be better off being suppressed as a feature.
- [ ] Property windows blanks after changing a path's color.
- [ ] Property window doesn't do multiple item manipulations rather adds more things to the notebook.
- [ ] if I select Circle; and draw a circle, and then click circle again. The ribbon button turns off; but doesn’t return me to “regular scene”. Regular scene in itself is kind of wonky as a description. But it should toggle back in some way. Same goes for regular scene. You can click it, it highlights, but then click it again, and it turns off but you are still in “regular scene” for control. 

Manipulation Handles
- [ ] Modifier key check to lock at 45 or 90 degree angles

Flow test ideas.
- [ ] Add in flow text extension text lists in elements.py replaced with {} from text objects during cutplanning. {variable} implies variable incrementing variable. {variable:0} implies a copy of the first variable used. {variable:1} is variable after first variable used. And pointer points to variable after the last one used.
- [ ] Console commands to control the word lists, duplicate every word x times, delete a word, add a word, load flat file from disk, reset pointers.
- [ ] If flow text pointer exceeds data then the current run-batch is degenerate. This should issue a warning during plotplanning.

Longer Term Issues
- [ ] initial settings for all 5 cameras are to point at the same default camera rather than having enumerated the cameras on the system at startup and add any newly found ones to the end of the existing list.
- [ ] Beep triggers on data send complete but not on wait-finish at the laser.

Non-Fix.
* The interpretation of lengths without units (i.e. mils) has changed.

dpticar, 2022-04-07:
After opening file in attachment, and changing Raster to Cut meerk40t is crashing. 

[image]

tatarize, 2022-04-08:
Yeah, I noticed that yesterday when I was trying to get hatch working. I thought for a while and I think my current solution is to remove that combo box and not allow you to shift between them. I don't love that solution but in 0.8.x the node types are different classes rather than typed differences of the same class. So it would need to either replace the node and the settings in the node which seemed like it hit some odd crashes itself or add a new node anyway.

tatarize, 2022-04-09:
Hatch got merged. While the main issue isn't fixed I did remove that functionality so it won't break anymore. This may or may not suck. I don't know how often people switch an operation from one type to another or how much harder it is to add in a replacement of the right type.

closed 2022-04-09 by tatarize