meerk40t/meerk40t #931: [Bug] 0.7,.6b2 Meerk40t.exe doesn't open

Sophist-UK, 2022-03-28:
Not sure what happened to 0.7.6b1. But the standard exe for 0.7.6b2 doesn't open a window. It doesn't crash just goes idle in the background.

joerlane, 2022-03-28:
Working fine on a Mac; and built for every other OS (and ran). Have you tried to launch it from the CLI? Not sure if MS Windows supports that function the same way...

[image]

tatarize, 2022-03-28:
Confirmed.

This probably relates to either introducing a bug which crashes in a way to prevent the gui from launching and runs meerk40t only silently as a thread, or is caused by some of @tiger12506's changes towards getting richtext support for console.

Windows has crap support for cli stuff. You could build a version that is a console version but we don't tend to do that. Windows has two different fairly unrelated modes being gui and console and you can't really make something that does well with both of them.

Sophist-UK, 2022-03-28:
I haven't found a way to bring up a console window in windows when run python without one, but I have found a method of hiding a console window from windows when you create one but don't want one.

So, in theory, if you run pyinstaller **without** the -w option you could have MK code which hides the console window if:

a. You are starting with GUI and
b. You do haven't run a "channel print x" command

and which unhides it if you ever run a "channel print x" command.

From [stackoverflow](https://stackoverflow.com/questions/54116088/bring-up-python-console-window-from-running-pyw-script):
```
import ctypes

def raise_console(console_toggle):
    """Brings up the Console Window."""
    if console_toggle:
        # Show console
        ctypes.windll.user32.ShowWindow(ctypes.windll.kernel32.GetConsoleWindow(), 4)
    else:
        # Hide console
        ctypes.windll.user32.ShowWindow(ctypes.windll.kernel32.GetConsoleWindow(), 0)
    console_toggle = not console_toggle
```

tiger12506, 2022-03-28:
Not sure exactly what's different about my machine, but the exe definitely popped up a crash dialog for me. It was the usual wx._adv, wx._xml popping up again. See pull request #934

tatarize, 2022-03-29:
nuked (removed windows things) and replaced with beta 3

closed 2022-03-29 by tatarize

tatarize, 2022-03-30:
Removing the window with ctypes was attempted. It always produced an unsightly flicker and couldn't be overcome. While it somewhat worked it was not useful enough to subject all the windows users to that pesky flicker since nearly none of them would ever care.

Sophist-UK, 2022-03-30:
yes - always creating a window and then hiding it if you start a GUI does create a flicker.

Which is why I researched and provided example code for how to create a console window when you run pythonw or pyinstaller with -w - so if there is a "channel print x" command, or you want to run without GUI, then you can create a console window for the process and start to use that - no flicker of a console window when it is not wanted, just a console window appearing when it is needed.