meerk40t/meerk40t #962: [Bug] png files still get transparent showing as black rather than white

Sophist-UK, 2022-04-07:
V: 0.7.6

I made these 6 PNG files at the same time using the same steps, yet 5 have black corners and 1 doesn't.

[Meerk40t black corners.zip](https://github.com/meerk40t/meerk40t/files/8443523/Meerk40t.black.corners.zip)

tatarize, 2022-04-09:
The first 5 are Grayscale 8 bit. The final one is RGBA 32 bit. Though apparently it's a palette image with 2 colors. It might be a PA mode image. Clearly a lot of graphic programs are correctly saying this thing has an alpha layer and meerk40t is not seeing it.

tatarize, 2022-04-09:
Should be fixed by #971

The issue is the palette image doesn't have transparency as a channel to be masked off. It has transparency in the info file. That gets converted correctly to RGBA but the code peels off the alpha mask first before any conversion but it doesn't have an alpha like that so it fails.

closed 2022-04-12 by tatarize