meerk40t/meerk40t #963: [Bug] Delete of Operation children deletes elements completely

Sophist-UK, 2022-04-07:
V: 0.7.6000

Delete of children of Operations should only delete opnodes not elements.

tatarize, 2022-04-09:
Hm. It does seem like that should be the case. It might have them emphasized and it's deleting all the emphasized nodes. I'll take another look into this. It's certainly improved, but there's likely an edge case here or there. There's also a time suck for some OSes when multi-selecting items they get unselected one at a time and redrawing them is apparently time consuming.

Sophist-UK, 2022-04-09:
Shouldn't Delete just delete all Selected items (and by iteration, any other nodes needed i.e. delete an element and opnodes get deleted too)?

tiger12506, 2022-04-10:
Most intuitive, IMO, is for delete elements to also remove references to those elements from operations, but deleting from operations should NOT touch the elements themselves. That might be tricky to pull off depending on how stuff is selected vs. emphasized in the tree vs. scene. I suspect it's just side-effects of messing with the code for selected vs. emphasized that is causing weird behavior every once in a while, not a misunderstanding of what delete should do though.

closed 2022-07-20 by tatarize