meerk40t/meerk40t #969: [Bug]: Crash at startup

UlisseDanilo, 2022-04-09:
### Summary Description

Crash at startup on windows 10

### Additional Details

[image]
[image]

### Crash logs

```shell
MeerK40t crash log. Version: 0.7.6000 on Windows:AMD64 - 4.1.1 msw (phoenix) wxWidgets 3.1.5
Traceback (most recent call last):
  File "mk40t.py", line 10, in <module>
  File "meerk40t\main.py", line 422, in run
  File "meerk40t\kernel.py", line 983, in bootstrap
  File "meerk40t\gui\wxmeerk40t.py", line 122, in plugin
  File "meerk40t\kernel.py", line 1956, in console
  File "meerk40t\kernel.py", line 2042, in _console_parse
  File "meerk40t\kernel.py", line 1378, in inner
  File "meerk40t\gui\wxmeerk40t.py", line 464, in window_open
  File "meerk40t\gui\wxmeerk40t.py", line 179, in run_later
  File "meerk40t\gui\wxmeerk40t.py", line 455, in window_open
  File "meerk40t\kernel.py", line 603, in open_as
  File "meerk40t\gui\wxmmain.py", line 268, in __init__
  File "meerk40t\gui\wxmmain.py", line 780, in __set_panes
  File "meerk40t\gui\wxmmain.py", line 950, in on_config_panes
  File "meerk40t\gui\wxmmain.py", line 962, in on_pane_lock
TypeError: '_thread.lock' object is not callable
```

### MeerK40t Version

0.7.6

### MeerK40t Type

Executable

### Your Operating System

Windows

tatarize, 2022-04-09:
Hm. Lemme check...

tatarize, 2022-04-09:
Well at least mine didn't autocrash so it's only some... lemme check that branch and the code in it.

tatarize, 2022-04-09:
```python
                if hasattr(pane.window, "lock"):
                    pane.window.lock()
```
Well it does call a `lock()` function and checked if it existed. Somehow the code is being applied to a _thread object which has a `lock()` function? Hm. Lemme see if I turn on all the panes if I can get something to break there.

tatarize, 2022-04-09:
Seems like it's the console panel. It has a _thread.lock() object on it that tests true for hasattr but then can't be executed but tries anyway.

UlisseDanilo, 2022-04-09:
send me a version and i try it now. TY

tatarize, 2022-04-09:
I see the bug and I've taken care of it. legacy7 branch is fixed. Since it's a bug that prevents startup, I'll push out another version.

tatarize, 2022-04-09:
https://github.com/meerk40t/meerk40t/releases/tag/0.7.6001

Has the correction. The exe files should compile soon.

* Start up preventing bugs are critical and should be fixed as soon as possible.

UlisseDanilo, 2022-04-09:
bug Fixed in 0.7.6001 ! TY

tatarize, 2022-04-09:
Ok. Thank you for the report.

If that's it, you can close the issue or I'll get to it in a few days.

closed 2022-04-09 by UlisseDanilo