meerk40t/meerk40t #973: [Bug]: Crash on start with old .ini file

dpticar, 2022-04-09:
### Summary Description

Version 0.7.6 is crashing with old ini file (in attachment).

Also **I think** 0.8.0-RC4 was crashing, but after removing old ini file and starting 0.8.0-rc4, it's not crashing with old ini file.
[MeerK40t_ini.zip](https://github.com/meerk40t/meerk40t/files/8457391/MeerK40t_ini.zip)

### Crash logs

```shell
Traceback (most recent call last):
  File "mk40t.py", line 10, in <module>
  File "meerk40t\main.py", line 422, in run
  File "meerk40t\kernel.py", line 983, in bootstrap
  File "meerk40t\gui\wxmeerk40t.py", line 122, in plugin
  File "meerk40t\kernel.py", line 1956, in console
  File "meerk40t\kernel.py", line 2042, in _console_parse
  File "meerk40t\kernel.py", line 1378, in inner
  File "meerk40t\gui\wxmeerk40t.py", line 464, in window_open
  File "meerk40t\gui\wxmeerk40t.py", line 179, in run_later
  File "meerk40t\gui\wxmeerk40t.py", line 455, in window_open
  File "meerk40t\kernel.py", line 603, in open_as
  File "meerk40t\gui\wxmmain.py", line 268, in __init__
  File "meerk40t\gui\wxmmain.py", line 780, in __set_panes
  File "meerk40t\gui\wxmmain.py", line 950, in on_config_panes
  File "meerk40t\gui\wxmmain.py", line 962, in on_pane_lock
TypeError: '_thread.lock' object is not callable
```

### MeerK40t Version

0.7.6

### MeerK40t Type

Executable

### Your Operating System

Windows

tatarize, 2022-04-10:
This crash seems to be thread.lock. It's the console bug still but not being used with 0.7.6001. The specific crash there is the same crash that caused the publication of 0.7.6 post1.

The ini file likely loads the console pane which is what was crashing since it had a `lock` attribute which couldn't be executed.

Double check the crash log for version. This crash is only possible in 0.7.6000 and should have been fixed in 0.7.6001

tatarize, 2022-04-10:
0.8.x does not use ini files anymore. The data got moved a bit of a different directory and put in a .cfg file, this is because some of the wx code being used wasn't available to command line interface and a few other issues. But, there is no settings sharing between 0.8.x and 0.7.x.

dpticar, 2022-04-10:
hi,

Sorry I haven't seen version 0.7.6001 , everything is ok with that version, no crash.
Also 0.8.0-rc5 is not crashing on start, probably I messed up versions.

tatarize, 2022-04-17:
I'm pretty sure this is fixed in 6001.

closed 2022-04-17 by tatarize