meerk40t/meerk40t #991: [Bug/Suggestiion] Greedy optimisation (various)

Sophist-UK, 2022-04-15:
1. Open vector subpaths are being cut in optimum order but not being cut in optimum direction
2. Images/rasters are not optimised correctly:
    a. Raster start point appears to be used to determine best travel **afterwards** not raster end point.
    b. Currently images can (in theory) start rastering at 2 points (on the side where the raster is starting from) - but there is currently no code to choose the best corner to start.
    c. There is currently no option to allow MK to choose e.g. whether to raster T2B or B2T and so no way to have rasters reversible and to raster in the opposite direction.
    d. I think that bbox is used to determine the raster start point (for optimisation purposes) rather than an analysis of the first / last raster lines.

tatarize, 2023-12-02:
-- "Raster start point appears to be used to determine best travel afterwards not raster end point."

This is actually a requirement of the lihuiyu raster protocol. While it's not the case for grbl and others. Basically because reversing direction causes the raster step-down it is the case that we must be beyond the current line and the next line before attempting a raster step.

Sophist-UK, 2023-12-02:
I think you misunderstand. **After** the raster has finished it travels to the nearest remaining burn to the raster start point rather than the nearest burn to the finish point.

Sophist-UK, 2024-10-04:
These suggestions are very minor optimisations in terms of time saved (cf. normal greedy optimisation which saves a heap of large travel moves). So I am happy for it to be closed, but please stop marking closed issues as completed when they haven't been completed, and provide a "won't do ever" explanation for closing it.

jpirnay, 2024-10-05:
Reopened to allow someone else to pick them up at a later point in time