meerk40t/meerk40t #1004: [Bug]: ValueError when opening svg

jaredly, 2022-04-20:
### Summary Description

[image]

Here's the SVG that triggers the error

### Additional Details

Looks like it might have to do with the transforms I've got on some svg elements?

### Crash logs

```shell
MeerK40t crash log. Version: 0.8.0006 RC5 git on Darwin:x86_64 - 4.1.1 osx-cocoa (phoenix) wxWidgets 3.1.5
Traceback (most recent call last):
  File "/Users/jared/clone/fork/meerk40t/meerk40t/gui/wxmribbon.py", line 98, in button_click
    my_code(0)  # Needs a parameter....
  File "/Users/jared/clone/fork/meerk40t/meerk40t/gui/wxmmain.py", line 558, in <lambda>
    "action": lambda e: kernel.console(".dialog_load\n"),
  File "/Users/jared/clone/fork/meerk40t/meerk40t/kernel/kernel.py", line 2014, in console
    self._console_parse(command, channel=self._console_channel)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/kernel/kernel.py", line 2093, in _console_parse
    data, remainder, input_type = command_funct(
  File "/Users/jared/clone/fork/meerk40t/meerk40t/kernel/functions.py", line 239, in inner
    returned = func(command=command, channel=channel, **ik, **kwargs)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/gui/wxmmain.py", line 1014, in load_dialog
    gui.clear_and_open(pathname)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/gui/wxmmain.py", line 2318, in clear_and_open
    if self.load(pathname):
  File "/Users/jared/clone/fork/meerk40t/meerk40t/gui/wxmmain.py", line 2336, in load
    results = self.context.elements.load(
  File "/Users/jared/clone/fork/meerk40t/meerk40t/core/elements.py", line 6229, in load
    results = loader.load(self, self, pathname, **kwargs)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/core/svg_io.py", line 496, in load
    svg = SVG.parse(
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 8712, in parse
    s = Group(values)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 7488, in __init__
    SVGElement.__init__(self, *args, **kwargs)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 3301, in __init__
    self.property_by_values(self.values)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 7510, in property_by_values
    Transformable.property_by_values(self, values)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 3344, in property_by_values
    self.transform = Matrix(values.get(SVG_ATTR_TRANSFORM, ""))
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 2471, in __init__
    self.parse(m)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 2631, in parse
    self.pre_translate(x_param, y_param)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 2903, in pre_translate
    self.pre_cat(Matrix.translate(tx, ty))
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 2878, in pre_cat
    self.a, self.b, self.c, self.d, self.e, self.f = Matrix.matrix_multiply(
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 3041, in matrix_multiply
    s.a * m.e + s.c * m.f + s.e * 1,
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 762, in __add__
    c += other
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 664, in __iadd__
    raise ValueError
ValueError

<_io.TextIOWrapper name='MeerK40t-2022-04-20_14_21_06.txt' mode='w' encoding='UTF-8'>
meerk40t on  main took 3m36s ➜ pipenv run python Meerk40t.py
DXF plugin could not load because ezdxf is not installed.
OpenCV is not installed. Disabling Camera. Install with:
        pip install opencv-python headless
GRBL plugin could not load because pyserial is not installed.

MeerK40t crash log. Version: 0.8.0006 RC5 git on Darwin:x86_64 - 4.1.1 osx-cocoa (phoenix) wxWidgets 3.1.5
Traceback (most recent call last):
  File "/Users/jared/clone/fork/meerk40t/meerk40t/gui/wxmribbon.py", line 98, in button_click
    my_code(0)  # Needs a parameter....
  File "/Users/jared/clone/fork/meerk40t/meerk40t/gui/wxmmain.py", line 558, in <lambda>
    "action": lambda e: kernel.console(".dialog_load\n"),
  File "/Users/jared/clone/fork/meerk40t/meerk40t/kernel/kernel.py", line 2014, in console
    self._console_parse(command, channel=self._console_channel)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/kernel/kernel.py", line 2093, in _console_parse
    data, remainder, input_type = command_funct(
  File "/Users/jared/clone/fork/meerk40t/meerk40t/kernel/functions.py", line 239, in inner
    returned = func(command=command, channel=channel, **ik, **kwargs)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/gui/wxmmain.py", line 1014, in load_dialog
    gui.clear_and_open(pathname)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/gui/wxmmain.py", line 2318, in clear_and_open
    if self.load(pathname):
  File "/Users/jared/clone/fork/meerk40t/meerk40t/gui/wxmmain.py", line 2336, in load
    results = self.context.elements.load(
  File "/Users/jared/clone/fork/meerk40t/meerk40t/core/elements.py", line 6229, in load
    results = loader.load(self, self, pathname, **kwargs)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/core/svg_io.py", line 496, in load
    svg = SVG.parse(
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 8712, in parse
    s = Group(values)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 7488, in __init__
    SVGElement.__init__(self, *args, **kwargs)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 3301, in __init__
    self.property_by_values(self.values)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 7510, in property_by_values
    Transformable.property_by_values(self, values)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 3344, in property_by_values
    self.transform = Matrix(values.get(SVG_ATTR_TRANSFORM, ""))
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 2471, in __init__
    self.parse(m)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 2631, in parse
    self.pre_translate(x_param, y_param)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 2903, in pre_translate
    self.pre_cat(Matrix.translate(tx, ty))
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 2878, in pre_cat
    self.a, self.b, self.c, self.d, self.e, self.f = Matrix.matrix_multiply(
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 3041, in matrix_multiply
    s.a * m.e + s.c * m.f + s.e * 1,
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 762, in __add__
    c += other
  File "/Users/jared/clone/fork/meerk40t/meerk40t/svgelements.py", line 664, in __iadd__
    raise ValueError
ValueError
```

### MeerK40t Version

fcc08ef474118980a8a888ba70cc3a499dbac741

### MeerK40t Type

Git

### Your Operating System

Mac (before M1/Bigsur)

jaredly, 2022-04-20:
Ah, the issue is that I've specified units on `translate(2mm, -2mm)`.

tatarize, 2022-04-22:
That sounds legal in svg and css.

closed 2022-11-09 by jpirnay