meerk40t/meerk40t #1019: [Bug] SVG linecaps not respected - always rastered as "round"

Sophist-UK, 2022-04-30:
V0.7.6

Non-closed SVG paths are displayed in scene and simulation (and so presumably burned) with round `stroke-linecaps` rather than the SVG stroke-linecaps butt or square defined in SVG (in Inkscape). 

I suspect (but haven't checked) that `stroke-linejoin` is similarly mis-rastered.

tatarize, 2022-04-30:
It's not taken into account. I think it might be possible to do (some of them), but there's only JOIN_BEVIL, JOIN_ROUND and JOIN_MITER and round is just the default value.

Sophist-UK, 2022-04-30:
Linecaps have a different set of values to linejoins.

Both are broken.

jpirnay, 2022-05-18:
Should be fixed with [PR 1046](https://github.com/meerk40t/meerk40t/pull/1046)