meerk40t/meerk40t #1020: [Bug] fill-rule:nonzero not respected

Sophist-UK, 2022-04-30:
V0.7.6

Overlapping text-to-path with explicit `fill-rule:nonzero` is rastered as `fill-rule:evenodd`

tatarize, 2022-05-08:
This is accurate. I'm not really sure why anybody would want the different fill rule but it should be pretty clear from loading that the non-zero isn't respected and all fill rules are evenodd.

tatarize, 2022-05-08:
I don't know any way to set the fill rule in wxPython, and it should be fairly obvious from the preview within MK that the rule will not be respected and from the simulation. This isn't really a bug, it's a defect since it's not an error it's just something that wasn't implemented since it seemed like it should be fairly rare to need such a thing. Though I'll apparently be hacking in end-caps and line-joins when I get free time next week.

Sophist-UK, 2022-05-27:
> I'm not really sure why anybody would want the different fill rule but it should be pretty clear from loading that the non-zero isn't respected and all fill rules are evenodd.

You want this because if you convert text to path and then merge the paths (to reduce the number of objects) then overlapping text has white spots.

tatarize, 2022-05-27:
Well, I get people might want it but I don't know anyway to make wxPython set a fill rule. If I can't draw it with wxPython then I can't render it with wxPython and fill rules are not able to be done. I could be looking in the wrong spot jpirnay did point out where gradients were hiding so there's certainly some things that aren't yet implemented but totally could be, but I don't know if or where fill rules could be set for a path to be drawn in wxPython.

Sophist-UK, 2022-05-27:
https://docs.wxpython.org/wx.GraphicsContext.html#wx.GraphicsContext.FillPath

`Fillpath` takes a `fillstyle` parameter wx.ODDEVEN_RULE or wx.WINDING_RULE.

In master:
* https://github.com/meerk40t/meerk40t/blob/main/meerk40t/gui/laserrender.py#L411=
* https://github.com/meerk40t/meerk40t/blob/main/meerk40t/gui/laserrender.py#L461=

In legacy7:
* https://github.com/meerk40t/meerk40t/blob/legacy7/meerk40t/gui/laserrender.py#L307=
* https://github.com/meerk40t/meerk40t/blob/legacy7/meerk40t/gui/laserrender.py#L332=

tatarize, 2022-05-27:
Yeah, that'll work. I'll hack it in where jpirnay put the end cap stuff. Should be about the same and needs a sort of per-path rendering or whatever.

Sophist-UK, 2022-05-27:
Whatever.

tatarize, 2022-05-27:
I raised the question with the wxpython folks.

https://discuss.wxpython.org/t/does-polygonfillmode-actually-not-include-non-zero/36003

It seems weird it would have winding but not non-zero.

Sophist-UK, 2022-05-27:
AFAICT "nonzero" is normally a shortened form of "nonzero winding" so they may actually be the same.

And wxpython is only a layer on wx and wx gets its svg from [nanosvg](https://github.com/memononen/nanosvg/tree/master/src). So good luck with that.:-)

tatarize, 2022-05-27:
nonzero winding is different than the actual non-zero rule. Unless I'm wrong on the svg spec. Ah, I am. nonzero actually does imply the winding rule. I always just considered that to be universal insideness as a rule.

jpirnay, 2022-05-27:
Whatever :-) Implemented it...

closed 2022-05-28 by jpirnay