meerk40t/meerk40t #1021: [Bug] SVG save of base64 non-conformant with RFC4648

Sophist-UK, 2022-04-30:
v0.7.6

RFC4648 Section 3.1 limits Base64 encoding to lines of no more than 76 characters.

Inkscape saved SVGs with Base64 encoding conform to this, but MK saved SVG files save the data as a single line.

tatarize, 2022-05-02:
This is accurate I posted a branch to try to conform to RFC2045 though the writing of newlines within the xml is a bit of a pain since it gets idiomized. I opened a SO question since I couldn't google any solid information on the problem and I have limited time just now.

https://stackoverflow.com/questions/72085514/write-a-literal-newline-in-elementtree-attribute

tatarize, 2022-05-04:
I'm not sure this is possible without writing my own xml writer, and I'm pretty sure it's not worth the implementation time. Though it does look better it has little impact on the current results.

Sophist-UK, 2022-05-04:
Agreed not worth the effort.

closed 2022-05-08 by tatarize