meerk40t/meerk40t #1049: Does MK support Inkscape 1.2 (released 16 May 2020) "multiple pages"?

Sophist-UK, 2022-05-25:


tatarize, 2022-05-28:
No. And I'm not sure how that would work even if it did. In theory if their implementation just put it in a different page it might load both pages stuff.

tatarize, 2022-05-28:
I'd likely want to see it in the spec, and maybe a few workflows that would make sense.

Sophist-UK, 2022-05-28:
> No. And I'm not sure how that would work even if it did. In theory if their implementation just put it in a different page it might load both pages stuff.

I think it would need to be treated as a group.

tatarize, 2022-06-20:
Currently it's processed in the namedview:
```xml
  <sodipodi:namedview
     id="namedview7"
     pagecolor="#ffffff"
     bordercolor="#666666"
     borderopacity="1.0"
     inkscape:showpageshadow="2"
     inkscape:pageopacity="0.0"
     inkscape:pagecheckerboard="0"
     inkscape:deskcolor="#d1d1d1"
     inkscape:document-units="mm"
     showgrid="false"
     inkscape:zoom="0.2102413"
     inkscape:cx="525.58655"
     inkscape:cy="558.88162"
     inkscape:window-width="1920"
     inkscape:window-height="1018"
     inkscape:window-x="-8"
     inkscape:window-y="497"
     inkscape:window-maximized="1"
     inkscape:current-layer="layer2">
    <inkscape:page
       x="0"
       y="0"
       width="210"
       height="297"
       id="page3950" />
    <inkscape:page
       x="279.38135"
       y="55.372883"
       width="205.13135"
       height="298.25848"
       id="page3952" />
```

The default SVG viewbox is:

```xml
<svg
   width="210mm"
   height="297mm"
   viewBox="0 0 210 297"
   version="1.1"
```

Which means this is a hack where the pages are registered within inkscape. The second page is outside the typical page and merely occupies more euclidean space. So when you do pages, it's just giving alternative viewboxes. If we set our viewbox to the second page there namely `page3952` it would give me the objects in the second page.

All objects however exist within the same space. And a default opening of the file with meerk40t would open all the objects regardless which page-space they are in and open them relative to the primary viewbox.

This is actually really clever but inkscape specific. I filed a request with the WC3 Working Group. Though that can take years even if they love the idea, which actually is a very lovely idea. 

---

MK-wise it wouldn't be too hard to scale things based on a different page. But, since I only extract the geometry it's a bit harder to turn objects on and off if they fail to fall within the boundaries of a page. This would basically ask that all objects that fall outside the bounds of the bed, be disabled and placed in regmark (or simply turned off). Then when we shift pages, we translate and scale based on the new coordinates.

In theory, much like we have `scene focus` where we focus on a particular section of the scene. We could have `scene page` where we could shift the coordinate systems such that the given page fall within the bed-bounds.

This also provides some interesting corollaries, for example the array-grid object, would offset the current location and run the same cutcode again in the new offset location. This could also be somewhat similar to a page, in that we would shift our position, and those things which fall outside that page are not rendered.

So if we do a 4 x 5 grid of identical objects or wordlist modified objects, this is somewhat the opposite of paging where we'd actually do the same object in the same place and counter-position the data. So we could have a real 4x5 grid of different objects and we'd run them overlapping each other as 20 different overlapping layers. 

At a basic level the Inkscape form of paging (though ignoring that it would imply scaling) could be seen as a laser-origin to object-origin shift. So we could start our same job located in the job-space at 0,0. And then repeat that job by shifting the laser-origin into a 4x5 grid. This is what step-repeat/grid does. With paging, we'd be doing the oppose. We'd take, for example, a physically moved 4x5 grid and run them all at 0,0.

For example, we could perform a z-raster splitting an image into 8 different images that are expected to run overlapping each other. And somehow (the gui methodologies here aren't clear) flag them run relative to their custom provided `(0,0)` thereby making their physical position within the space a moot point.

tatarize, 2022-06-22:
That sounds a bit like just having layers. Which is somewhat doable, apparently inkscape just has them as a fancy type of group. Which would be perfectly reasonable, turning all descendants off and on in visibility and likely execution. So each node would be flagged with `output` and `show`.

Then you trigger these likely with a right click (since wx tree probably can't support items with visible, enabled tree buttons). If you flag a layer as output, all items within that layer are disabled. Whereas if you flag a layer as visible these either show or hide them from being displayed.

Sophist-UK, 2022-06-22:
In my head, items would always be shown in the tree. Disabled items would be greyed out in the tree and not shown in the scene. If it is possible, I might put a radio box next to Pages and a checkbox next to every other item.

tatarize, 2022-06-22:
It might be possible, just not in the wxTreeCtrl thing. But, currently there's only maybe right clicking or whatever.

closed 2024-10-04 by jpirnay