meerk40t/meerk40t #1052: [Bug] Paths with no fill load with fill

Sophist-UK, 2022-05-27:
SVG has (excerpted for key info):
```
<style
   type="text/css"
   id="style2944">
   <![CDATA[
    .fil1 {fill:#4D897C}
   ]]>
</style>
<path
    class="fil1"
   style="fill:none;fill-opacity:1;stroke:#ff0000;stroke-width:9.8404;stroke-dasharray:none;stroke-opacity:1"
/>
```

[Seahorse.svg](https://user-images.githubusercontent.com/3001893/170679021-7c2dae9b-862d-4c81-ba5a-7b1a4337a6f3.svg)

[image]

[image]

jpirnay, 2022-05-30:
svgelements needs a fix:
Lines 8632 ff read as:
```
8631:               # Split style element into parts; priority highest
8632:                if SVG_ATTR_STYLE in attributes:
8633:                    style += attributes[SVG_ATTR_STYLE]```
```
a ";" needs to be inserted...
```
8631:              # Split style element into parts; priority highest
8632:                if SVG_ATTR_STYLE in attributes:
8633:                    if len(style) != 0:
8634:                        style += ";"
8635:                    style += attributes[SVG_ATTR_STYLE]```
```
et voilà:
[image]

jpirnay, 2022-05-31:
Resolved