meerk40t/meerk40t #1105: [Bug]:

jaredly, 2022-06-14:
### Summary Description

Clicking "Set Current" on the config screen produces an exception (trace bwlo). This is with the k40 plugged in, m2nano board, and jogging around manually works.

### Crash logs

```shell
MeerK40t crash log. Version: 0.8.0007 RC6 git on Darwin:x86_64 - 4.1.1 osx-cocoa (phoenix) wxWidgets 3.1.5
Traceback (most recent call last):
  File "/Users/jared/clone/fork/meerk40t/meerk40t/lihuiyu/gui/lhystudiosdrivergui.py", line 419, in on_button_set_home_current
    native_x = self.context.device.native_x
AttributeError: 'LihuiyuDevice' object has no attribute 'native_x'
```

### MeerK40t Version

0.8.0007 RC6

### MeerK40t Type

Git

### Your Operating System

Mac (M1/Bigsur)

tatarize, 2022-06-15:
Okay. Easy fix. It's running older code for that. Might be a lot of other little things like that in 0.8.x but I need to finish up the corrections to rasterwizard and go down the list looking for them.

tatarize, 2022-06-20:
This is fixed in latest main. I'll close this next RC.

closed 2022-07-01 by tatarize