meerk40t/meerk40t #1106: Animations for Widgets/Scene

tatarize, 2022-06-15:
While it hasn't mattered much yet, the original code I wrote in android for scene and widget stuff had animations and I never ported any of that stuff over. Doing so would allow widgets access to animations. Basically they call `start()` this adds them into the timed elements list in scene which performs ticks in its own thread (this would likely be in `kernel.threaded()`) this thread adds and removes anything that are in the add-remove lists (in a synchronized fashion) and then performs the ticks on the list of animated widgets. These widgets are expected to call stop() when they are done but otherwise animate by updating their current position (where they are drawn), until they reach their goal position (where they stop).

There's not much known need for this yet, but some animations like moving the laser head position over several ticks rather than instantly can help explain to the user what's going on. There's also some things like direction of marching ants to show the directionality of a curve that be useful in laser software.

Widgets would need to implement:
```java

    public void start() {
        getScene().addTimedElement(this);
    }

    public void stop() {
        getScene().removeTimedElement(this);
    }

    void tick();

    void onStartMain();

    void onStartThreaded();

    void onStopThreaded();

    void onStopMain();
```

Scene would need to permit widget animations. 

```java
    @Override
    public void run() {
        ArrayList<TimedElement> runningElements = new ArrayList<>();
        halting = false;
        while (true) {
            if (!addTimed.isEmpty()) {
                synchronized (SceneView.this) {
                    for (int i = 0; i < addTimed.size(); i++) {
                        TimedElement element = addTimed.get(i);
                        if (!runningElements.contains(element)) {
                            element.onStartThreaded();
                            runningElements.add(element);
                        }
                    }
                    addTimed.clear();
                }
            }
            if (!removeTimed.isEmpty()) {
                synchronized (SceneView.this) {
                    for (int i = 0; i < removeTimed.size(); i++) {
                        TimedElement element = removeTimed.get(i);
                        element.onStopThreaded();
                        runningElements.remove(element);
                        notifyStoppedElement(element);
                    }
                    removeTimed.clear();
                    goInvalidate();
                }
            }
            if (halting) {
                synchronized (SceneView.this) {
                    for (TimedElement widget : runningElements) {
                        widget.onStopThreaded();
                        notifyStoppedElement(widget);
                    }
                    runningElements.clear();
                    goInvalidate();
                    halting = false;
                }
            }
            if (runningElements.isEmpty()) {
                break;
            }
            long timeBetweenTicks = System.currentTimeMillis() - lastTick;
            long waitTick = tick - timeBetweenTicks;
            try {
                Thread.sleep(Math.max(waitTick, TICK_REQUIRED));
            } catch (InterruptedException e) {
                break;
            }
            lastTick = System.currentTimeMillis();
            synchronized (SceneView.this) {
                for (TimedElement widget : runningElements) {
                    widget.tick();
                }
            }
            //not forced full revalidate, just partial if requested.
            goInvalidate();
        }
        getActivity().runOnUiThread(new Runnable() {
            @Override
            public void run() {
                running = false;
                if (!addTimed.isEmpty()) startIfNeeded();
            }
        });

    }

    private void notifyStoppedElement(@NonNull final TimedElement element) {
        getActivity().runOnUiThread(new Runnable() {
            @Override
            public void run() {
                element.onStopMain();
                goInvalidate();
            }
        });
    }
    public void addTimedElement(@NonNull TimedElement widget) {
        synchronized (SceneView.this) {
            if (!addTimed.contains(widget)) {
                widget.onStartMain();
                addTimed.add(widget);
            }
        }
        startIfNeeded();
    }

    public void removeTimedElement(TimedElement widget) {
        synchronized (SceneView.this) {
            if (running) {
                if (!removeTimed.contains(widget)) {
                    removeTimed.add(widget);
                }
            }
        }
    }

    private void startIfNeeded() {
        synchronized (SceneView.this) {
            if (running) return;
            running = true;
            Thread thread = new Thread(this);
            thread.start();
        }
    }

    private void halt() {
        if (running) {
            halting = true;
        }
    }
```

jpirnay, 2022-06-20:
That's interesting - I was working on something along that line as well - my usecase was somehwat along the ants-stuff for emphasized elements to make them visually stand out.
I wanted to have three GDI buffers, the main one and then two more that would be populated by drawing the outer shape of emphasized stuff with two different stroke-types that would blend over properly.
If # emphasized = 0 no need to do something, if != 0 then every half a second blit the second buffer with xor over the main buffer.

tatarize, 2022-07-22:
This is live in #1188.

closed 2022-07-22 by tatarize