meerk40t/meerk40t #1160: [0.8.0Beta5] Issues

Sophist-UK, 2022-07-12:
The latest BETA release of 0.8.0 is Beta 5. **This issue is for reporting issues with this specific version of MeerK40t.**

Please do all further testing on this release. Please do not report any further bugs with previous beta releases. Do give any other types of feedback, questions, concerns or comments.

Sophist-UK, 2022-07-12:
* Config - window is wider than the default size and has a horizontal scrollbar.
* Config - Limit write buffer is checked but input field is locked.
* Preferences - default window size has vertical scrolling - can we make it bigger?
* Preferences - can we put the first two options (DPI Scaling / Shutdown) into a group and put it under the SVG PPI group?
* Preferences - add text to Language tooltip to say that MK needs restarting after changing language.
* SVG save - `Operations` tag is not SVG compliant - should be `Meerk40t:Operations`
* Tree / SVG load - inkscape labels are not shown.
* Tree / SVG load - element details not shown - not sure that this is a major problem - but some attributes (like stroke and fill colours) can be useful.
* Tree - image label in elements different from image label in Operations - resolution is useful in elements - can we have name/label + resolution + step in both tree areas?
* Tree - entries in operations have a * prefix - what does this mean?
* Tree - default operations no longer have a tick mark indicating default.
* Tree SVG tag is now an extra group - is this needed? (Can you have more than one SVG tag per file?)
* Tree Context menu - `Remove file` is now `Delete group` which is confusing.
* Tree Context menu - No right click entry on images for Raster Wizard (only apply script).
* **Position pane does not work!**

jpirnay, 2022-07-13:
> Position pane does not work
Works perfecly for me - you need to press enter to apply changes

Sophist-UK, 2022-07-13:
* Selecting in the scene should give some indication in the tree what has been selected if the parent nodes are closed - perhaps an underscore.
* Should selecting in the scene "select" elements in the tree rather than (as present) "emphasising" them?
* Selecting in the scene all elements in a group should (IMO) emphasise the group also (or possibly select the group and emphasise the children).

Sophist-UK, 2022-07-13:
> * ANSWER: Indicates this node to be a reference node

What's a reference node?

jpirnay, 2022-07-13:
The link to an element appearing under operations - it's no real element but linking to it's parent. The "virtual copy" to be handled within the operation

Sophist-UK, 2022-07-13:
Is this indicator of a reference node actually necessary? Aren't nodes in the elements branch originals and nodes in the Operations branch copies?

If it is necessary could we use something like a '.' instead of '*'?

tatarize, 2022-07-15:
> Is this indicator of a reference node actually necessary? Aren't nodes in the elements branch originals and nodes in the Operations branch copies?

No. Copies would be if they were copied and a second copy of the original data existed. It doesn't. These are references. They simply refer to the original data. The distinction might not be needed, but it's certainly useful to understanding. These are more common in Computer Science, but they are sort of minimized in that they are indicated by merely `*`.

tatarize, 2022-07-16:
I think the "*" is fine, but it is the case that it's not correctly duplicating the information on the other nodes. This is because of some shortsighted code where I call the `__str__()` function of the node which doesn't get processed through the formatter. So we end up with some nodes that use old str() creation stuff and newer formatter stuff and the info doesn't mesh. (correcting now)

tatarize, 2022-07-16:
Also, as part weirdly, there's a function that happens during cutplanning where we take these elements and we replace the references with actual genuine copies. This is so that there's no knock-on effects modifying references, and because we need to scale these to the native resolution of the actual device we're using.

jpirnay, 2022-08-05:
Continued in https://github.com/meerk40t/meerk40t/issues/1224

closed 2022-08-05 by jpirnay