meerk40t/meerk40t #1162: [0.7.7] Delete key on selected operation nodes deletes the element nodes too!!!!!

Sophist-UK, 2022-07-12:
This is just plain wrong.

The delete functionality is easy - it deletes **selected** nodes and any children and does not delete **emphasised** nodes.

I fixed this in #793 so no idea why it is now broken again. But without Undo it is **bloody annoying** when you try to delete opnodes and it deletes the elements too.

jpirnay, 2022-07-12:
Fixed in 0.8

tiger12506, 2022-07-12:
Well the conflicting demands here seem to be conflicting definitions of what emphasized means. It was my understanding that emphasized was a workaround to support broken tree selection code on other platforms (mac). The idea being that emphasized and selected would be kept in sync. So it then came to be that emphasized meant 'selected in the scene' and selection meant 'clicked on in the tree'.
However, the code for that has been changed at least a thousand times probably before, during, and since you looked at it Sophist because the emphasized->selection->emphasized event handling was recursively triggering on other platforms and causing massive slowdowns. And then changed again because of broken behavior. Etc. etc. rinse, repeat.

Sophist-UK, 2022-07-12:
My understanding of `emphasised` is that it is children of selected nodes **and** equivalent nodes in the other part of the tree i.e. opnodes for selected / emphasised element nodes and possible (or maybe not) vice versa.

It seems to me that there needs to be clarity on the definition of these things otherwise:

a. The code to emphasise nodes may end up being incorrect;
b. Selected nodes may be used when emphasised nodes should be used or vice versa.

jpirnay, 2022-07-24:
We can leave this open for 0.7, but it's confirmedly fixed in 0.8

closed 2023-03-13 by tatarize