meerk40t/meerk40t #1163: [0.7.7] Classification no longer classifies filled items to matching raster operations

Sophist-UK, 2022-07-12:
The 0.7 classification routine used to classify filled elements to a matching Raster operation with the exact same colour (if it existed). Because you have to preserve and respect the order of filled items in a raster operation, it would group overlapping items and then add the whole group to any non-default Raster operations that had the exact same colour as any element in the group.

This meant that for test designs (where you create many non-overlapping raster items of slightly different colours and matching operations with e.g. different speeds or powers) you could save and restore and the items would be re-classified back again automatically.

But now when I restore, my test designs have every filled element allocated to every Raster operation and then takes a **lot** of effort to manually get right again.

jpirnay, 2022-07-12:
Would you care to share your test designs? These could end up in a unit test

Sophist-UK, 2022-07-12:
[Step vs Speed LASER7.zip](https://github.com/meerk40t/meerk40t/files/9096845/Step.vs.Speed.LASER7.zip)

Sophist-UK, 2022-07-12:
See also the 0.8Beta5 issues list - because node labels have far, far less info, and particularly do not show the colours, it is actually now impossible to manually drag & drop elements to Raster operations with matching colours without opening the properties and then the colour picker to see exactly what colour each element and operation is.

tatarize, 2022-07-15:
[image]

Is the current 7.7 logic, it classified everything in all the rasters.

[image]

8.0 seems to classify everything in a Disabled Raster *and* the first raster.

The fill seems to confirm that these fill values are *mostly* black.

```
[06:21:57] fill
[06:21:57]     ----------
[06:21:57]     Fill Values:
[06:21:57]     0: fill = none - Path:rect1179
[06:21:57]     1: fill = #040004 - Path:rect411
[06:21:58]     2: fill = #040003 - Path:rect1998
[06:21:58]     3: fill = #040002 - Path:rect2000
[06:21:58]     4: fill = #040001 - Path:rect2002
[06:21:58]     5: fill = #040000 - Path:rect2004
[06:21:58]     6: fill = #030004 - Path:rect2100
[06:21:58]     7: fill = #030003 - Path:rect2102
[06:21:58]     8: fill = #030002 - Path:rect2104
[06:21:58]     9: fill = #030001 - Path:rect2106
[06:21:58]     10: fill = #030000 - Path:rect2108
[06:21:58]     11: fill = #020004 - Path:rect2112
[06:21:58]     12: fill = #020003 - Path:rect2114
[06:21:58]     13: fill = #020002 - Path:rect2116
[06:21:58]     14: fill = #020001 - Path:rect2118
[06:21:58]     15: fill = #020000 - Path:rect2120
[06:21:58]     16: fill = #010004 - Path:rect2124
[06:21:58]     17: fill = #010003 - Path:rect2126
[06:21:58]     18: fill = #010002 - Path:rect2128
[06:21:58]     19: fill = #010001 - Path:rect2130
[06:21:58]     20: fill = #010000 - Path:rect2132
[06:21:58]     21: fill = #000004 - Path:rect2136
[06:21:58]     22: fill = #000003 - Path:rect2138
[06:21:58]     23: fill = #000002 - Path:rect2140
[06:21:58]     24: fill = #000001 - Path:rect2142
[06:21:58]     25: fill = #000000 - Path:rect2144
[06:21:58]     26: fill = #010101 - Path:path4380
[06:21:58]     ----------
```

0.8.x will very much not care and just classifies anything with a set fill into a raster operation. 0.7.7 classification works this way regardless whether legacy or non-legacy classification is used.

In 0.8.x the issue would be mostly resolved by user classification being saved to disk. So the original classification would merely hold without resorting to minor tweaks to get things to classify in this manner.

---

For what it's worth, that classification scheme cited, only actually kicks in if `rasters_one_pass = True` and since the first and second raster operations do not share the same op_color then rasters_one_pass gets set False and stays False.

The last time this was touched was 11 months ago by Sophist.

https://github.com/meerk40t/meerk40t/commit/9be5b3e9aad1d0bf5ace8ac605896a03dfad3be9

tatarize, 2022-07-15:
That version also classified the given file the same way. This was `0.7.0 RC11ds`.

tatarize, 2022-07-15:
I can find no version where this classified the way described. Going back to the pre-RC10 days.

Sophist-UK, 2022-07-15:
Hmmm - actually the code is still there in 0.7.7, but apparently not working.

https://github.com/meerk40t/meerk40t/blob/1f6b80eb7f55d730ae6aaeaac84c9d33b49bb236/meerk40t/core/elements.py#L7531-L7557

tatarize, 2022-07-15:
Tracing back, it's never worked. If I forced `rasters_one_pass = True` that behavior did actually kick in. I assume it's an edge case where if each raster is a different color then elements should be classified based on their fill color rather than stroke. But, the check there has always been checking that the color != the previous color and then it sets that to False. Which is checking that all rasters are the same color rather than all rasters are different. Thus triggering the behavior wrongly.

But, it appears that it has never classified such operations correctly so nobody else is going to have such a test file that stopped working because that edge case never worked.

Sophist-UK, 2022-07-16:
Nevertheless, calibration files are a use case, and IMO if we want to encourage a more scientific usage of lasers it is a reasonably important one (though not as important as real production burns of course).

jpirnay, 2022-07-17:
The amended classification routines in 0.8b6 should address most if not all of this

Sophist-UK, 2022-07-17:
Once the executable is available I'll take a look.

closed 2022-08-12 by jpirnay