meerk40t/meerk40t #1187: Problem when importing DXF or DWG files[Bug]:

LeonardoCT, 2022-07-22:
### Summary Description

DXF files not importing correctly. 

### Additional Details

When i try to import dxf or dwg files the round edges become straight:

[2D cad software]:
[image]
[meerk40T]:
[image]
[K40 whisperer]:
[image]
As you can see, the 2D drawing is fine, when i import it in K40 whisperer it works fine, but when i try to use Meerk40T the round edges become straight, and i have no idea how to fix this, as it seems like an import issue.
Thanks

### Crash logs

```shell
Nothing to put here
```

### MeerK40t Version

V0.7.7001

### MeerK40t Type

Executable

### Your Operating System

Windows

tatarize, 2022-07-24:
I might need an example of that file. Basically dxf has a *lot* of different bits that can be messed up and figuring out what's wrong requires an example file. The smaller the better. Removing as much as you can that isn't relevant to the failure is certainly helpful.

LeonardoCT, 2022-07-25:
[Koala.dxf.zip](https://github.com/meerk40t/meerk40t/files/9184181/Koala.dxf.zip)
Here is the file, however i found a very simple work around,
i switched to svg files and it works perfectly

tatarize, 2022-09-11:
[image]

closed 2022-09-11 by tatarize