meerk40t/meerk40t #1199: [Bug]: Clip Image to Path in SVG not working

markdesouza, 2022-07-27:
### Summary Description

In Inkscape 1.2, I created an svg that contained an embedded image that is clipped to a path (a rectangle with rounded corners). When loaded in meerl40t, the image is loaded unclipped. When loading in k40whisperer, the image is correctly clipped.

### MeerK40t Version

0.7.7001

### MeerK40t Type

Executable

### Your Operating System

Raspberry Pi

jpirnay, 2022-07-28:
The underlying graphics library of wxpython (the GUI framework Meerk40t is using) has no clippath functionality. The complete graphics engine would need to be ported to something different (i.e. https://docs.wxpython.org/wx.lib.wxcairo.html ). 
So there is only a limited chance this will be resolved in the near future. Rectangular clips are supported though, but I am not sure whether it would make sense to implement this feature-subset.

tatarize, 2022-07-28:
I don't think that's entirely true. While the clip-path for cairo is an issue, the Image to Path clipping might be a lot more doable because here we have Pillow. We can render the filled shape giving us a black and white image. Then we can use Pillow to clip the image down to just the pixels that rendered. 

* We'd need the ability to add clip-paths to the `elem image` nodes. And we'd need the ability to have those clip paths create transparent pixels outside the clip-path and opaque ones within them but it would be possible.

tatarize, 2022-07-28:
Also see #70.

Sophist-UK, 2022-07-28:
This is another one of those things (like the new Raster Scripts) that you would need to register against the object(s) when the user set the clip path, but actually run at burn time (because - in Inkscape at least - you can move the clip path or image and the clipping changes dynamically).

tatarize, 2022-07-28:
Yeah, and you'd likely want to be able to scale either or both. So you'd need the clip to be able to move and scale relative to the image etc. So you'd need to be able to add a heart shape, over an image. Then you'd want to be able to adjust the image inside the clip. So where the clip occurs relative to the image.

I did some work with some yearbook software and I assume some others work the same way but it let you click and move things. Then if you double clicked it would let you move the actual image itself the clip box. So you could have a full sized image there and you'd zoom in on one tiny section. So you'd typically move the clip-box around and the size of the widget would be the size of the clipbox there. Then when you need you get the full image under that which can be moved around and scaled such that you could find and put the relevant clip over the right face.

You'd have this stuff available within meerk40t at all times, just like it always keeps an intact copy of your original image. Then it would do the cropping and virtual scaling of the image itself, and cropping that. Really you'd probably want to crop the image to the clip-box first and then apply the relevant scripts etc, including not just the original transparency masking of the image but also the extra cropping to the clipbox and the clipshape itself. (You'll never need to clip less than the axis-aligned bounding box of the shape you'll be using as a mask there).

This went from needing a pretty big redesign to get working, to needing a bit of work to get working. The parser already tends to parse the `clipPath` though it doesn't really do anything with that data since there was nothing in MK that could handle it. But, with the ImageNode class being much more able be coded. Raster-script and clip-path would just be two clear aspects of that class. Though notably only clip-path would make sense to save in the in the SVG. I guess you could save the operations of raster-script too.

Also, while not really part of this topic, there's also a need for a more strictly enforced passthrough option. Where if you load an image you should be able to strongly enforce that this image should not be manipulated. You set the DPI (version 8 allows DPI set, it provides variable step-distance to do this) and that should strictly enforce the size the image is allowed to be.

This was done a bit with post-script applied rasterscripts in 0.7.x but there's some growing use and of `imag-r` and we need to be able to give folks a 1:1 from the initial setup.  Since if you lock the pixels in place where it will not create new pixels, then dots are held constant and the DPI sets the image size more directly. Since size will change the dots per inch by increasing the number of inches without changing the dots. Currently you set the DPI and you set the size. And it determines if it needs to resample the image to make those two things fit. But, it's hard to match these currently so that we can be certain there will be no post-processing done.

jpirnay, 2022-07-30:
The clippath functionality in SVG is wider though, acting as a viewport for any object. So this would and could act as one but relevant usecase

tatarize, 2022-07-31:
True. `clipPath` in svg is supposed to be an overall viewport object that affects the rendering but not the geometry of underlying objects. So in theory you can clippath a path, though it would do this only visually. To do this more properly we'd need to treat the clipPath more like a group, which is somewhat how it ends up in the hierarchy though there are still some additional differences since they can attach at any node within the SVG structure or whatever. But, then the hack gets a bit weirder, because we're not quite dealing with a tree then. We have clipPath objects that attach at particular locations in the tree and they contain their own path objects (which again can also contain clipPath), and all of this stuff affects the render tree though not really the geometry tree. But, for the basic use-case of just clipPath attached to an image we could just add clipPath onto the ImageNode objects. Otherwise we have to treat the elements tree like the render tree and provide viewports within that that cause different rendering. So you could have a clipPath parent to a bunch of objects and those objects would engrave the same but if you tried to render them they would need to render according to their parent objects and whether there's a clipPath in that parent.

We could also I guess reconfigure this stuff a bit, And maybe pass the clipPath down to the child nodes so if you put any elem-node in a raster object it would have attached to it a clip-path which would affect the rastering of that object (though not the geometry, but you could actually force the clipping of the path if you used the cag tools) some people might want the geometry of the object to be affected. But, this then gets a bit more confusing.

We could maybe also make clipPaths a special type of group node and have them actually show up in the element tree heirarchy. So if we rendered the clipPath we would have a reference to the clipPath and either references to the children or just know that this is a renderable object alone and whatever children are within it are actually caught up in the reference there. So we could have a clipPath with an image child. And if we put the clipPath in the raster object it would render the children of that object regardless whehter those children were image or path-type nodes.

We could also, potentially, not allow after the fact editing and just force actualize the clipPath. All the path-like objects would be converted to actual paths and force clipped with the cag tools. All the images would be actually edited and rendered actual clip paths to those objects. This seems the least desirable solution.

I think the happiest medium might be to just allow clip paths on images. Unless there's a sort of clip-path objects occur as objects just below the root and contain references to points in the tree and thus are like images themselves that makes some kind of considerable sense that I'm missing. The full SVG functionality would basically allow clipPaths to all objects and affects the render tree. We don't really have a render tree, though I guess we could maybe make one. But, then you'd have path objects within that tree that could be referenced by the operations so they could be cut but not seen.

Make no mistake. I have seen some svg objects where without clip-paths the paths look wrong since they used a clip-path to cut off some circles to make a hat sort of thing, and it looked really dumb without the clip-paths and my proposed clip ImageNode addition would not fix that. Though again we wouldn't really fix that since the clipPaths are viewbox rendertree affecting attributes (which reference objects in the shadowtree) so we'd still cut the circles on that hat thing outside of the lines. 

I can see an easy solution to the just image problem. We let images have a clip-path and let you set that with tools, and do the typesetting stuff in the GUI. So you could adjust the path or the image and it'll just fix that in the cutplanning stage. But, I don't see how we could have an arbitrary node that contains somewhat virtual copies and is supposed to render but may or may not need to cut actual geometry. 

---

There was a old idea of have modification node things were we could have a semi-rendered stage. Where we could put the path objects in a raster-operation really inside a "render" object and put that inside an image operation. Then we wouldn't need the raster-operation at all. We'd have image operations which would contain `image` objects (with all the relevant info on how to raster them) and `render` objects (with all the relevant info on how to raster them). And in cut planning we'd merely do the rendering of the render objects. This way we wouldn't need two different raster-like operations. Since in the end we're just rendering the paths to an image and then putting that basically an image operation that can only contain one image. This would in theory be more versatile. We could, in such a system also have clip objects which could be applied to `image` or `render` since during the cutplanning render objects would become image objects. So you have clippath with an image or you could have clippath with a render with an bunch of paths. Then you process these modification node types from the leaf nodes backwards. You might be able to do some of these others operations too, after all cut is really just an engrave that is flagged to permit inner-first cutting. And a hatch is really just a path operation [... cut ...]
I think images get clips isn't a bridge too far, and svg's wider function of viewports for any objects (without a really brilliant implementation idea) might be. I often have some far fetched ideas like make images that show that they are composed out of renderable objects, and then clipped images give us this other functionality of viewboxes for free. Though other people try to use clip-boxes to clip geometry which isn't actually a thing they do in SVG... and the ideas just sort of get muddled.

closed 2024-10-04 by jpirnay