meerk40t/meerk40t #1221: [Bug]:

lagnajeet, 2022-08-04:
### Summary Description

[image]

When trying to burn SVG, Meerk40t is very inaccurate in the cut. See picture above. Not sure if this is a known issue or there is a workaround for it. 

The following file was used for the cut
[image]

### MeerK40t Version

0.8.0-Beta7

### MeerK40t Type

Executable

### Your Operating System

Windows

joerlane, 2022-08-04:
This appears to be a difference in speed setting, based on the irregular outlines. 

What was the Vector Cut speed set to in K40whisperer? What was the red Operation Cut speed set to in MeerK40t? 

Did all of those examples cut through the material? Did they appear to run at the same speed, and cut at the same power setting?

If the speeds appear the same; could you use the Save As function in MeerK40t to save the SVG? That newly saved file would include the MeerK40t operation settings to allow for further comparison.

lagnajeet, 2022-08-04:
They both were cut through the wood. Both were running at 5mm/s and same laser power settings. From the look of it they both look like 5mm/s . I have attached the file saved as SVG from Meerk40t

[image]
 
Looking at the file in a text editor seems like Meerk40t has approximated the arch as line segments which it has to do and when i open the SVG in any vector imaging program it look s fine.

joerlane, 2022-08-04:
This appears to be a bug. I was able to reproduce it with your file, and my own files that contain curves. 

Curves do work fine in the stable version 0.7.7 Post 1, version 0.8.0 is still in Beta after all...

lagnajeet, 2022-08-05:
I can confirm. I tried it with version 0.6 and it works fine.

jpirnay, 2022-08-06:
That should be resolved with the latest patches in main

tatarize, 2022-08-07:
The issue was it was turning on the constant_move_x value which is specifically designed to trigger a plotplanner thing called smooth which is expected to keep things from going from pure y to pure x movements because when rastering that would cause clanks. It's supposed to only turn on for some rare forms of rasters but engaged by accident when loaded from a file because `"False"` the string is actually `True` the value. And this attribute wasn't validated after reloading. This has been changed into a protected-non-saved value. I'll do the same for `raster_alt` and `force_twitchless` too which also will do the same thing but won't have a pronounced effect except your ability to turn on twitches won't happen and it'll do NSE-rastering regardless whether you turned that on or not.

closed 2022-08-31 by tatarize