meerk40t/meerk40t #1224: [0.8.0Beta7] Issues

jpirnay, 2022-08-05:
The latest BETA release of 0.8.0 is Beta 7 (https://github.com/meerk40t/meerk40t/releases/tag/0.8.0015). This issue is for reporting issues with this specific version of MeerK40t.

Please do all further testing on this release. Please do not report any further bugs with previous beta releases. Do give any other types of feedback, questions, concerns or comments.

jpirnay, 2022-08-05:
- [ ]  Config - window is wider than the default size and has a horizontal scrollbar.
- [ ] Tree Context menu - No right click entry on images for Raster Wizard (only apply script). In fact, I can't find the raster wizard at all.
- [ ]  Selecting in the scene should give some indication in the tree what has been selected if the parent nodes are closed - perhaps a squiggly blue underscore?
- [ ]  Should selecting in the scene "select" elements in the tree rather than (as present) "emphasising" them?
- [ ]  Selecting in the scene all elements in a group should (IMO) emphasise the group also (or possibly select the group and emphasise the children).
- [ ]  When you select preference 'Classify in reverse', overlapping filled paths are combined in the wrong order in simulate (and presumably burn). You need to check this setting and reverse the order when you create the raster image.
- [ ] 'Wavy line' bug (see https://github.com/meerk40t/meerk40t/issues/1221)
-

tatarize, 2022-08-09:
```Tree Context menu - No right click entry on images for Raster Wizard (only apply script). In fact, I can't find the raster wizard at all.```

It's located in the main scene and individual bits are editable when you call up the image properties.

Sophist-UK, 2022-08-29:
Suggest this is closed as beta 7 is ancient history.

closed 2022-08-30 by jpirnay