meerk40t/meerk40t #1228: Svg import scale

RyanOtero, 2022-08-06:
Super impressed with the feature set, but I've tried loading multiple svgs from adobe illustrator at 72 dpi with two different versions of meetk40t and the svgs import to fill the cutting bed size, is this default behavior? I wouldn't think so. It seems like a pita to have to rescale each vector manually. Versions tried 8.0014 b6 and 0.7.5000. driver is lyhstudio, board m2.

joerlane, 2022-08-06:
Inside of Preferences there's an option "SVG Pixel Per Inch". One of the options is 72 px per inch for Illustrator.

RyanOtero, 2022-08-06:
Yes, that is the one selected

joerlane, 2022-08-06:
Sorry I misread that. Is the Units set correctly (Inches or mm; or whatever)? Can you post a file? I don't have illustrator handy here.

RyanOtero, 2022-08-06:
[image]
1 inch square

I thought the units in the preferences were just for visualization, do they affect import scale? I work in inches in Illustrator, but generally in mm in cutting software because feed rates are easier to remember in even increments, e.g. 20/s, 450mm/s, etc. as opposed to decimal equivalents in inches. So it seems that the svgs come in roughly double native size, but it is not exact, I checked with a 1/2 in square and it was not half the size as the 1 in square on import, close though.

joerlane, 2022-08-06:
I get a square that's 72.5 pixels in inkscape. But you are right. It's filling the bed (even with the 96 px/inch setting) in MeerK40t.

I also can't seem to change my Units to Inches to even test. Passing this on; it appears to be a legitimate bug.

RyanOtero, 2022-08-06:
Yeah, I see in notepad++ it says 72.5 for the bounds must be the bounds of the stroke thickness? But it says 72 in the rect block. I was really hoping you were just going to tell me I missed a check box somewhere.

joerlane, 2022-08-06:
Ok; so one thing I was able to do. I saved your file in Inkscape and changed the document properties to mm. That opens correctly in MeerK40t. It doesn't seem to like a file defined with px as a measurement unit. 

Still a bug. It should support files with px as a unit.

RyanOtero, 2022-08-06:
Ok, will try. Thanks for the prompt feedback

joerlane, 2022-08-06:
Still not importing the correct scale. 

Looks like the feedback on the Unit setting in preferences isn't happening; but it did change me to inches. 

More bugs... None of the DPI settings are correct still, with either inches or mm set as a Unit.

joerlane, 2022-08-06:
I tried to change my inkscape import preference to be 72 px/inch. And now I'm locked out of document properties and preferences. 

That might actually work with those inkscape settings correct. But ideally you should be able to import Illustrator files into MeerK40t directly. 

I'll keep working on it; but someone will look at the underlying issues soon enough. 

Thanks for the bug report.

RyanOtero, 2022-08-06:
I tried opening a svg that was exported from AI and resaving with Inkscape, still did not work. A real shame for me, as I was really excited about using Meerk40t instead of k40 whisperer. I will be happy to provide whatever files you need for testing. Thanks, and I humbly await the squashing of this bug!

joerlane, 2022-08-06:
I just installed a fresh copy of Inkscape on another machine. It does the same thing. It's not doing the conversion properly. It opens the file and shows a 72 px square... But I can't make that convert or save correctly from Inkscape even. 

You could try setting a different output from illustrator. mm; should work.... 

Still don't have a handle on the scope of this bug. But I'll keep toying with it.

tatarize, 2022-08-07:
In preferences you should be able to set your SVG loading Pixels Per Inch to 72.

[image]

tiger12506, 2022-08-07:
Yeah, but the point of the issue is that it doesn't work

tatarize, 2022-08-07:
```svg
<?xml version="1.0" encoding="utf-8"?>
<!-- Generator: Adobe Illustrator 24.1.2, SVG Export Plug-In . SVG Version: 6.00 Build 0)  -->
<svg version="1.1" id="Layer_1" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" x="0px" y="0px"
	 viewBox="0 0 72.5 72.5" style="enable-background:new 0 0 72.5 72.5;" xml:space="preserve">
<style type="text/css">
	.st0{fill:none;stroke:#FF0000;stroke-width:0.518;stroke-miterlimit:10;}
</style>
<rect x="0.3" y="0.3" class="st0" width="72" height="72"/>
</svg>
```

There is no height or width. Therefore the height and width are 100%. `0 0 72 72` defines the viewbox. We should do then is scale this up to the size of the viewport (note not viewbox) that means we scale that up to the bed size which is what we get.

[image]

tatarize, 2022-08-07:
If you load this in Chrome you'll see it doesn't go to being 1in 1in, it goes to being the size of the window.

tatarize, 2022-08-07:
There should be something else in illustrator to correct this. The 1in 1in only works if the viewport is by definition the size of the viewbox which isn't strictly true. Our viewport is the size of the bed. We're discussing how to correct this in discord.

RyanOtero, 2022-08-07:
Hmm, does this mean I should always set my canvas size to be my bed size? Currently the canvas size is always set to be the vector bounding box.

tatarize, 2022-08-07:
The issue is the vector bounding box is causing it to become weird. It's actually within spec, it's supposed to set the height and width to 100% 100%. If you set it to the bed size you'd probably have no issues.

tatarize, 2022-08-07:
The issue is that because your viewport is 0 0 72.5 72.5 and your rect is 72 72. That doesn't mean that your rect is 1 inch if PPI is actually set to 72. Actually it means we scale up your design to be the size of the bed. Because we're giving width and height omitted and thus 100% 100%, which is taken to mean the bed size.

Even if we correct this and allow you a checkbox (which we're debating now) to turn off the assumption that your viewport is the size of the bed, the alternative is that the viewport is the size of the viewbox. Which means we have it at 72.5 72.5 which means our 72px 72px rectangle actually gets scaled up by a factor of 72.5/72 which would *still* be slightly wrong.

tatarize, 2022-08-07:
So your rectangle because of the SVG spec is supposed to be either the size of the viewport. This is because we have a viewbox but height and width are omitted. This means we 100% x 100% which is percent size of the viewport. The viewport in meerk40t is the bed size. So the 72.5 72.5 viewbox is scaled up to be the size of our bed. This makes it about 1% smaller than than the entire bedsize. If we permit an option to say viewport is actually the size of the viewbox we then get our 72.5 72.5 viewport with a which shouldn't scale up, but if it did we'd have it scaled by 1.00694... and the code is saying it's 1.01 inch which seems weird since it I checked the parsing and it didn't change. There's some chance the scaling factor given is off by that amount but that doesn't seem very likely but is certainly possible.

I'll need a few more checks. But, if you don't set the size of your design to the bounding box that would give you actual units for a standard size and should correct all your current problems. I need to find that 1% but it seems like this option I'm cooking up should also fix it, but still 1% isn't nothing so I'll need to figure that out.

RyanOtero, 2022-08-07:
Well I am fairly sure the canvas size is the strange 72.5 because of the stroke width. Normal workflow for me is to set the canvas size to artwork bounds just before saving as svg because k40 whisperer uses that as 0,0 top-right corner of the svg. Am I the only one using AI and trying to use meerk40t?

tatarize, 2022-08-07:
Okay, I found the 1%. MeerK40t is actually checking the bounds with the stroke width (need to if we raster things). All is well, though that might be potentially confusing but. You have a solid work around and we have a checkbox that fixes that loading so it's not an issue.

tatarize, 2022-08-07:
Yeah, I figured out the issue and it was *also* stroke for me too. Actually same thing since the 1% 72.5/72 scale stroke thing was also applied in meerk40t. So there's now an option for `SVG Viewport is Bed` which will correct the issue for your original file. Since what the viewport is was a thing that could be different.

tatarize, 2022-08-07:
Did you need anything else? Kick off a version to add the option in 0.8.x, add the option to 0.7.x line? Something else? Is this mostly settled? Does the work around to set the Illustrator canvas size directly work and you're good for now?

RyanOtero, 2022-08-07:
If you want to just add it in to whatever your next update is. Let me know and I'll grab that version when available. The bed size = canvas in AI is a good workaround for now. Thanks so much.

tatarize, 2022-08-07:
It'll be in the next version of 0.8.x which will be whatever beta or full release is next (we're pretty close to a full release).

tiger12506, 2022-08-09:
I still think the true fix for OP here is to actually get Illustrator to output file with a meaningful size, if OP wants meaningful size interpretation in other programs. I've not used Illustrator, but this page seems to show how to do it.
https://webcoursesbangkok.com/how-to-change-measurements-in-adobe-illustrator/

RyanOtero, 2022-08-09:
This article is about changing units of measurement within the program itself for various items (artboard, stroke and font size) which I already use. It does not describe units for export. And as it happens there are no unit settings on the svg export dialogue besides dpi which is 72 dpi by default.

tiger12506, 2022-08-09:
Hmm.. I guess Illustrator is dumber than I originally thought. I figured it would at least be equivalent to Inkscape. Shows what I know. Sorry for the noise.

closed 2024-10-04 by jpirnay