meerk40t/meerk40t #1279: [Bug]: Error when submitting work to laser engraver.

RossMich1, 2022-08-24:
### Summary Description

Hi There,

When I execute the work for engraving I get this error:

[image]

I'm trying to engrave this onto a piece of cardboard:

[image]

I've followed this [YouTuber ](https://www.youtube.com/watch?v=ynTFqAd5HR0&ab_channel=UKRocketry)and I've followed his steps exactly and keep getting the same error. I'm still new to this and this is probably me being inexperienced with laser cutters but would appreciate some help.

### Additional Details

Laser cutter K40 with Moshi Board.

### MeerK40t Version

v0.8.00.19

### MeerK40t Type

Source (zip/tar file from Github)

### Your Operating System

Windows

tatarize, 2022-08-24:
It's probably a crash bug. It's very close to done, but I've not broken out my moshiboard for that kind of testing yet so it probably went unseen.

tatarize, 2022-08-24:
https://github.com/meerk40t/meerk40t/releases/tag/0.8.0020

Here pushed a new version. Should have files soon. Includes several small Moshi fixes that prevented it from working. If there's any more trouble you should probably switch to the stable version of 0.7.7

closed 2024-10-04 by jpirnay