meerk40t/meerk40t #1283: [Bug]: text dominant-baseline not accounted for

will-janz, 2022-08-27:
### Summary Description

The dominant-baseline attribute for text isn't handled correctly. In this example, dominant-baseline is set to middle but the default is used.

I checked the [SVG Namespace](https://github.com/meerk40t/meerk40t/wiki/Tech:-SVG-namespace) wiki entry but it asked me to fill it out and had nothing in it.

### Additional Details

[image]
[image]

### MeerK40t Version

v0.8.0020 Beta12

### MeerK40t Type

Executable

### Your Operating System

Windows

tatarize, 2022-08-27:
The SVG Namespace thing would be for the meerk40t specific sections of the svg we use convering notes and operations and custom elements.

--

This appears to be correct, though wxPython doesn't provide easy hooks for that. Supporting some of the cryptic parts of the SVG text standard might be harder. You can do text-to-path in inkscape if you need to the text to work consistently there. The dominent-baseline for middle and hanging would in theory be possible though it would tend to just require moving the text object down some. There's some oddities in the text that aren't supported like rotation where each character gets rotated from the previous character or some of the left-to-right or vertical text. This may eventually get added but I think it might just require biting the bullet here and permitting the defect to continue. Unless a bunch of people are using that feature and it's fairly easy to fix. It's *mostly* right now, but still has a few defects and a critical bug. Apparently linux has a habit of changing the text size to an integer text size which can cause it to be larger than expected and lose a bit of the left side of the text. But, the failure to implement a few things might not make the immediate list of things to fix. Especially since wxPython doesn't really give clear insight as to where they baseline should be. In theory `baseline-shift` and `dominant-baseline` might eventually get added but I don't think its critical just now.

But, do text-to-path in inkscape and they'll be right where inkscape showed them to be.

tatarize, 2022-08-27:
I updated the link given. It should have pointed to: https://github.com/meerk40t/meerk40t/wiki/Namespace

will-janz, 2022-08-27:
I'm typing these out by hand and generating them with a script (because weird brain). I'll have to see about automating text to path. No worries, gotta prioritize. Mind if I make an "SVG Pitfalls and Warnings" page in the wiki?

will-janz, 2022-08-27:
For anyone from the future who wants to do this in the command line: 

`inkscape --without-gui --file="SOURCEFILE" --export-text-to-path --export-plain-svg --export-filename="SOURCEFILE-pathed.svg"`

With Inkscape 1.1

tatarize, 2022-09-16:
While not perfect, I actually did some work adjusting the baselines since wxPython natively had hanging since all text was drawn from the upper-left corner so I had to move the baseline with the best adjustments I could muster and so I also added `middle` for the baseline there. It didn't make the element sign perfect, but it did make it markedly improved.

will-janz, 2022-09-23:
Dang, nice! I think that puts it ahead of Lightburn in terms of text handling (I'm not 100% sure of that though.)

tatarize, 2022-09-26:
Nah, there's a pretty gapping hole in that I don't actually solve the text for the shape of the text. And I sort of need to be able to do that. So you can't really vector cut text or convert text-to-path within meerk40t.

closed 2024-10-04 by jpirnay