meerk40t/meerk40t #1285: [0.8.0Beta12] Issues

Sophist-UK, 2022-08-29:
The latest BETA release of 0.8.0 is Beta 12. This issue is for reporting issues with this specific version of MeerK40t.

Please do all further testing on this release. Please do not report any further bugs with previous beta releases. Do give any other types of feedback, questions, concerns or comments.

Sophist-UK, 2022-08-29:
* [ ] Operations Properties windows take a long time to open
* [ ] Image operation properties - Estimated burn time - Calculation button doesn't calculate
* [ ] Preferences window default window size is taller than it needs to be
* [ ] Ctrl-mouse wheel doesn't zoom the scene.
* [ ] Camera window - either resize window to fit camera image or resize camera image to fit within the window
* [ ] Camera window - I have no idea what the slider does. Apart from going from 0-60 (in 4.7s? {Jeremy Clarkson}) it doesn't seem to have any effect. Aside from making it work, can we please have a caption title and a tooltip explanation?
* [ ] Camera window - the Correct Perspective checkbox seems to be counter-intuitive at first (i.e. you only get handles when you disable it) - we need a tooltip to explain how to use this.
* [ ] Camera window - the Correct Fisheye checkbox doesn't appear to do anything - and it could do with a tooltip explanation too.
* [ ] Camera window - I am not sure whether it makes sense to allow the handles to be dragged outside the picture.
* [ ] Camera window - the Reset Perspective / Reset Fisheye menu items should be reversed in position to be in the same order as the checkboxes.
* [ ] Camera window - the Detect Distortions / Calibrations needs a tooltip to explain how to get the checkerboard pattern and how to use it.
* [ ] Tools/Camera Interface vs. Camera button on toolbar - they store separate window positions/sizes - is this intentional?
* [ ] View / Show variables - what does this do?
* [ ] Camera - How do you delete the background image from the scene once you have set it? Also might be useful to have a toggle check-box which shows / hides the camera image. This appears to exist for Simulation, so it may well be the next issue that is the cause.
* [ ] Preferences/GUI/Use in scene tool menu - when checked, right click on background toggles a drawing menu rather than giving a normal right click menu - this does NOT feel right - we need a real right click menu for background (e.g. to show/hide the camera) and toggling an in-scene drawing menu should be an option on that. (I think that it is actually this option that prevents the right click menu from Simulation from also being shown on main scene.)
* [ ] Camera - Is it possible (and does it make any sense) to have the live camera image as background of the scene?
* [ ] Scene bounding box - seems to fit correctly on selected items horizontally (i.e. vertical lines match extremes of selected object), but vertical boundaries (horizontal lines) are incorrect (too close together).
    [image]
* [ ] Tree - Inkscape labels are not shown in tree though they are shown in properties. But they are shown on older files, so Inkscape may have changed the way it saves them.
* [ ] Raster Wizard - Despite being an experienced Meerk40t user, I have no idea how to do image Raster Wizard. There is no Raster Wizard toolbar button, no Tools menu entry, no right click options that make any sense. Until I find out how to do this, I cannot really use MK 0.8 in production.
* [ ] Tree labels - I actually think the shorter labels in 0.8 are better than the length labels I did for 0.7, but there is probably more information that would in fact be useful - like path fills.
* [ ] Keymap - defined keys don't seem to work!!
* [ ] Help/Discord - Please create a new invite link as this still has my Discord nick showing.
* [ ] Scene background - should probably be different for each device. Each device should probably also have a default camera associated.
* [ ] Notes pane should not show "Automatically open notes" when pane is docked.
* [ ] Object sizes appear to be slightly wrong. For example, when I create a path that is **exactly** 230mm in width with a stroke-width of 0.2, it loads into MK with a width of 230.09mm. (See above image as example.) Similarly I position this at **exactly** x=0.9mm (including stroke width i.e. at **exactly** x=1mm) and it loads as x=0.96mm. Again, for me, this makes MK 0.8 unusable in production. (EDIT: I need to look into this further, as older SVGs I have saved appear to load accurately. So may be an inkscape setting or bug. However loading the same file in 0.7.7 and 0.8b12 gives different sizes and positions!!)
* [ ] Tree labels - File nodes are shown as "File:" with a colon, others are shown as e.g. "Path" or "Group" or "Image" without a colon. Personally I would prefer to see everything with colons, but they should IMO be consistent.
* [ ] Preference/Classify in reverse - When checked, raster operations process images in wrong order so elements that should be on top are hidden. (This has been reported before but not yet fixed.)
* [ ] Properties window not showing element properties.
* [ ] Operations nodes in tree showing "@default" or "default" when the default checkbox in op-props is not set. What do these means and what is the significance of the "@"?
* [ ] Dots in SVG do not create a Dots op.

Here is the test file I use to test classification and raster optimisation etc.
[Element classification test.zip](https://github.com/meerk40t/meerk40t/files/9444051/Element.classification.test.zip)

tatarize, 2022-08-30:
>  Operations Properties windows take a long time to open

Fine for me. Might depend on the property settings. My system is 12 years old.

> Image operation properties - Estimated burn time - Calculation button doesn't calculate

Fixed. Now they are better than the other estimates!

> Preferences window default window size is taller than it needs to be

It's too short to show the calculations. And it gets different sized preference windows in it.

> Ctrl-mouse wheel doesn't zoom the scene.

Alt + Right-mouse zooms. I'll add ctrl-middle.

> Camera window - either resize window to fit camera image or resize camera image to fit within the window

Ok. Done.

> * Camera window - I have no idea what the slider does. Apart from going from 0-60 (in 4.7s? {Jeremy Clarkson}) it doesn't seem to have any effect. Aside from making it work, can we please have a caption title and a tooltip explanation?

Ok. Done.

> * Camera window - the Correct Perspective checkbox seems to be counter-intuitive at first (i.e. you only get handles when you disable it) - we need a tooltip to explain how to use this.

Solidish maybe. But, when you're correcting the perspective everything is weird which means you can't because it made whatever wonky quadrilaterals you put into there into a square. 

> *  Camera window - the Correct Fisheye checkbox doesn't appear to do anything - and it could do with a tooltip explanation too.

Ok. Done. 

> *  Camera window - I am not sure whether it makes sense to allow the handles to be dragged outside the picture.

It does. There's a few rare corner cases where your camera isn't going to capture the other corner but you still want a perspective.

> *  Camera window - the Reset Perspective / Reset Fisheye menu items should be reversed in position to be in the same order as the checkboxes.

Ok. Done. Good eye!

> *  Camera window - the Detect Distortions / Calibrations needs a tooltip to explain how to get the checkerboard pattern and how to use it.

Ok. Done.
Really this whole setup needs a camera wizard thing to go through the process. 

> *  Tools/Camera Interface vs. Camera button on toolbar - they store separate window positions/sizes - is this intentional?

Yes. You might want your laser-plasma camera smaller than your bed camera.

> * View / Show variables - what does this do?

Text Variables in wordlist, and whether they should be shown with their given value or shown with the pre-replaced values.

> * Camera - How do you delete the background image from the scene once you have set it? Also might be useful to have a toggle check-box which shows / hides the camera image. This appears to exist for Simulation, so it may well be the next issue that is the cause.

Double-click brings up a special menu. Remove Image.

> Preferences/GUI/Use in scene tool menu - when checked, right click on background toggles a drawing menu rather than giving a normal right click menu - this does NOT feel right - we need a real right [...]

Yeah, this stuff is why the secret menu for that is under double-click. There certainly does need to be a better solution. But, it's not likely for 0.8.x

> *  Camera - Is it possible (and does it make any sense) to have the live camera image as background of the scene?

You can set a timer to basically do that if you want. 

`timer 0 0.2 .camera0 background`
It will give you 5 frames a second. You can set the 0.2 lower for more.

If you close the window with the camera you might need to do `camera0 start` to get the camera running again. (it doesn't really need the gui but closes if all know things using the camera are closed).

[... 30 lines cut ...]

Ok. Done.
Your colon should now be regular. 

> *  Preference/Classify in reverse - When checked, raster operations process images in wrong order so elements that should be on top are hidden. (This has been reported before but not yet fixed.)

We're looking into this. But, it might be one of those things where we made moving things around a lot easier for the classification stuff.

> * Properties window not showing element properties.

They aren't showing references properties. That's a reference node, not a element node.

> *  Operations nodes in tree showing "@default" or "default" when the default checkbox in op-props is not set. What do these means and what is the significance of the "@"?

The `@<default>` is speed I think. This basically says that no speed is set for that operation and the default is being used. This could be improved but if you see that you should probably set those values or get whatever the default value might be.

> *  Dots in SVG do not create a Dots op.

This is true. They do not. We don't detect them anymore and you'd need to use the Point tool to draw them. I think dxf which has real points might actually make them. This could be restored. But, generally it got lost when all the code for `elem` type nodes got subclassed and Point became a real thing. It's not gotten all the hacks back in it after I took all the hacks out.

> Here is the test file I use to test classification and raster optimisation etc. [Element classification test.zip](https://github.com/meerk40t/meerk40t/files/9444051/Element.classification.test.zip)

Jpirnay's been working on that. We might actually just force the user to do that manually.

There might be other ways to do this and we absolutely hate that the operation order and element ordering are causing radically different results.  You have white circle and black circle you could get donut but you could also get black circle covers everything. But, what's actually a problem is that you'd have donut in the scene and black circle on the work. (Unless you check simulation).

Sophist-UK, 2022-08-30:
> > Operations Properties windows take a long time to open
> 
> Fine for me. Might depend on the property settings. My system is 12 years old.

My system is also old but is an i7. Please believe me when I say that there is a graphics performance problem showing Property windows.

> Solidish maybe. But, when you're correcting the perspective everything is weird which means you can't because it made whatever wonky quadrilaterals you put into there into a square.

The issue is not a bug - and I think that how it works is correct - but usability when there is no explanation of how to adjust the perspective is pants.

> > * Camera window - the Detect Distortions / Calibrations needs a tooltip to explain how to get the checkerboard pattern and how to use it.
> 
> Ok. Done. Really this whole setup needs a camera wizard thing to go through the process.

Yes. A wizard would be great, but I am not using camera, just testing.

> > * Tools/Camera Interface vs. Camera button on toolbar - they store separate window positions/sizes - is this intentional?
> 
> Yes. You might want your laser-plasma camera smaller than your bed camera.

No - this is not the same thing. Your explanation would be saving different sizes / positions for different cameras (which may also be true) but my bug report is that windows should have the same position whether they are opened with Tools/something or a toolbar button. The question about whether undocked-panes & windows should share the same size and position is also a different question.

> > * View / Show variables - what does this do?
> 
> Text Variables in wordlist, and whether they should be shown with their given value or shown with the pre-replaced values.

So what is the wordlist and what are text variables. This needs documentation / tooltip explanations.

> Double-click brings up a special menu. Remove Image.

Then swap the usage - make double-click bring up the drawing menu and right click the normal context menu.

> Yeah, this stuff is why the secret menu for that is under double-click. There certainly does need to be a better solution. But, it's not likely for 0.8.x

Then swap the usage - make double-click bring up the drawing menu and right click the normal context menu. I cannot see why this needs to be delayed for 0.9.x?

> > * Camera - Is it possible (and does it make any sense) to have the live camera image as background of the scene?
> 
> You can set a timer to basically do that if you want.
> 
> `timer 0 0.2 .camera0 background` It will give you 5 frames a second. You can set the 0.2 lower for more.

Ok - then lets have some GUI functionality to do this as IMO (as someone without a camera) this is something that is likely needed frequently.

> If you close the window with the camera you might need to do `camera0 start` to get the camera running again. (it doesn't really need the gui but closes if all know things using the camera are closed).

I am sure that you can do a camera autostart whenever access is needed and have an automatic timer turn it off (say) 2-5 secs after it was last used.

> > * Scene bounding box - seems to fit correctly on selected items horizontally (i.e. vertical lines match extremes of selected object), but vertical boundaries (horizontal lines) are incorrect (too close together).
> >   [image]
> 
> That seems accurate.

No - it is broken.

> You would apply the script. Then manipulate properties in the properties window, if that was needed. You can also tweak the other more important properties all the time and actualization during the [...]

Can I have an explanation in English please. There appears to be nowhere to apply a script (e.g. Gold). The properties window has DPI.

> They are all super configurable now, and can be overwritten by different services if that's required. So you can get, for example, frequency for balor when you don't care for an M2 Nano.

I can't find any configuration screens to change what is shown.

> The cameras are all independent. They are services and you can launch any number of cameras (though scripted we do five). Scene background might need to clear out if you switch devices though.

As I have previously suggested, the UI needs to reflect the workflow the user is doing at the time. Design (drawing), Layout (moving stuff around), Operations settings (i.e. simulation but in main scene with access to ops settings), Pre-Burn (where you want to position the material and have live camera view), During-Burn - also may want live camera and spool info.

The last 3 should be related to a selected device, and camera should be associated with that device.

> The stroke width is half on each side like wxPython draws. There might be a need for actual exact values without stroke-widths in the calculations. They are however accurate values the sizes just [...]
[... 3 lines cut ...]
Please look into this as it sure seems like a bug to me.

> We're looking into this. But, it might be one of those things where we made moving things around a lot easier for the classification stuff.

I fixed this in 0.7 but that fix was part of the commits that were not included in 0.8. When you convert raster elements to an image you need to check whether classify_reverse is set or not.

> > * Properties window not showing element properties.
> 
> They aren't showing references properties. That's a reference node, not a element node.

Is there any reason why the related element node properties should not be shown?

[... cut ...]
> This is true. They do not. We don't detect them anymore and you'd need to use the Point tool to draw them. I think dxf which has real points might actually make them. This could be restored. But, [...]

So how does 0.8 burn a point then?

> Jpirnay's been working on that. We might actually just force the user to do that manually.
>
> There might be other ways to do this and we absolutely hate that the operation order and element ordering [...]

See fix description above - it is a couple of lines of code to reverse the direction of iteration when you change the raster elements into an image.

tatarize, 2022-08-31:
> My system is also old but is an i7. Please believe me when I say that there is a graphics performance problem showing Property windows.

I guess I can test it on my other system without a graphics card. But, I didn't even have performance issues on the rPi here.

> So what is the wordlist and what are text variables. This needs documentation / tooltip explanations.

Word lists are on the ribbon bar and are activated when you try adding text. So you could do names on pages of pencils. And move on to the next page next time etc. We have it working well enough to test but it could still use the edges smoothed out. 

They could use some better interfacing and work but we'll deal with this after the next version. Basically they are list replacement text. 

> Then swap the usage - make double-click bring up the drawing menu and right click the normal context menu.

Double click on a real object would bring up the object properties and that's fairly standard. There likely needs to be a better way to remove the background.

> Then swap the usage - make double-click bring up the drawing menu and right click the normal context menu. I cannot see why this needs to be delayed for 0.9.x?

I could see including remove background image within the right-click context menu, though currently there's nothing there. 

The current plan is after 0.8.x to move along that series and work on design issues for 1.0.0. 0.9.x would be more if we needed massive changes to the kernel and we honestly shouldn't need any more kernel changes. There's hasn't been anything major needed changed in the kernel for a while, and services seems like a really solid way of doing things.

> Ok - then lets have some GUI functionality to do this as IMO (as someone without a camera) this is something that is likely needed frequently.

There is some need to write a frames server that you could use to get frames from the machine. Like if you had a rasperry pi controlling the camera and you still wanted access. Double-clicking the image in the camera frame will update the background. I'm not sure of the utility of live-mode but it could be done with a timer. You could even do `batch add ...` and it would do invoke that timer when you start MK.

> I am sure that you can do a camera autostart whenever access is needed and have an automatic timer turn it off (say) 2-5 secs after it was last used.

Well you can certainly start the camera and you might even be able to do `camera start background` and it might try starting if not started. I'd need to check tho.

> > That seems accurate.
> 
> No - it is broken.
Yes, that's accurate. It's broken. Though I might need a copy of that file, or fairly a subset to track down why.

> Can I have an explanation in English please. There appears to be nowhere to apply a script (e.g. Gold). The properties window has DPI.

This should be made more easy in beta13. We tried to make it clear. You should be able to load an image, select it, right click and rasterwizard that thing in a somewhat understandable way.

> I can't find any configuration screens to change what is shown.

Oh, highly configurable for the developers, yeah, you can't control that. Seems like it would be some rather extreme feature creep.

> As I have previously suggested, the UI needs to reflect the workflow the user is doing at the time. Design (drawing), Layout (moving stuff around), Operations settings (i.e. simulation but in main [...]

This has come up a few times recently. It's certainly not a thing that we could tuck in at the 11th hour for 0.8.x

> Please look into this as it sure seems like a bug to me.

I'll make a note to let you turn off stroke widths included in selection size. They might be needed for rendering an image but they aren't really needed in the bounding box stuff, and hard selection of the geometry would certainly be helpful to figuring if what you're seeing a bug or not.

> I fixed this in 0.7 but that fix was part of the commits that were not included in 0.8. When you convert raster elements to an image you need to check whether classify_reverse is set or not.

Yeah, there were a lot of complex sorts of just-so code in the advanced classification routines. Some parts of that were impossible to fix when the switch was made to having each type of element be its own node. Basically all the nodes could maybe include an svg.Rect() but they are migrating towards not including such a things. Text and Image already lack the SVGText and SVGImage object backing them. Some of the shapes still have them but mostly don't really need to. They are solidly positive changes but the readability of expert classification made porting that over too hard. The same sort of thing befell the optimize split rasters things. And focus was made towards making manual classification of elements significantly easier. 

> Is there any reason why the related element node properties should not be shown?

There was not and jpirnay fixed that for beta13.

> I'll try to recreate this and see if setting speed makes it disappear.

I have no doubt that it's accurate but basically the operations have settings and settings are now pure dictionaries. So you do not necessarily have a speed associated with an operation. It will insert the default speed but it doesn't necessarily have that already done until you mess with the settings so it uses "default" as a placeholder.

> So how does 0.8 burn a point then?

You go to the points tool and add some point in there and move them around if you need to.  It'll create a dot. SVG loading doesn't have anything for dots so they aren't being automatically converted. If you put them into the dots operation I think they anything that has a first point would burn that point.

Sophist-UK, 2022-08-31:
> > My system is also old but is an i7. Please believe me when I say that there is a graphics performance problem showing Property windows.
> 
> I guess I can test it on my other system without a graphics card. But, I didn't even have performance issues on the rPi here.

My system has a built in Intel GPU and a discrete Nvidia one - and you have to tell Windows to use the Nvidia one - when I get a chance I will see whether changing to Nvidia makes a difference. But I suspect that the difference of opinion is only one of expectation - I see the performance as well below what I would expect for this, you see it as acceptable. I would say that it seems to be drawing elements one by one - so it may just be that the elements need to be created before the window is shown.

> Word lists are on the ribbon bar and are activated when you try adding text. So you could do names on pages of pencils. And move on to the next page next time etc. We have it working well enough to [...]

So it's the equivalent of Microsoft Word's mail merge function - you can burn object 1 with name 1, then move to object 2 and burn that with name 2. Does this get saved in the SVG file?

1. I think that slightly different terminology might help make it understandable to newcomers.
2. Some simple explanation in tooltips would help.

> > Then swap the usage - make double-click bring up the drawing menu and right click the normal context menu.
> 
> Double click on a real object would bring up the object properties and that's fairly standard. There likely needs to be a better way to remove the background.
> 
> I could see including remove background image within the right-click context menu, though currently there's nothing there.

This is already the case in the Simulation scene. I suspect that the same right-click context menu would be shown in the main scene if it had not been overwritten by the code to make the drawing menu appear/disappear. (Toggling the drawing menu could presumably be allocated to a Keymap anyway.)
 
> The current plan is after 0.8.x to move along that series and work on design issues for 1.0.0. 0.9.x would be more if we needed massive changes to the kernel and we honestly shouldn't need any more [...]

This is NOT a major design change - this is a bad minor UI choice that requires a minor tweak to the UI and which should be included in 0.8.x.

However, I agree that major UI changes should be in a 0.9.x branch.

> There is some need to write a frames server that you could use to get frames from the machine. Like if you had a rasperry pi controlling the camera and you still wanted access. Double-clicking the [...]

Personally I have no issues with the concept of using the console to start a timer rather than having a UI for it, but many users are not comfortable with command lines and without UI functionality for it they will never discover that it is possible.

> This should be made more easy in beta13. We tried to make it clear. You should be able to load an image, select it, right click and rasterwizard that thing in a somewhat understandable way.

I was right clicking in the tree - and didn't try right clicking in the scene. I will take another look at b13 when I have the chance.

> > I can't find any configuration screens to change what is shown.
> 
> Oh, highly configurable for the developers, yeah, you can't control that. Seems like it would be some rather extreme feature creep.

To be clear, I am not looking for user config for the labels - but I am looking for labels to have more of the attributes that would actually be helpful. If it would help, I would be happy to look at the various types of node and suggest what might be useful to add to the labels.

> This has come up a few times recently. It's certainly not a thing that we could tuck in at the 11th hour for 0.8.x

No - as above, major UI changes should be for v0.9.

> I'll make a note to let you turn off stroke widths included in selection size. They might be needed for rendering an image but they aren't really needed in the bounding box stuff, and hard selection [...]

As I already said, this is **NOT** an issue with Stroke Widths. The stroke widths are a known size, and the variance is NOT this size. This is a significant BUG.
 
> Yeah, there were a lot of complex sorts of just-so code in the advanced classification routines. Some parts of that were impossible to fix when the switch was made to having each type of element be [...]

This is not the same issue as the optimise split rasters (which is much more complicated). This is a SIMPLE issue to fix.

However, the optimise split rasters should still be possible., You can either make default assumptions about acceleration . deceleration distances that make sense for all lasers, or you can code device routines that provide the metrics needed to make such assumptions. (I think the former would be the way to go.) I should add that having used 0.7 with raster split optimisation the I think that using a zero margin in the raster step direction was a mistake - it is slower to stop and move a small distance and restart and if e.g. head is at y=0 and you are doing B2T rasters, it will do (correctly) do the separated rasters from top to bottom but do each raster from bottom to top which looks weird and has longer travel distances. A more significant margin the step direction would be beneficial.

[... cut ...]
> You go to the points tool and add some point in there and move them around if you need to. It'll create a dot. SVG loading doesn't have anything for dots so they aren't being automatically [...]

I don't have any real files with DOTs that need to be burned - I think I have a couple for registrations, but they are not important - so not sure if there are any real life uses of dots in SVG files that need to have backward compatibility. So I am good with this as it is.

But my question was whether points are burned in 0.8 as dots were in 0.7 i.e. as a dwell.

Sophist-UK, 2022-09-08:
Closing because current beta is 15.

jpirnay, 2022-09-08:
Beta 15 has been temporarily withdrawn, so I'll reopen this as well temporarily

jpirnay, 2022-09-08:
[image]
I can't reproduce this behaviour - any chance you could share the culprit file?

jpirnay, 2022-09-08:
Sorry, wrong reopen

closed 2022-09-08 by jpirnay

jpirnay, 2022-09-09:
-  Camera - Is it possible (and does it make any sense) to have the live camera image as background of the scene?

Tat already explained the existing console method, starting with b17 you can as well issue such a thing from within the camera panel:
[image]
[image]