meerk40t/meerk40t #1293: [0.7.7] Semi-transparent white PNG pixels being burned but with laser off

Sophist-UK, 2022-08-31:
PNG raster lines containing all fully-transparent pixels or partially-transparent white pixels, are being burned with laser off. Resulting image is correct, but takes longer to burn.

I am not expecting this to be fixed in 0.7 - I am reporting it so that this can be corrected in 0.8 if it still exists. (My limited experience with 0.8 is that it didn't happen.)

tatarize, 2022-09-01:
I need more details and maybe an example of this. Fixes in 0.7.x are still permitted and will probably be done. They just have to be done with an extremely narrow scope that modifies as little code as possible to fix a very specific error. We actually have a few things in 0.7.x that need fixing. If the fix isn't difficult and will not create any additional bugs that's absolutely a thing that goes into maintenance mode.

Secondly, please actually recheck that the laser was *entirely* off. The only reason it would go to a place is to laser it in raster mode. Get the example bit you checked and see that it didn't actually fire a single laser pulse or two in that area. You might need to look at an angle or whatnot. But, the code doesn't really allow the behavior you're describing. If it's going to that location in a raster, it's going there to fire at least one pulse.

---

* Ruidacontrol RunStatus reply is now known.
* Corrections for wx4.1 / py3.10. (There's crashing if they run this combo from source)*
* Correctiions for wx4.2 / py 3.10. (This actually mostly works and a few windows need fixing)*

`*` - python 3.10 changed the binding rules to no longer silently permit floats to be cast to ints when going to C++. These need to be overt, wxPython is entirely made of those things so we have a special test case of `wx.Size(0.5)` and if that crashes we know the python and wxPython aren't compatible, the exception being wxPython 4.2 which mostly fixed everything internal but we'll need to cast a few values to int in MK source.

Sophist-UK, 2022-09-01:
I checked the png - they were semi transparent white pixels.

I have no idea whether the laser was firing because it was burning faintly om paper i.e. 100ppi and low power so beam was not visible.

The PNG file is the one on the 0.8beta13 bugs issue.

You can see the artefacts if you use the raster wizard on this PNG - you get black dashes around the edge.

closed 2024-10-04 by jpirnay