meerk40t/meerk40t #1300: Wordlist Editor Improvements

tiger12506, 2022-09-04:
To get the most out of a new feature so that Tat doesn't end up hiding it under a rug before release, we want to make a couple changes to the Wordlist window.

[image]

1) Encourage manual editing of the content of variables by allowing rows to be added/edited/deleted in the right grid control.
2) Make clear what order variable content will appear in by providing an index column in right grid control
3) Make clear what "Current Index for Data" does by perhaps renaming it to "Starting index" ?
4) Remove Index column from left grid control if it serves no purpose. (I can't get it to change value ever?)
5) Provide for a bit of flexibility in inputting data. We should be able to copy csv data from another application and paste it into this window (perhaps with button press? Or get fancy with drag and drop?)
6) Can we transition from using the textbox and Add buttons to directly adding/editing rows in the grid controls? I think that would improve readability for the user a lot.  Both the left and right grid could have independent + x buttons if you like for adding/deleting a row...
7) Backup and Restore wordlist should focus on saving and loading existing variable list and existing content rows (does it? I can't edit content, so I don't know)

tatarize, 2022-09-04:
Rug is easier. It'll hold all that code and take 5 minutes.

tatarize, 2022-09-04:
The index value actually does have a purpose it's the index in the word list you're at. It only really counts up if you check a wordlist that's like `{names++}` and you need to have more than 1 names in your list.

tiger12506, 2022-09-04:
Dumping this here so it doesn't get lost:
[image]
[image]

RuKeBo, 2022-09-28:
> Make Import CSV-File one action, we don't care about the filename once we've picked it, so Import CSV should do both ellipsis button's handler, and it's own, get rid of the other controls. Should be same "level" as the Add variable content button.

I'm not quite sure I understand this but I do know that when I browse for the CSV file and select it, it Imports it without me clicking on the "Import CSV" button.  This is not how it worked previously.  I think this is what you are saying in that it is one operation now but it sounds like you are saying the "Import CSV" should do both operations, not the ellipses button.  I will say that the way you had it sort of made more sense to me especially since you provided the preview.  What is the point of the Preview now if it's going to import it anyway?  It is helpful some though.  I assume your thought is that if you say change the file outside then want to Import it again you hit the "Import CSV" and it re-imports?  Just my thinking but this then by-passes the Preview since now you are Importing something different than the Preview shows.  If you had to load the file then Import it, that would seem to flow more logically, but then it is two steps.

RuKeBo, 2022-09-28:
Also, you can chalk this up to another time or not at all but it would be really nice if the preview could show the expected import scheme.  For instance, if the "Contains Variable-Names" is selected it would show the Variable Name in bold then the values under it in normal...or maybe indent the values under the Variable Name? I know this is a bit overboard.  It also gets more complicated when you have multiple Variables and values for them.  

[image]
  
If the "Contains Data" or the "Auto-Detect" detected data, then it might list just the data separated by a space so they were in groups.        

[image]  
  
Or Maybe in columns like so:  
  
[image]

[image]

closed 2022-10-24 by tatarize