meerk40t/meerk40t #1305: Can you please make it for home right and it down go toward left side ? after home button clicks

Irfan478, 2022-09-07:
### Summary Description

My machine home is in right direction and after it homes the machine the head move towards left side 

### MeerK40t Version

0.8.0022 Beta 14

### MeerK40t Type

Executable

### Your Operating System

Windows

Irfan478, 2022-09-07:
the reason behind this is that when my head goes towards the left side it doesn't come towards the right side after homing.

Sophist-UK, 2022-09-07:
Go into the Config window and there is an option for `Home right`.

tatarize, 2022-09-08:
Hm. I'll check. It could be the home routine messing up. It should go be completely able to treat your upper-left as machine home.

tatarize, 2022-09-08:
```
[20:08:27] channel open 19
[20:08:27]     Watching Channel: m2nano-home-right/usb_send
[20:08:31] channel open 20
[20:08:31]     Watching Channel: m2nano-home-right/recv
[20:08:34] home
[20:08:35]     [255, 206, 0, 0, 0, 1]
[20:08:35]     [255, 206, 0, 0, 0, 1]
b'\x00IPPFFFFFFFFFFFFFFFFFFFFFFFFFFF\xe4'
[20:08:35]     [255, 206, 0, 0, 0, 1]
b'\x00IBzzzzzzzzzzzzzzzzzzzzzzzzzzzz)'
[20:08:35]     [255, 206, 0, 0, 0, 1]
b'\x00zzzzzzzzzzzzzzzzzzz220S1PFFFFFO'
```

This is correct. The command IPP does the home event (`b'\x00` is header stuff and the trailing `F` are filler). Whereas the IBzzzz is actually moving to the left (x-swapped, typically B is right otherwise). I assume your machine is also swapped X axis in addition to being home-right. This is likely due to the home-offset being at 0,0 as "off" but that's not actually correct since if you homed in the top-left then you're actually at `(0, 330mm)` or so. And it shouldn't actually try to mess with that. 

The issue is that adjust_x and adjust_y previously were hard-coded so when they checked against 0,0 it was fairly okay.

```python
        adjust_x, adjust_y = self.service.physical_to_device_position(
            adjust_x, adjust_y
        )
        if adjust_x != 0 or adjust_y != 0:
            # Perform post home adjustment.
            self._move_relative(adjust_x, adjust_y)
            # Erase adjustment
            self.native_x = 0
            self.native_y = 0
```
Given irfan's settings 0,0 is now located exactly where his laser homed which isn't 0,0 in the native coords but actually located pretty notably away from that section.

I'm moving away from home-offset coords anyway since there's a more official move_origin command so at a minimum we likely now need an off switch for home adjusting. If not the home-adjustments to be completely removed.

This beta release I needed to integrate some of the commands like "move origin" into the cutcode more directly since the spooler would issue `move_abs 0 0` commands when it reached one of those `special operations` the problem is because the spooler now has everything as a `spoolerjob` we hit some additional problems. Namely it was spooling `home` and `move_abs 0 0` as spooler operations which could not actually run until the current job (which is what created those spool objects) would finish. So if you had 20 loops where you issued a `home` command, it would only issue the `home` commands 20 times after all 20 loops.

So I moved towards integrating some of those sections and added a real `set_origin` value though jobs do not execute with a particular frame set. So they would all execute absolute_coordinates which is what has been standard since MK started. But, you could issue `move_origin` command that would move local to that origin point. It's notably a limited feature because I didn't want to throw any big wrenches in anything while finishing up the betas and working towards a premier release of version 8.

The home adjustment also works a bit like `set_position` which was also somewhat scrapped in this next beta. So I'm somewhat torn but probably okay with removing that somewhat broken feature to make room for others. The goal is to allow you to run a job a dozen times vs. a dozen different origin positions that could be defined a few different ways. So you could have your template/regmark stuff also provide positions for a job and perform that job in a loop at all the different locations, and make this also allow you the frame of "current location" as to where to execute a job. In theory both current-location and origin would move back to that location when your job finishes. 

Here the home-offset is being a problem and I'm inclined to just remove it even though this would technically remove some a feature it doesn't seem the way lightburn or ruida went and if I'm intending to properly mimic a ruida-device I would need the ability to set an origin point and start a job based on the origin framing.

tatarize, 2022-09-09:
There was a weird issue with panes always crashing Beta16 should be out fairly soon. I need to check on the drawables for the Raster operation thing.

Irfan478, 2022-09-13:
Okey can i Download Beta15 ? Is it working fine ? Okey i will check this

Sophist-UK, 2022-09-13:
beta17 is the latest beta. It claims to be free of all known bugs and the last Beta - though this remains to be seen.

Irfan478, 2022-09-14:
hmm beta17 is ready for Right side homing as well ?
imean my machine home is physically in the right side.

joerlane, 2022-09-14:
Have you tried it? I can do all the same testing if you run into issues.

Irfan478, 2022-09-15:
unfortunately in beta14 I made some settings. due to changings in beta14 I'm unable to run Beta17.

Irfan478, 2022-09-15:
Software stucks ,whenever i download the Beta17 and try to run it

Irfan478, 2022-09-15:
from where ican delete the old version settings files ? as per done by me in beta14 version ?

joerlane, 2022-09-15:
If you open About MeerK40t from the menu, and click on the System-Information tab; it will tell you the path where the configuration is stored on your computer.

Irfan478, 2022-09-15:
Okey let me check

Irfan478, 2022-09-15:
Okey, it's great I have deleted the previously saved settings files and now it's working fine.

Irfan478, 2022-09-15:
I have checked the Beta17 Version and its working fine as per settings. my machine home is in right side. 
Only one little thing is missing iwould like to share. The machine Limit laser movements to bed size is not working. in beta14 iwas working.

Irfan478, 2022-09-15:
the Beta17 version is not limiting the Machine movements to bed size.

tatarize, 2022-09-30:
No version completely limits machine movements to bed size. And it depends a bit on whether the bed size is right. And Home-right tends to need flip-x also checked or it goes the wrong direction at times.

Irfan478, 2022-10-03:
sorry but beta 14 version was been working for me. 
it was limiting the machine 6040 bed size.

closed 2024-10-04 by jpirnay