meerk40t/meerk40t #1306: [0.8.0Beta14] Issues

Sophist-UK, 2022-09-07:
The latest BETA release of 0.8.0 is Beta 14. This issue is for reporting issues with this specific version of MeerK40t.

Please do all further testing on this release. Please do not report any further bugs with previous beta releases. Do give any other types of feedback, questions, concerns or comments.

Sophist-UK, 2022-09-07:
This comment is a holding place for the team to consolidate issues from previous betas that remain unresolved in Beta 14.

Sophist-UK, 2022-09-07:
* [ ] I previously reported an issue with zooming not working. The issue is actually that the modifiers are not checked in the [`event` method in `/meerk40t/gui/scene/scenespacewidget.py` at around line 109](https://github.com/meerk40t/meerk40t/blob/main/meerk40t/gui/scene/scenespacewidget.py#L109)

tatarize, 2022-09-08:
> * [ ]  I previously reported an issue with zooming not working. The issue is actually that the modifiers are not checked in the [`event` method in `/meerk40t/gui/scene/scenespacewidget.py` at around [...]

We never really fully figured out what you were saying that you wanted `Ctrl-mouse wheel doesn't zoom the scene.` But, holding control and moving the mouse wheel did actually zoom. We also added Ctrl+middle mouse button to do normal pull down zoom which was previously part of right+alt only.

Mouse wheel still zooms so I'm going to need additional information as to what you're talking about to address this. that line doesn't check the modifiers which is true but wheelup has always been zoom, and still is zoom, regardless of whether the control key is pressed or not.

Sophist-UK, 2022-09-08:
It depends whether you have checked the `Preferences/Gui/MouseWheel Pan` checkbox or not. With it checked, Ctrl-Wheel does not zoom. With it not checked, Ctrl has no effect and it Zooms unless you have Shift pressed - and you cannot scroll vertically using the MouseWheel at all (I would suggest that Ctrl-Wheel should do vertical scroll when this Pan checkbox is **not** checked).

Sophist-UK, 2022-09-08:
Closing as current beta is now 15. I will not be testing further betas until all (or most) of previously reported bugs are fixed.

closed 2022-09-08 by Sophist-UK

jpirnay, 2022-09-08:
Thank you for testing efforts