meerk40t/meerk40t #1311: [0.8.0Beta16] Issues

Sophist-UK, 2022-09-09:
The latest BETA release of 0.8.0 is Beta 16. This issue is for reporting issues with this specific version of MeerK40t.

Please do all further testing on this release. Please do not report any further bugs with previous beta releases. Do give any other types of feedback, questions, concerns or comments.

Sophist-UK, 2022-09-09:
This comment is a holding place for the team to consolidate issues from previous betas that remain unresolved from previous Beta releases.

Sophist-UK, 2022-09-09:
I will wait for the list of unresolved bugs to be added before testing this release so that I don't waste time testing stuff that remains broken.

Sophist-UK, 2022-09-10:
* [ ] Right click in scene, then right clic again slightly different position - scene moves as well as right-click menu - scene should stay still, right click menu should move.

jpirnay, 2022-09-10:
> Where do I set before and after special operations for every execute?

You have now special operations "util xxx" that you can add to the operation branch in the tree that will provide this functionality (and more)
[image]

Sophist-UK, 2022-09-10:
Nope - not the same functionality at all. Same special operations, but from a user perspective not the same use case.

What I am looking for are the default functions that are executed automatically at the start and end of every Execute.

jpirnay, 2022-09-10:
They are if you add them to the operations - the other functionality you were looking for has been removed in favor of the per file use case.

jpirnay, 2022-09-10:
- [ ] Position pane does not execute a move element if no resize is required

Sophist-UK, 2022-09-10:
> They are if you add them to the operations - the other functionality you were looking for has been removed in favor of the per file use case.

Well that is by itself a good reason to stay on 0.7 branch. Automated before and after special operations were IMO an essential part of the application.

tatarize, 2022-09-13:
> If you are thinking about explicitly removing functionality, you might want to think about getting some user feedback before making the decision (though of course this doesn't apply if this functionality got dropped in error).

I was the one who made the decision and removed it. Mostly because a lot of the core functionality was duplicated there within the injectables. It was discussed and we went with Jpirnay's fix of adding some of them back in as operations during the cut planning `copy` stage. Codewise we also lost `home(x,y)` offset, since that's going to, at least a bit later, be replaced with a more advanced coordinates routine bringing `relative`, `origin`, and `absolute` coordinate systems around. But, currently that bit is removed.

Sophist-UK, 2022-09-13:
It doesn't matter who made the decision, deleting the functionality without knowing whether users will be impacted by it and without documenting the change so users know what has happened is not great practice. Since the repeat-grid functionality was also on that menu, where has this now gone to?

Sophist-UK, 2022-09-13:
Closing because [Beta 17[(https://github.com/meerk40t/meerk40t/issues/1329) has been released.

closed 2022-09-13 by Sophist-UK