meerk40t/meerk40t #1320: [0.7.7] Burns partially inside a closed cut should be burned first

Sophist-UK, 2022-09-10:
At present if you have `Burn Inner First` set, burns that are partially inside a closed cut are **not** burned first (even though the inside may drop out before it is burned).

1. An inside burn with the outside end actually on the boundary of the closed cut are not considered inside. If point 2, isn't addressed, we should add a small (0.5mm?) margin to the outside of the closed cut to check whether a burn path is inside.
2. A non-closed cut or any other type of burn which **crosses** the closed cut should be considered inside so that it gets burned before the piece drops out. To avoid difficulties with two closed path cuts which cross, where deciding which is inside which is ambiguous, you should probably only check for closed cuts being entirely inside other closed cuts.

[Cuts not treated inside.svg](https://user-images.githubusercontent.com/3001893/189484010-f8eb7932-6f32-4feb-ab9c-3d4be3719e61.svg)

jpirnay, 2022-09-10:
[image]
I fail to see the issue in your file. Which part of the burn are you referring to?

Sophist-UK, 2022-09-10:
It may depend on Execute settings - I use Group Inner Burns and the short red open line inside the outside closed path are cut at the very end with Group Inner Burns set rather than being cut before each of the outer curves.

tatarize, 2022-09-11:
> A non-closed cut or any other type of burn which crosses the closed cut should be considered inside so that it gets burned before the piece drops out. To avoid difficulties with two closed path cuts [...]

This actually gets notably harder to detect because it would also hold true for any line segments which do not have a point within the shape. Picture a circle with two line segments going entirely across the circle with no vertexes inside that. 

[image]

You actually would need to check a variety of line segments whether they were ever inside another line segment. It wouldn't be enough to merely detect whether or not a bit of path ever had a point inside another closed path. And really this ambiguity applies to closed paths too.

[image]

Here we have three closed paths, none of which could be burned in their entirety if we are going to burn them in drop order. To do this we would need to use the `O(N²/log(n))` Bentley-Ottmann algorithm. Which would really only apply to line segments though we could use Zingl-Bresenham rendering to create these line segments out of our curves (assuming we're not using something like gcode that could allow us to use circular arcs directly and thus not need the linerization, but any-way I digress). We'd need to find all line intersections and which of these chopped segments are contained within other segments.  So we'd break the shapes down into the sorts of graphs that clipping algorithms use. We could probably use the same sort of scanline work we did with Bentley-Ottmann (find all line intersections between all lines we have here) to solve for the insideness of the particular segments. This actually is somewhat how the current insideness routine works. It does a scanline system to determine whether if all the points of a particular shape are inside another shape. And the non-closed shapes and the closed shapes would need to be potentially broken up in weird ways.

And this is seemingly to cover a somewhat rare issue that actually should not come up that often.

tatarize, 2022-09-11:
Fun quirk, the second example there can actually be solved there in a single path as one would assume from the clearly Eulerian nature of the nodes, however it can't be solved if you start from any of the three inner-most vertexes, which would expect given you need to cut the center bit first. You actually must start at any of the 3 outer-most vertexes.

I don't quite see how that would be easily solved. I guess 2-Opt with constraints might be able to solve that, but this stuff gets pretty deep in graph theory.

--- My official recommendation for this problem is not to fix it. It's one of those nagging edge conditions like cutting a circle inside a raster. If you order that correctly the raster which has bounds entirely outside the circle is filled and actually occupies all parts of that circle. So the correct order is raster then cut, but it's kind of hard to detect that situation.

Sophist-UK, 2022-09-11:
I accept that there are some topologies that are not easy to solve - but that does not mean that the cases described in this issue should not be addressed.

inspectionsbybob, 2022-10-26:
I also have an issue with outer cuts "falling out" before the inner cuts are made.  If outer cuts cause a fall, the inner cuts may be out of focus and may not cut clean. 
Maybe part of Optimise to try to find the innermost cuts and work out so that when objects drop/fall, nothing needs to be cut/rastered onto the part that dropped. That test may be easier, find the innermost and work outwards vs out to in...

jpirnay, 2022-11-02:
[image]
The horizontal lines are overlapping the outer shape by just so much. If I add a tiny tolerance to the algorithm to establish the 'inside', then it would be properly optimised.
[image]

So the question is whether that would not make sense to apply as a general algorithm (maybe device dependent).

jpirnay, 2022-11-02:
> 

That should be properly addressed with the 'Burn Inner First' optimisation potentials.

[image]

jpirnay, 2022-11-02:
Partially addressed in https://github.com/meerk40t/meerk40t/pull/1470. Partially as it does not fully cover all cases, but the 'culprit' case mentioned would effectively be covered

closed 2024-10-04 by jpirnay