meerk40t/meerk40t #1332: [0.8.0Beta18] Issues

tatarize, 2022-09-16:
Checking through all the housecleaning.
- [ ] Translation templates updated to recent versions.
- [ ] GUI Translations to German at least checked. Competed automatic translation if time crunch.

tatarize, 2022-09-17:
RPi. #879 

* Call Blob more meaningful. 
    * We're moving away from ExecuteJob and will probably drop the whole thing at some later date. Blob internally means cutcode-ify.
* rPi is missing tree open/close arrows - @joerlane says this is a Linux issue and depends on windows manager. Probably not fixable.
    * This is annoying but I'm not going to mess with it, it's not really a software issue within meerk40t.
* When last CutObject clears and background reverts from red to grey cannot use emergency stop.
    *  I believe this was fixed with TCP and it does send abort sequences even without knowing anything's in the buffer, and can quit the wait for finish routine.
* View invert - doesn't work on rPi - should we drop it altogether?
    * View invert is more hidden now. Won't be dropped.
* Rasters appear grey in simulation but vectors show as colours - this is inconsistent. Should everything be grey scale and based on burn PPI i.e. use grey scale to show vector PPI
    * Colors serve a purpose.

tatarize, 2022-09-17:
RC-3: #909
- [ ] configuration files do no persist if the program crashed during execution.
- [ ] GAG-Union can cause long infinite looped pause.
- [ ] Property windows blanks after changing a path's color. *yup*
* Property window doesn't do multiple item manipulations rather adds more things to the notebook.
    * True but not fixing just yet, might need multiple object special property windows.
- [ ]  If flow text pointer exceeds data then the current run-batch is degenerate. This should issue a warning during plotplanning. (this needs to be checked, what happens if wordlists have no value)

tatarize, 2022-09-17:
RC-4: 
- [ ] Preferences is “gone” mac style in the 0.8.0005 frozen build. No longer in the menu or working with “Apple-,” keyboard command. (Not resolved)

tatarize, 2022-09-17:
RC-5 #967 
- [ ]  Ruidacontrol needs to respect (receive) EOF before starting the received job.
- [ ]  Ruidacontrol has movement issue when moving between raster objects. This actually shuts down my Y axis stepper driver chip; and no more Y moves happen on my machine until I reboot the k40.
- [ ]  Rasters at even 100mm/s are problematic. 200mm/s seems unusable.
- [ ] ez cad shows an elapsed time for the last "execution" in the status bar at the bottom of the window, which is extremely helpful in tuning a job for cutting as fast as possible for example. (While we can display this it's the actual time it took to send the raw galvo data, and not the time the laser itself took.)
- [ ]  Wobble should be able to specify inner or outer edge; or center of vector path; currently it requires extensive work in Inkscape to make an inset to be able to use on a specific path for that reason (to be able to respect the path as a cut).
console Operation frequency xxx is not a valid command
- [ ]  .rd files do not load. (attempting to load one crashes)

tatarize, 2022-09-17:
Beta5:
- [ ]  Selecting in the scene should give some indication in the tree what has been selected if the parent nodes are closed - perhaps a squiggly blue underscore?
    * This is accurate and does show the tree has lost some of these indicators. We used to indicate when a selected node has references in the scene or the parent of that node. We do not currently have any of that other than mere selection. This is certainly reduced functionality that we can accept but should probably make a note for the release notes.
* Selecting in the scene all elements in a group should (IMO) emphasise the group also (or possibly select the group and emphasise the children). 
    *   This seems somewhat prudent but won't be done given the lack of other indicators.
* Should selecting in the scene "select" elements in the tree rather than (as present) "emphasising" them?
    * There was a bug in the GTK implementation that made emphasis a thing. No. Selecting them can suffer some secondary issues. With wx4.0.x which we could actually stop supporting at some point.
- [ ]  When you select preference 'Classify in reverse', overlapping filled paths are combined in the wrong order in simulate (and presumably burn). You need to check this setting and reverse the order when you create the raster image.
    * This is accurate and will probably not be fixed. The order in which filled objects are placed into the operation actually gives the drawing order for the rendered raster image. So you may see on screen Bigger Blacker Circle under Smaller Whiter Circle but if they are in the other order inside the operation they will actually render showing only the Bigger Blacker Circle. In later versions of MeerK40t I am anticipating directly permitting operations view mode in the scene.

tatarize, 2022-09-17:
* Kerning adjustment (change spacing of the text rather than size of the text for matching things).
    * This requires having direct access to the actual glyphs themselves which probably means parsing the font itself.
* Click on element and trying to move as a single action did selection rather than moving the thing around.
    * While still true, this might be added as an option later.
* No distinction between the designing of the objects and the controlling the laser. It was not clear whether some buttons she would hit like "Home" were controlling the laser or returning her to the bed view.
    * This won't be addressed directly but there are future plans for coordinate system upgrades which would draw a lot more things in the scene and one of those useful things would be the ability to have different views of the data. So rather than just drawing the elements we could draw the operations as they stand, or draw the operation in relative draw mode.
*  She expected a ribbon bar buttons for adding an image.
    * This would be easy but won't be added.
*  She may have benefited from a zoom-to-bed button as well.
    * There exists a button in view for this, but it's not on the ribbon bar.
*  She wanted to do design work and the tools page in ribbon weren't obvious though she did find them within like 30 seconds.
    * There is some need to fix this cosmetic stuff.
*  Heat Impact calculation chart within properties. Average pulses over area time. PPI/DPI/ Speed to calculate equivalent speed.
    * We currently run time calculations as part of there it wouldn't be too hard to calculate an equivalent setting. But not for this release.
* Selection rect in rPi looks solid rather than dashed.
    *  This is because the scale of the drawing of the rectangle is big at the scale rather than small. So the rectangle is large so consequently the drawing of the rectangle looks solid rather than dashed because they are very tiny dashes.
*  Tools should not toggle. You hit text tool again, it should stay selected.
    * This would take modifying button_click in wxmribbon.py and I didn't have too much luck doing a quick hack to do this.
- [ ] Ribbon should have an option to turn off the text labels for the items. (takes lots of room).
    * The branch `micro-ribbon` is testing art that turns off these labels and makes them take nearly no room (makes them tiny font and the color of the background). We could also make them blank.
* Options/Preference menu needed, shouldn't be in the ribbon. Should be used more rarely but have it's own menu.
    *  Hm. Mac has this as a requirement.
* Click drag and move right away in single action if you're in the selection tool. If you're in say circle tool, that should draw a circle.
    * We're exclusively click then click and drag for now.
*  Units can give full name in preferences (there's enough space).
    * Tried this they are evenly spaced in the radio box so inches and mils get less space and it wasn't confusing.
*  The center dot for moving should maybe just use the entire center of the selection as movement.
    * Won't fix this time around but a large center rect for moving would be fine.
*  Click center rather than other options should be a toggle between scale and rotate handles.
    * This might actually want to be fixed. It would make things probably look better.
* Super-basic object nesting based on bounding box.
    * This would be moved to 0.8.2 at a minimum, probably as an advanced alignment tool.
*  On import zoom to newly loaded elements rather than bed.
    * I have been informed this is to spackle over some movement issue or something that I don't seem to get.
*  alpha/black view did not disable correctly was still cached.
    * This may need to be reviewed but few people disable that or know what it does.
- [ ] Get rid of the move outside the bed option completely, rather than giving people a button. Always stay inside the bed.
    * I might actually want to do that. The option to jog outside the bed doesn't seem very useful. Projects can still move there but it would just refuse with the gui. You could probably still command it outside the bed with the console.
*  jpirnay reiterates the need for path offset.
    * Requires calculating tangents to curves, and thus normal angle from curves and placing points a certain distance away from the normal angle.
- [ ] Point tool's point object did not scale correctly in GTK, they were super tiny and needed to be zoomed way in to find the points added.
- [ ]  Command W does not work on Keymap and Rotary. (This is confirmed to still be an issue)

tatarize, 2022-09-17:
Beta 12: #1285

* Preferences window default window size is taller than it needs to be
    * For the elements it's shorter than it should be. It's a very dynamic window.
*  Tools/Camera Interface vs. Camera button on toolbar - they store separate window positions/sizes - is this intentional?
    * Yes. It's intentional. It's so you can have lots of different cameras in lots of different places.
- [ ] Scene bounding box - seems to fit correctly on selected items horizontally (i.e. vertical lines match extremes of selected object), but vertical boundaries (horizontal lines) are incorrect (too close together).
    * Some work was done to use painted vs. geometric bounding boxes the home was this was somewhat the cause but it's not clear that it is. 
* Tree labels - I actually think the shorter labels in 0.8 are better than the length labels I did for 0.7, but there is probably more information that would in fact be useful - like path fills.
    * A feature to make this configurable is in the 0.8.2 branch for post-full release.
* Scene background - should probably be different for each device. Each device should probably also have a default camera associated.
    * Scene background is not. In fact, the scene is general and universal regardless of the backend device by design. Some things like the bed size should matters but that's just for displaying.
    * Object sizes appear to be slightly wrong. For example, when I create a path that is exactly 230mm in width with a stroke-width of 0.2, it loads into MK with a width of 230.09mm. (See above image as example.) Similarly I position this at exactly x=0.9mm (including stroke width i.e. at exactly x=1mm) and it loads as x=0.96mm. Again, for me, this makes MK 0.8 unusable in production. (EDIT: I need to look into this further, as older SVGs I have saved appear to load accurately. So may be an inkscape setting or bug. However loading the same file in 0.7.7 and 0.8b12 gives different sizes and positions!!)
    * The hope is this is paint-bounds vs. geometric bounds and fixed but it's not confirmed.
* Operations nodes in tree showing "@default" or "default" when the default checkbox in op-props is not set. What do these means and what is the significance of the "@"?
     * The default values here are not given but this actually needs a global value for all operations that are explicit and used when normal values are not given. So a Null would not be needed here. I think `@default` is Power.

tatarize, 2022-09-17:
Beta13 # 1291
- [ ] Weird issue adjusting pane sizes. 3 panes in left zone, adjusting split between 2 and 3, as I drag down when pane 3 get to minimum size as you continue to drag down the top of pane 2 starts to move upwards.
    * I've heard reports of this and it should probably be investigated. 
- [ ]  When multiple nodes are selected, Grey out (i.e. disable) all right-click menus that only operate on some but not all of the selected nodes.
    * This should probably be done.
- [ ]  Image raster burn failed - I had spurious burn marks (faint - suggesting that they were a single raster line across @ step 2). These extended outside the right side of the image bounds, but the head didn't lose position, but eventually the head went so far outside the image bounds as to hit the stops. Switching back to 0.7 until this is resolved. (I should add that I had previously done a test burn of this part of the image with no issues.)
    * Critical bug should probably be checked faint marks issue. Logging for testing.
* SVG contains an image at 1000 DPI. In 0.7 I need to use a raster wizard to reduce to step 2. In 0.8 default DPI is 500 and it burns as I would like. But why is default DPI 500 and not 1000? I have always felt that the default step in the 0.7 wizard should have been 2 rather than 3, so IMO 500 is the ideal default value, but it will result in different burns in 0.8 from 0.7.
    * The wizard thing differed by script. It wasn't actually just picked one. It was 2 generally speaking. If 1000 DPI it would be step 1 and that burns things and people complain. The argument is the default settings should work well enough for anybody who wants to try them without anything needing to change.
- [ ] Note: There are lots of new functions in 0.8 (like image properties and wordlists/text) and I have not yet tested that these are all saved in SVG files and restored intact. This is something that would probably benefit from unit tests (because it is not dependent on GUI).
    * Unit tests to save and restore things are actually a fantastic idea. Unit tests absolutely let you flag a file for deletion after a test and you would simply do some console commands make something, save it load it and see if it lost or retained properties.
* Raster Wizard menu items do set image properties as expected, however image properties window needs a preview of the image (like 0.7 raster wizard) to provide equivalent functionality. Based on experience with 0.7 I think that this preview is essential not only for usability but also so you can spot defect artefacts that are created. We also need a Raster Wizard reset to reset Image properties to zero in case you don't like what you have done.
    * This is a misperception. The image in the scene *is* the preview image. It doesn't actually take effect until cutplanning and even then only for the image that is being planned. It doesn't actually apply the raster-wizard script except for previews and this script is able to be altered and fiddled with before, during and after burning. It only gets a real concrete value after you it's sent to the planner.
*  Ctrl-Mouse-Wheel still not zooming.
    * It should. I don't know why holding control would matter though just regular mouse wheel zooms. And I added Mouse3 + CTRL for the pull and zoom.
* Pinch-to-zoom on trackpad also needs to be added.
    * This was tested a fairly short time ago and worked, it was too fast and zoom_scale was restored.
* Ctrl+= is zoom in (from View menu) - though it should be Ctrl++. This appears to be a wx bug as the code is clearly Ctrl-+ and not Ctrl-=.
    * That's where the + is, capital shift+= and we don't do. This is the key I would hit to zoom as such.
* Group inner burns is NOT working even though the checkbox is ticked.
    * This appeared to work in my tests.

tatarize, 2022-09-17:
Beta16 - #1311
*  I used the raster wizard in 0.7 with step 3 (i.e. 333.33 DPI) and saved. When I load in 0.8 it shows DPI of 500 in Image Properties. Not sure how it will raster - but the UI is very misleading.
    * It would raster at a DPI of 500. No misleading. It doesn't know how big the image was. However it might if save images with their correct DPI and 0.8.x now loads that DPI more directly.

tatarize, 2022-09-17:
Beta17 - #1329

*  Camera - select a different camera in the dropdown on the button bar and then click the camera icon - correct camera is often not opened in the camera window.
    * The camera loaded is tied to the camera window. The windows have URI and it will say what URI that camera window is trying to open. This is not in error, though it maybe poorly defined. Modernly it loads the URI equal to the window by default and uses URI manager to put other ones in there.
- [ ]  Path properties - small X under the stroke/fill colour is not great UI. Can we use a much bigger border for the colour button to indicate this. Mouse over on colour button changes the border (good) but clears the colour (bad). 
    * This is accurate the small X under the color is not great UI. I wasn't going to look a gift horse in the mouth though.  It would be better to give the one that matters an X but in a way that doesn't conflict with the transparent button (the one that means transparent rather than the one that you can't actually see).
* Step repeat functionality in Execute Window still missing. 
    * This functionality was lost and will not be restored until after the 0.8.x release. We have an origin point or several and we will use those for step-repeat in the future. I didn't want to mess with the release by changing a bunch of code there. But due to  #1307 the home offsets were removed to deal with issues with them being at 0,0 (which actually caused them to be really weird for top-left home users). And the additional later functionality will cover the step-repeat and then some. This loss of functionality will be noted in the release notes and you could still use grid, though it's clearly not the same.
- [ ] Two-stage collapse rather than collapse all children.

tatarize, 2022-09-17:
### Issues from very old crash logs. 

- [ ] SVGs can reference Images located in external locations. If these images do not exist, it does not tell the user that that image was not valid. It silently doesn't include it.
- [ ] DXF Hardening and recovery #739

### Additional Perhaps covered above:
- [ ]  Loading RD files crashes.
- [ ] The wx.ID_PREFERENCES is not used in tools and so Mac does not load Preferences with the button it tends to have.
- [ ]  Passthrough mode for image -- A missing essential feature is to mark images as passthrough such that they do not ever alter the image and restrict the image to specific sizes. They would be required to set their DPI equal to the size and never invent new pixels.
- [ ]  RasterWizard applied to Images does not save the raster wizard script to the SVG. This means loading and saving of that image is *lossy*. You will only get the original unaltered image and must reapply the script.
- [ ] RasterWizard images cannot be forced into being real and saved out. This prevents MeerK40t from working as a image preprocessor for other programs which was a feature from 0.7.x/0.6.x. If this is done within MeerK40t it should force the image into Passthrough mode, which is actually quite similar to how 0.7.x worked.
* Functionality Loss Note Rotary (doesn't do the stretches like it used to. You could apply this in as a belt stretch in some cases and it would be fine, but more proper rotary required more work so it doesn't even apply weaksauce rotary currently. For example if you could tick a rotary with the GPIO of a M3:10 Nano (new stock board) you'd still have access to your y-axis. And this sort of thing would need to be accounted for typically with splits, how much work can you do before you need to tick your rotary.

Sophist-UK, 2022-09-17:
Apologies for today's testing being on beta17 - I should have checked to see if beta18 was available.

* [ ] Beta 18 is being detected by Microsoft Security as `Win32/Contebrew.A` and deleted after download!!!

tatarize, 2022-09-17:
I told it to rerun that. Hopefully better luck on the false positives.

tatarize, 2022-09-17:
When finished it should update the timestamp to today rather than yesterday.

jpirnay, 2022-09-17:
- [ ] Each burn is now shown as a single item in spooler even with special ops and no merge ops. I can see this is a good thing when e.g. you are running several lasers in parallel and you want to see each job as a single line. However this means that you cannot e.g. set a job upon with several passes on engrave and then several passes on cut, and kill off individual passes once you have enough burn. This was a VERY useful ability in 0.7 and I miss it already in 0.8. Can we please have the ability to select a burn in the spooler and open a window showing individual "operations" bearing in mind merges etc. and e.g. cancel later ops?
- [ ]  !!! Execute - multiple operation passes not burned !!!
- [ ]  Spooler - For executing job, estimate column shows elapsed time not estimated total time.
- [ ]  Spooler - Name column should show project name or similar (for multiple projects to multiple lasers simultaneously) - number of items should be a separate column.
- [ ]  Spooler - I would prefer to see all spool jobs to all devices (rather than have a dropdown), and have a separate device column to show the device it is sent to.

jpirnay, 2022-09-17:
- [ ] Despite a job bring abortted/ended mk might still warn you about pending data

tatarize, 2022-09-17:
> Each burn is now shown as a single item in spooler even with special ops and no merge ops. I can see this is a good thing when e.g. you are running several lasers in parallel and you want to see [...]

They are all loaded into LaserJob class which itself has the ability to run loops. So you can run an infinite laser job that will loop forever and won't reprocess anything and will just run that bit of cutcode over and over again. There are plans to expand some of the coding in operations since it's basically able to set up a kind of step by step program with certain numbers of loops. Oddly, no. If you are running several lasers in parallel they each have their own spooler and they don't combine spoolers. You'd need to switch to those devices and check the spoolers, and I think the spoolers window might let you select the individual devices too to view the spooler.

Generally there are cases with galvo lasers and probably some others where you might want to run 1000 passes of some job, and then 10 passes of a different couple after that. We've batted around the idea of having post-ops and pre-ops as parts contained within the operation branch or outside the operation branch. And doing things like making operation groups that run different loops a different number of times. I did not, as it happen consider the workflow described in that.

You could perhaps mimic this if you checked properties on `Operations` and set Loop Continuous then selected your engrave operation and said `execute operations` or `simulate operations` and then hit start. Then do the same for the *other* job. you want to run likewise.

[image]

You then get this. Run the first one as long as you want and when you've had enough cancel it and you'll start doing the next job.

I think the DPI change may have broken some other stuff. My loaded image simulated and ran in a completely wrong position. I don't understand why.

tatarize, 2022-09-17:
Keep in mind those are two different jobs since I started them differently (you could do the same thing with disabling the operations too, though in that case you'll get the beeps and homes each loop since it'll do the default operations). But, run one for a while until you're satisfied. Then run the next, by canceling the first. If anything it might be cleaner. Though the Spooler 1 items isn't the most helpful bit of information there.

Sophist-UK, 2022-09-18:
I am sorry, but clearly I have failed to explain clearly enough why this is an issue.

As a user I would like sufficient granularity of access to the compiled burns that I can easily create (i.e. by increased operation passes) a superset of the burns I want, and then use the UI to cancel those that turn out to be unnecessary. I could do this in 0.7, I can't do this in 0.8. That is functionality that is now missing!!

Explaining how I can jump through hoops to create a similar but different solution is (yet again) dismissing a user issue without seriously considering whether a change has been detrimental or not. If you are going to be dismissive about my input, why should I bother?

tatarize, 2022-09-18:
> As a user I would like sufficient granularity of access to the compiled burns that I can easily create (i.e. by increased operation passes) a superset of the burns I want, and then use the UI to [...]

You're probably right. I don't see an easy way to correct this problem. The reason for the switch is that LaserJob was introduced, this was to combat a few different problems. Basically galvo light-jobs need to be submitted and run in a way that weren't just a precompiled bunch of commands. For example, you can have live outlining job. So the laser will outline the job. And anything you select in the scene gets outlined right there for you with the laser as it's running. This isn't something you could do with the older methods. And there was a need to do huge loops. Like infinite loops of the number of cuts for galvo. Like Make an infinite loop, and run this until I tell it to stop. So that stuff couldn't be pre-compiled into cutcode, since I can't make infinite cutcode.

So we have jobs that are generally arbitrary. And the old work that was done in the older spooler class is now done in the LaserJob class. Which actually contains a list of cutcode within it and loops over that cutcode the number of times it was told to. The spooler consequentially just has the laserjob and data about that job. We could add in an editor for the laser job. Which could let you view what's inside it (a list of compiled cutcode) and maybe fiddle with that if we could do so without things crashing. But, that would require some features we currently don't have.

So yes, this loses some functionality, but it gains a lot more functionality, it's just the functionality gained isn't necessarily in the same place as the lost functionality.  You'd need to be able to peer inside the running laser-job and edit the various parts. Or I guess tell it to stop the execution of one part and just directly move on to the next part. That's a fine feature (please go ahead and open a separate issue for it since there's some other needs for more looping and configurable laserjobs across the board) but it's not something that can be rectified in the first version for a full release here.

BTW: Operations branch itself has properties on it for setting the loop count of the laserjob. And more things could be put in there as they are needed.
[image]

You can absolutely run parts of jobs for any number of times you want and let it loop until you're good and cancel them and see how many times it looped for. So you need to cut something in 200 passes you can do that. Stop it when it worked and have the pass count for when it finished up.

Sophist-UK, 2022-09-19:
I understand why you needed to do what you did, but in essence we need another layer of granularity inside a LaserJob which reflect the granularity of optimization i.e. if you are merging stuff (and optimizing the merged burns) you cannot break that into smaller chunks, but if you are not merging then each pass of each operation needs to be a separate sub-LaserJob.

Then you need a window which you can open for a selected LaserJob which shows these sub-jobs (in a very similar layout to the spooler), and a button to open this window on the current executing LaserJob (i.e. so you can open it instead of the spooler).

RuKeBo, 2022-09-19:
Is there a reason that under "Config", "Default Actions" there is no button to "Add to Job End"?

RuKeBo, 2022-09-19:
The default color for the "Hatch" operation is a dark-green but this is not available as a color in the Elements Properties for Fill or Stroke Color.

RuKeBo, 2022-09-19:
> > The default color for the "Hatch" operation is a dark-green but this is not available as a color in the Elements Properties for Fill or Stroke Color.
> 
> I had the same problem finding it. It is at the very bottom of the window - the UI is adequate but a bit confusing IMO. Some right-click options might be helpful and some tooltips.

This Window?
[image]

tatarize, 2022-09-19:
[image]

It's there tiny bit of nothing at the edge there just after the add to job start button, that's the add to job end button.

RuKeBo, 2022-09-19:
> [image]
> 
> It's there tiny bit of nothing at the edge there just after the add to job start button, that's the add to job end button.

Ok.  I thought he was referring to the Hatch color issue since that is what was quoted.

tatarize, 2022-09-19:
> The default color for the "Hatch" operation is a dark-green but this is not available as a color in the Elements Properties for Fill or Stroke Color.

Yeah, turns out I short handed that as `Color("green")` since it'll just read those according to the SVG spec. So it seemed pretty reasonable but that light green #00FF00 is sometimes called "lime" and normal "green" is #008000 or so. If you type `dialog_stroke` you which would open a color picker for the stroke, and let you specifically select it. I suppose changing it to lime is going to be the easiest.

tatarize, 2022-09-19:
I switched the hatch default to `lime` which should give you normal #0F0 color for that. Though it's more of an angry fruit salad color of green.

RuKeBo, 2022-09-19:
> I switched the hatch default to `lime` which should give you normal #0F0 color for that. Though it's more of an angry fruit salad color of green.

This is not to instigate a change, just for thought.  The numbers of colors shown is obviously limited.  If I do change the stroke or fill color to (dark) green, then there is no indication what the color is since it only displays the 8 primary colors plus "none".  My thought was why not just have a box that shows the current color, or "X" if none, and you click on it to change it using the color picker.  The draw back to this, at least the one I can think of, is that it is more steps and you need to kind of know the difference between different shades of colors.  You need to select the exact shade of blue that the default Engrave is? I'm not saying this is necessarily hard or easy but there is more to choose from and I did need to look around.  Do we really need that many options, after all, there are really only a handful of Operations and I would think most people will only use 3 most of the time.

tatarize, 2022-09-19:
In theory you can bind a key to `dialog_stroke` if you want. 
Console `bind c dialog_stroke` then just hit the `c` key and set the color of the whatever you selected to anything. You can also set the colors in console, there's actually 16 million colors to choose from, it's just that it made more sense to give the gui a small subset of those so people classify things or whatever.

Sophist-UK, 2022-09-19:
> This Window?

No. I quoted the wrong comment. "Is there a reason that under "Config", "Default Actions" there is no button to "Add to Job End"?"

jpirnay, 2022-09-20:
a) Default actions:
[image]

b) Color for objects (stroke/fill):
[image]

RuKeBo, 2022-09-20:
## Hatch  Operation
I got this to work but seems inconsistent. If I add a Hatch Operation, go in and change the angle using the slider to change the angle, then close the dialog, the angle does not change.  The slider does update the angle number display dynamically.  If you enter a number into the Angle input, it does not update the slider dynamically.  If I put 45 in the angle, the slider does not move to the 45 position.  If you change the angle by entering an angle or using the slider, it will not save the value when you close the dialog.  What you need to do, is press <ENTER> in the Angle input field in order to save the value.  You can change the angle using the slider, which does dynamically update the input field, but you must put your cursor in that input field and press <ENTER> in order for the display below it to update and save the value.  If you change the angle in the input field, it does not update the slider dynamically and you need to again, press the <ENTER> key in order for the display below it to update and save the value.  If you use the input field for the angle, press <ENTER>, close the dialog and then go back into the Hatch properties, the slider will be updated to the proper location.   
I don't know if this same procedure occurs elsewhere in the program, I would need to check. I am assuming that automatically saving the value is difficult in programming.  If so, maybe an "Update" button would be more intuitive instead of having to press the enter key in the input field?  Personally I think the slider should move dynamically with the number in the field but not sure how that would work when you want to enter 180, would it update to 1, then 18, then 180? Is that logical, maybe so.  At the least the slider should probably update when the "Update" button is pressed, it does not update when you press <ENTER> in the field, only when you open the dialog again.

RuKeBo, 2022-09-20:
## Units   
You are able to select between mm, cm, inch, and mils in Preferences.  This changes the bed scale.  However, it does not carry through to the rest of the program.  For instance, in the Properties for an object, the "X", "Y", "Width", and "Height" always show in "mm".  I would think this should display in the chosen units in Preferences.  I happen to use "inch" but then need to convert everything in other places.  I know you can enter the numbers here using "in" but when you close and re-open the dialog it goes back to "mm".

jpirnay, 2022-09-20:
This was a very deliberate design decision, you can enter data at will into the fields, but mk will work with your changes only after one three things are happening: 
1) you press enter inside the field
2) you press tab to jump to the next field
3) you click on another element /fields inside or outside the field
While you could argue that within one tetxfield every single keystroke should be directly applied, there are many situations where that could lead to undesired consequences, hence mk waiting patiently for your conscious decision to apply a value/a set of values

RuKeBo, 2022-09-20:
> This was a very deliberate design decision, you can enter data at will into the fields, but mk will work with your changes only after one three things are happening:
>
> 1. you press enter inside the [...]

Ok, fair enough.  I didn't think it updated when you moved to another field but it does as you stated.  
However, the slider does not update with the input field even after you move to another field, press <ENTER>, or <TAB>.  Also, I am able to put in a negative number but the slider cannot display this since it starts at zero "0".  There is no way to deal with this though, just mentioning it for some reason.  If I wanted a cross hatch I found it easier to put in 45 for one operation and -45 for the other, I could just have easily put in 135 instead of -45, but that was more math to do in my head.

jpirnay, 2022-09-20:
> However, the slider does not update with the input field even after you move to another field, press , or . Also, I am able to put in a negative number but the slider cannot display this since it starts at zero "0". There is no way to deal with this though, just mentioning it for some reason. If I wanted a cross hatch I found it easier to put in 45 for one operation and -45 for the other, I could just have easily put in 135 instead of -45, but that was more math to do in my head.

That's a bug - fixed. Thank you

RuKeBo, 2022-09-20:
I might be missing something here.  If I open a file with a filled circle, square, and ring into Meerk40t, set the operation to hatch, and Simulate a burn, it seems to be hatching the stroke, which has no color and did not have a stroke in Inkscape.  
[image]
Note the only the edge of each is hatched.  
Here is what it looks like in Meerk40t  
[image]  

If I create a rectangle directly in Meerk40t, give it a fill and set it to a Hatch Operation, it looks correct in the Simulator, it will show it hatching the interior of the rectangle.  I cannot add a rectangle, or anything to the SVG I opened that was made in Inkscape to test them together, it crashes every time.

tatarize, 2022-09-20:
> Ok, fair enough. I didn't think it updated when you moved to another field but it does as you stated.

There's some debate as to whether or not it should change each keypress. It used to for something, and it might be somewhat reasonable to allow it. If the thing you typed was a valid length it could well update the underlying value. It's sort of nice to see things change as you change them.

tatarize, 2022-09-20:
> If I create a rectangle directly in Meerk40t, give it a fill and set it to a Hatch Operation, it looks correct in the Simulator, it will show it hatching the interior of the rectangle. I cannot add [...]

What crash? Seems like fixing that should be pretty straight-forward and perfect for the current work.

It looks like it might be working but isn't showing that. The edge bits there are not the hatch. That's actually the movement along the edge of the hatch. There's actually a chance that your hatching is right but there is no stroke value and without the stroke value it's making the hatch but transparent.

tatarize, 2022-09-20:
300mm/s is pretty fast for a hatch unless you're running a galvo laser. This stuff is somewhat technically still a vector cut that does an angled fill of various shapes. So in a gantry laser 300mm/s is likely unsafe speeds.

If you check the simulator and move the progress bar you'll see that it *is* doing a hatch there. But since it did that for vector objects with no strokes the strokes themselves are invisible which isn't ideal. It actually *does* work, it doesn't show you that fact though. Also, the edge bits are the movements where the laser is off.

RuKeBo, 2022-09-21:
I did confirm last night that it does hatch the actual shape on the K40.  
  
I believe the 300 is ppi not speed. but I was only using it for testing, not an actual burn.  The speed says "default", more on the later. When I open the properties for that operation it is 35 mm/s and 300 ppi.  
  
Funny thing is that is seems to be working this morning, partially.   I wanted to experiment with a cross hatch so there is one hatch on 45 deg. and a second at 135 deg.  It will show the lines for the first 45 deg. hatch but does the same thing as before with the second hatch, does not show the hatch lines but shows the ticks at the edges.  I did two things;  I rebooted my computer and I added a rectangle from within MeerK40t, the later is what seems to have helped, not rebooting.  The rebooting may have helped with the crashes.  As I said, it was crashing every time I added a shape in MeerK40t, I sent the crash logs.  It does not do that now that I rebooted.  Since I was able to add the shape in MeerK40t, it will show the first hatch operation lines.  If I remove the shape, it goes back to how it was before, does not show the hatch lines at all.  
  
Here is another peculiar thing.  I decided to add a 3rd Hatch Operation to see what would happen, I duplicated the first one, change the angle to zero "0".  This means in the tree I first had a Hatch at 45 deg., then one at 0, then one at 135 deg.  It showed the hatch pattern in the simulator for the 1st and 2nd hatch operations.  I duplicated the 2nd hatch operation and changed the angle to 90.  It then showed all but the last again.  I duplicated the 2nd again and changed the angle to 135 deg, the same as the last hatch operation that was not working and disabled the last one.  It showed all four.  There appears to be an issue with that last Hatch operation.  I am not sure how I originally created that whether it was added as a Hatch operation or an Engrave was converted to a Hatch.  You might notice it says "default" for the speed of that operation while all the ones that work say 35.0 mm/s (I'm guessing this in and of itself is not an issue).  I did change it to 25, didn't help, then back to 35, still didn't show they pattern in the simulator.  These seems to be something going on with that Hatch Operation, will try and do more testing on it.  

[image]
  
Just to wrap this up, if after I have the four hatch operations and they all show in the simulator then delete the rectangle I added from within MeerK40t, re-run the simulator, the hatch lines do NOT show up.  It will only show the lines if I add a shape in MeerK40t.    
  
Ok, here is the reason, or why it will or won't show...In order for the hatch lines to show up in the simulator, I need to add a shape in MeerK40t.  In order for the hatch lines to show in the Inkscape shapes, the MeerK40t shape needs to be in that Hatch Operation AND, it needs to be last (below) any and all of the Inkscape shapes in the Operation tree. If I put the MeerK40t shape 2nd in line, it will not show the hatch lines for any shape, including the MeerK40t shape, it has to be last.  I assumed it would just not show the hatch lines for the Inkscape shapes that were above it but no, it won't show any of the Hatch lines for that Hatch operation for any shape in the project, including the Meerk40t one.  I tested this by putting the MeerK40t shape above 3 Inkscape shapes in a Hatch Operation, that Hatch did not show up in the Simulator.  I moved it down one so that one Inkscape shape was above it, 2 below it.  Nothing showed in the simulator.  I moved it down again, nothing.  Moved it down again to the bottom, the hatch lines then showed up in the simulator.

RuKeBo, 2022-09-21:
## Move Laser to  
This is  acting a little strange but predictable.  I am using inches units so it looks like this as default:  
[image]
Notice it says "0inch"  
I can type over the "0" with a "4" and get this:  
[image]
If I remove everything in the box it will not allow me to put back in "inch":  
[image]
After the "c" it does allow the letters: "e", "t", "I", "i", "p", "l", "x", "c", "n", and "m".   
  
It will take "in" for "inches" and work properly:  
[image]

tatarize, 2022-09-21:
Hm. It's supposed to just let you type anything in there and flag it red if it is not going to work and rather throw a value error.

RuKeBo, 2022-09-21:
> e the progress bar you'll see that it _is_ doing a hatch there. But since it did that for vector objects with no strokes the strokes themselves are invisible which isn't ideal. It actually _does_ [...]

I didn't understand what you were saying at first but putting a stroke on the object in Inkscape does show the hatch line sin the simulator without having to add an object in MeerK40t.  Why and how adding the MeerK40t object to the project works to show the hatch lines in the simulator is a mystery though.   
  
Also, what determines the Hatch line color in the simulator.  I can't get this to change or know what the color will show up as.  Doesn't matter at all to me, just curious if there is a reason for the color it uses.  All the objects are the same fill color and same stroke color and same Operation color but one of the hatch lines in the simulator is lime and the others are blue.

tatarize, 2022-09-21:
The strokes are there but invisible. You can see them if you animate it in simulator. I'll take a bit of time to fix it in a bit. I got the whole inches bug thing. Basically "transparent" is a color for the stroke and it's what the hatch is colored in in the simulator, which isn't a color you can see.

tatarize, 2022-09-22:
- [x] Hatch color being None (transparent) if the hatched object's stroke is initially None.
- [x] Text box refuses to type "inch" since no set available unit had an `h` in it.

RuKeBo, 2022-09-22:
Not sure how this is supposed to work exactly so maybe just confirm if it working as intended,    
  
If I uncheck everything in Preferences, Classification and add a circle in MeerK40t, it does not classify it, as I expect it should not.  If I then click any of the colors in the statusbar, either stroke or fill, it will create Image, Raster, Engrave, and Cut Operations in the tree and depending on the color I select, may classify it.  Not exactly what I would expect since I turned all the classifications options off.   Similar thing happens if I change the color from the Properties dialog. I also wonder why it creates other blank operations when only one may be needed. If I change the stroke to red I would expect it (if I had some Classification option set) to only create the Cut Operation, not Image, Raster and Engrave as well.  I am not sure exactly  how "Default Operation Empty", "Assign to default operations", and "Autogenerate Operations" In Preferences work with respect to this but I would expect it to not make any Operations or classify anything if they are all unchecked.
  
This is difficult to describe.  In the Properties dialog, if the object does NOT have a fill or stroke color assigned to it and you select one of the colors for it, it will not classify it.  For instance, if a circle has a blue fill but no stroke and you click on the red stroke color, the stroke color changes to red but it is NOT classified.  If you click on the red stroke color again, it will classify.  If the stroke (or fill) already has a color assigned and you click on a color, either same or different, it will classify it. Basically it will not classify the stroke or fill unless it already has a color assigned to it.   This only appears to occur in the Properties dialog and not if you chose the color from the statusbar.  This is also with the "Immediately classify after colour change" option ticked in Properties.
 
Changing the color to "lime" for either stroke or fill does not create the Hatch Operation or classify it as a Hatch.  If the Hatch operation exists (Appended manually), and the stroke color is changed to "lime", it will classify it as a Hatch.  It will not classify it as Hatch if the fill color is change to "lime".

RuKeBo, 2022-09-22:
When I add a Hatch or Dots Operation, it shows up in the tree with (stop) at the end.  If I go into the Properties for it, it is not checked.  If I check "Stop" then un-check it, the (stop) goes away.

jpirnay, 2022-09-22:
> Changing the color to "lime" for either stroke or fill does not create the Hatch Operation or classify it as a Hatch. If the Hatch operation exists (Appended manually), and the stroke color is [...]

Neither Hatch nor Dots operations are created by default, as they overlap with the classical engrave operation too much. So you need to manually create such ops

RuKeBo, 2022-09-22:
> * [x]  This is difficult to describe.  In the Properties dialog, if the object does NOT have a fill or stroke color assigned to it and you select one of the colors for it, it will not classify it.  For instance, if a circle has a blue fill but no stroke and you click on the red stroke color, the stroke color changes to red but it is NOT classified.  If you click on the red stroke color again, it will classify.  If the stroke (or fill) already has a color assigned and you click on a color, either same or different, it will classify it. Basically it will not classify the stroke or fill unless it already has a color assigned to it.   This only appears to occur in the Properties dialog and not if you chose the color from the statusbar.  This is also with the "Immediately classify after colour change" option ticked in Properties.
> 
I see this has been worked on and I don't want to jump the gun, just letting you know as I test it.  
When the stroke color in the Properties panel is selected, it now classifies whether the object had a stroke color already or not.  This seems like it was revised and working correctly.  
  
I thought this was not working correctly but playing with it I suppose it is. The title "Default operation Empty" doesn't seem to explain what occurs, or at least it wasn't intuitive, but not sure there is a better wording.  I assumed that un-checking it would not create Operations in some way since the other options do this, they allow Operations to be created when checked and don't when not.  Since this confused me so much, but maybe I am the only one, I would say that maybe this should be revised in some way.  It just took me a long time to figure out what it was doing, maybe I am in the minority here.  Maybe it should say "Create Default Operations", that's probably not good either.  Something for thought.
    
Having "Default operation Empty", "Assign to default operations", and Autogenerate operations" checked and have no Operations in the tree, have a circle object, then select a red stroke, it will make a Cut Operation, and ONLY a cut Operation, and classify the circle (ellipse) as a Cut operation.  If the circle has a stroke and I select a black fill color, it does NOT create a Raster Operation or classify the fill as Raster if there was no pre-existing Raster Operation.  If there is no stroke color, it will create the Raster Operations and classify the black fill as Raster.  If the Raster Operation already exists and there is a stroke color, clicking on the black fill will classify it as a Raster.  I assume you can't really Engrave a fill so a blue fill does not classify in any case.  Also interesting is that if I remove all the Operations and have a circle with a read stroke and black fill then click on the black color fill in the statusbar, a Cut Operation is created and the circle classified there but the Raster Operation is not created and thus nothing is classified in it. I get the same result doing this through the Properties panel.

tatarize, 2022-09-22:
I'm still working my way through all the old crash reports. We'll probably release Beta20 after that which hopefully should need no more updates. 

Some of them are for 0.7.8 or so which ironically might end up getting a full version sooner than 0.8.1 since it's pure bug fixes and needs some updates for py3.10 wx4.1+.

---

Default operation empty is a bit cryptic. Basically if you have empty ops and it sees this it would sometimes give you a Red/Blue/Other arrangement of ops which might have been what you wanted there. But, since you could have requested nothing with `new` it was decided to give you an option as to whether to fill in blank operations with our sort of blanket default operations.

RuKeBo, 2022-09-22:
Selecting the white color from object Properties displays as Stroke(white=#ffffff) but the indicated color box selected is the "X" (none)...the box the blue line under it is none, I'm thinking it should be the white box.

tatarize, 2022-09-22:
The red X thing was because the box was supposed to be "transparent" or no-stroke if it failed to properly convey that it's a problem (a minor cosmetic one, but still a problem). You can tell it to use white but usually with console or custom colors. Whites area  bit of a pain. Valid colors but hard to see on white backgrounds.

tatarize, 2022-09-22:
Apparently no, there's a legit white color that does white. Ah, I see, it's underlying the transparent. Yeah, that's wrong.

tatarize, 2022-09-23:
Okay, finally finished up the entire backlog of crash logs, fixed everything it had for me.

I need testing and yeah/nah for actually releasing a full version.

RuKeBo, 2022-09-23:
Pressing the <Delete> key while an Operation is selected that contains one or more Elements deletes the Element(s), it does not delete the Operation fully as I would expect.  If you want to remove the Operation and all Elements using the <Delete> key, you need to select the Operation, press <Delete> to remove the Elements then press <Delete> again to remove the Operation.   you could also select each Element and delete them individually. 
  
Cool safety feature or inconsistent?  You can right-click on the Operation to delete everything.  I just expected that if I select an Element in the Operation and press the <Delete> key it would delete that Element and if I selected the Operation and press the <Delete> key it would remove the Operation.

RuKeBo, 2022-09-23:
I need to take this one at a time as there are so many options it is confusing what is affecting what.  
  
Since I newly discovered an option for "Default" in the Operation Properties, I think I have a slightly better understanding, or at least different view, on the classification process.  I think the terminology used throughout needs some tweaking though.  
  
I have "Default Operation Empty" and "Classify after color-change" enabled only. I created Two Raster operations, one I set to "Default", the other I changed to the color Magenta, all other settings unchanged.  If I have a circle element and change the fill to black, does not classify (I think it should classify to the Default Black Raster Operation).  If I change the fill color to Magenta, it will classify to the Magenta Raster Operation.  If I uncheck the "Default" option for the Black Raster Operation and change the fill to black, it will classify to that Operation.  To be clear, I am indicating that I believe that it should classify a black fill to a Default black raster operation.
  
Just another note for this example...if I enable "Assign to default operations" and change the fill color to black, it classifies in the Default Black Raster Operation when it would not before with this option disabled.  This indicates to me that for some reason, the black fill is not classifiable to that Operation, but with the option enabled, it assigns it to the 'default' operation because that is what I think it tries to do by design. One more thing to note, I can never get it to un-classify itself.  If I remove the fill, it stays in that Default Operation.

Sophist-UK, 2022-09-23:
Here is my specially created, go-to file to test classification which I have shared several times previously.
[Element classification test.zip](https://github.com/meerk40t/meerk40t/files/9636311/Element.classification.test.zip)

This file does not contain MK operation details. When I do File New and then load this in Beta 20 I find the following issues:

1. **IT CRASHES!!!!!!** Disappointing to say the least that this happens at this late stage when this file has been available for testing for over a year.
    ```
    MeerK40t crash log. Version: 0.8.0028 Beta20 on Windows: Python 3.7.9: AMD64 - wxPython: 4.1.1 msw (phoenix) wxWidgets 3.1.5
    Traceback (most recent call last):
      File "meerk40t\gui\wxmmain.py", line 2983, in on_drop_file
      File "meerk40t\gui\wxmmain.py", line 2938, in load
      File "meerk40t\core\elements.py", line 9757, in load
      File "meerk40t\core\svg_io.py", line 902, in load
      File "meerk40t\core\svg_io.py", line 496, in process
      File "meerk40t\core\svg_io.py", line 783, in parse
      File "meerk40t\core\svg_io.py", line 792, in parse
      File "meerk40t\core\svg_io.py", line 792, in parse
      File "meerk40t\core\svg_io.py", line 700, in parse
    KeyError: 'dpi'
    ```
    I get the same error when I load Joe's go-to test file also (his LUIGI file). [Basic24.svg](https://user-images.githubusercontent.com/3001893/192034547-10640409-8d4a-4921-86b0-bdb95fcc38a4.svg)

2. This file doesn't crash - but it doesn't give an error either which it should do.
     [MALFORMED.svg](https://user-images.githubusercontent.com/3001893/192033817-74a8094a-9fd3-4e04-bef5-01bda372934b.svg)

Once I can load my go-to classification test file without it crashing, I'll take a look at whether it is classifying in any sensible way (because the last time I looked at this in a 0.8.0 beta the classification of filled paths was broken).

Sophist-UK, 2022-09-23:
> It's probably a safety feature and you can right click and say you want to delete them fully and it'll trash that completely. There's a lot of context sensitivity with regard to the delete key. And it took a lot to get it mostly right.

AFAIK, deleting selected tree nodes is easy - you just need to delete nodes that are selected (and any children) rather than consider deleting emphasised nodes. I had this working perfectly AFAIK in 0.7 - I don't remember seeing any reports where it wasn't working as expected.

[14 more replies]

closed 2022-09-30 by tatarize