meerk40t/meerk40t #1345: [0.8.0Beta21] Issues

Sophist-UK, 2022-09-23:
The latest BETA release of 0.8.0 is Beta 21. This issue is for reporting issues with this specific version of MeerK40t.

Please do all further testing on this release. Please do not report any further bugs with previous beta releases. Do give any other types of feedback, questions, concerns or comments.

Personal note: I admire the optimism expressed in the release notes of this beta: "Hopefully last beta" and "Known errors are down to effectively nothing." I am looking forward to confirming all previously reported issues are fixed and that this version does not have any bugs when I try to use it in production.

Sophist-UK, 2022-09-23:
Reserved for carry forward of previously reported issues.

tatarize, 2022-09-25:
> Insert Special Op / Home Beep Interrupt - I think this should be Origin Beep Interrupt because a home may end up with slightly different home position due to slight randomness in end-stop detection.

Origin didn't really exist when that was added and I don't really know who added it. I don't love that since it's just click a three different things or whatever. I'd probably be more inclined to remove it than swap it out. Also, we did some home repeat tests and it's less random that you'd think. If you want origin, put in an origin. Though with some of the changes in Origin-Coord setup and multiple origins that I am planning on adding in future it might deviate a bit more. Also, it's just a weird setting that does three things you could have done manually. I dunno why it's there.

> * MK SVG Extensions for save / reload do **NOT** follow proper SVG conventions e.g. `<path ... type="elem line">`. Assuming that you need to save `type` (which I am unclear about) it should probably be `meerk40t:type`. Ditto for special attributes in the `<image>` tags.

They save all the types of attributes for a particular element. The image tag had a lot of things that didn't save and recover. The Operations (raster script) needed to be saved for the image, the DPI, the invert, view_invert, dither settings. There were a bunch of them that needed to be saved so I told it to save everything that was an attribute. This is notably somewhat non-standard for the namespace and would be somewhat reasonable as `meerk40t:<attr>` but the namespaces don't really do anything for programs that don't care about them when they load. No program is going to read "type" there and freak out.

There was some discussion about how needed it is for other programs to open meerk40t files, and while they should be SVG files they are somewhat more guidelines than exact 1:1 code objects. That said type could be excluded since it doesn't really do anything. And `Line` as a type of `elem line` should probably save as a `<line/>` type in svg. 

> Some Inkscape attributes are not preserved between load and save e.g. <defs>, <sodipodi:namedview>, <metadata>, some attributes of <svg>, inkscape/sodipodi attributes of elements etc. I appreciate that without understanding/handling e.g. UoM-specific fields in these tags, there might be a problem on reload, but we should try to preserve these so that you can re-edit in e.g. Inkscape a meerk40t saved file without having lost Inkscape specific information.

There's not really any place that stuff would get saved, since there aren't really 1:1 storage of them and there's certainly no editing of that data. We're somewhat moving away from maintaining the objects fully, some of them do save a pile of random `attr` that it found on the object and that stuff will save through, but other things like `group` don't really have much of a home. As things progress there's more of a useful node-tree list for the data structures and less things that can be shoe-horned into SVG. For example, if we added a barcode object, it wouldn't be too hard since there's ( https://github.com/WhyNotHugo/python-barcode ) which is a pretty straight-forward single dependency. We could save the element to disk and reload them, but there wouldn't be any sort of barcode in svg. We could probably give an image with the original text and style of barcode, and enough data to rebuild and edit it in meerk40t and while a nifty idea and might be helpful to people, it wouldn't be svg-ish. The same applies the other way, groups trigger `type="group"` nodes to maintain the structure but there's not really any of those attributes save. SVG is similar in that I think it somewhat corresponds to a "file" like node, but none of those attributes actually preserve. 

> Attributes on saved <text> tags are bonkers - before and after...

They somewhat work which is more than we sort of had before. Some of those values aren't really helpful, other ones are just set there to somewhat get the same sorts of positioning. For example the default baseline is set towards `hanging` because wxPython drew things in the upper-left hand corner and we somewhat needed to replicate that in the file we saved. Yes, some of these properties are clearly not needed. Saving out the whole `settings` thing should probably be prevented since it's what we typically get as attrs anyway. And it's long and fairly useless.

>  Saved file has extremely long lines - can we have image data with a space after each 76 characters link Inkscape, and can we break other long tags into several lines like Inkscape does? It would be ideal if these matched to make it easy to compare pre- and post-saved files, but that is probably going to be too difficult.

I checked into this and it's not really possible without rewriting or writing my own xml save parser. It always HTML'izes the `\n` if you try to add them. It's basically not very feasible to do that. And I guess don't look in those files or whatever. There was some documentation that suggested breaks for the files but the ElementTree just isn't able to do it, and I checked into a bunch of methods to try hacking it in.

> And I apologise for showing up all your "this is now the final beta" optimism.

Pah, don't worry about it. The current beta has a pretty bad bug that breaks things. Position Panel forces some infinite loop stuff, if you basically select an object while it's visible, hell on Raspberry Pi it quickly crashes.  It's fixed for Beta 22, but yeah, absolutely breaking bug. The trickle of little bugs that still exists isn't nothing. I don't expect to find many serious issues left and it should be good enough for a release, but I've had this "this time it's all good to go" parade rained on with some extremely good beta-testing a few times.

Sophist-UK, 2022-09-25:
> No program is going to read "type" there and freak out.

Except some other program that ignores namespacing and uses the type attribute. This is namespace pollution.

> I checked into this and it's not really possible without rewriting or writing my own xml save parser. It always HTML'izes the `\n` if you try to add them. It's basically not very feasible to do [...]

I remember you trying this with \n but I realised yesterday that it was a space rather than a line break and it was the editor that was wrapping the lines nicely because of the space.

> And `Line` as a type of `elem line` should probably save as a `<line/>` type in svg.

Shape elements absolutely should be saved as their native shape and not converted to `<path>`.

RuKeBo, 2022-09-26:
This is sort of strange and again, hard to explain.  This is NOT new to this version. I have "Treat 'Black' as raster", "Classify after color-change", "Don't autoload operations on empty set", and "Autogenerate Operations", enabled in Preferences.    
  
If I have no operations in the tree and add a rectangle element and change fill color to black, it will classify as a Raster (this is working properly).  If I change the stroke color to red, it will not create a new Cut Operation and does not classify it (I think this is not working correctly).  I can also change the stroke to blue and it will not create a new Engrave Operation. (I don't think this is working correctly).
  
If I change the fill color to blue, it stays classified as a Raster, and creates a Cut Operation (because it has a a blue fill and red stroke so I think this is working correctly)  
  
If I then change the fill color back to black, it stays classified as a Raster and keeps it (stroke) classified as a cut operation  (correct again).
  
If I change the fill back to blue and remove the Cut Operation, then change the fill to black, it does not create the Cut Operation or classify as a Cut (it still has a red stroke).  I think this is not working correctly as above.  
  
Going back to the black fill and red stroke that is classified as both a Raster and Cut...If I change the fill to any other color, it will un-classify it as a Raster but keep the Cut.  I believe this is incorrect or inconsistent with the above since without the stroke any fill color classifies as a Raster.

Maybe this is more succinct:  
First, should any color fill be classified as a Raster (with the settings I chose)?   I think this will determine which of the first two below is not working correctly.  

If the element does NOT have a stroke, it will classify ANY fill color as a Raster. 
  
If the element HAS a stroke, it will only classify the color BLACK as a Raster, all other fill colors are NOT classified.  
  
Black (only) filled element does nor create an Operation (Cut or Engrave) when the stroke color (red or blue) is changed.  Other fill colors will.  If the Operation (Cut or Engrave) EXISTS and the fill is black and the stroke color is changed, it will classify the element as either a Cut (red) or Engrave (all other stroke colors).  It just won't CREATE the Operation if the stroke color is changed and the fill color is black.

RuKeBo, 2022-09-26:
I suppose another question is whether "Classify after color-change" should classify everything or only what color changed.  If I change the stroke color, should it classify everything or just the stroke?  That is both a question of what MeerK40t is supposed to do right now and also a question on what the philosophy should actually be.

jpirnay, 2022-09-26:
It will only reclassify the element that was changed

jpirnay, 2022-09-26:
Automatic classification has a lot of advantages, as it will reduce user-interaction (especially for files with a lot of elements). 
On the other hand it has become very simple with some clicks to assign all elements of a given color (either stroke or fill) to an operation manually. As MK stores the assignment this is a one-time only exercise.
As you stated this is a matter of personal preferences.

Sophist-UK, 2022-09-26:
> @Sophist-UK you were on top of the stuff in the past - any comments?

I would need to play with a copy of the file and have a detailed description of which element is classifying incorrectly and how @RuKeBo would like to see it classified. But the tests I did with my specially crafted classification test file and with Joe's mario one, suggest that it is not doing that badly - but there were a few oddities.

jpirnay, 2022-09-26:
Classification on load most probably does not expose that feature as no ops are in existence  at that time. The auto-generation of an operation is not executed if a hit was found. And that could e.g. be a fill hit was found, but none for a stroke. Are we happy that an op raised its hand or doe want a hit on both attributes?

Sophist-UK, 2022-09-26:
From memory I think I came to the conclusion that any element with a fill was a raster element and should not automatically be classified to a new op as a cut/engrave just because it had a stroke set. A red stroke and blue fill in front of a bigger black element should definitely **not** be classified as a vector operation.

jpirnay, 2022-09-26:
Any solid fill will classify into a raster op. But as you can have multiple assignments you could still have cut/engrave in addition. This is currently not working out of the box.

tatarize, 2022-09-26:
Keep in mind when heading down this rabbit hole. There's a reason why I am heavily in favor of making assigning elements to operations manually to be very easy. There are simply a lot of hair splitting and gut-check guesses as to how most people would expect the automatic process to go. If have a lime (#00FF00) vector shape would it get automatically classified into a newly invented Hatch operation since sometimes Hatch default is lime? What if there's already a hatch operation in the operations list? Would you make a different one or put it in the one that already existed, and all the stuff about classifying white filled objects inside rasters in particular orders regardless of the order they appeared in the list. 

At *some* point it must be the case that a judgment call is made and that stuff could go either way. Also, it used to matter a lot more because saving a file wouldn't save the assignments of the which element went where. Now it does so you can assign them and just save the file rather than jerryrigging it to assign these things perfectly or whatever.

---

And while not part of the program proper. This sort of thing really should maybe have some unittests. I do recall a classification change caused a crash the other day. It would be nice to know when that happens. I guess I could write a quick one for just make me a few hundred different object types at random and classify them a few hundred times into different randomish operations in randomish orders.

jpirnay, 2022-09-26:
Still the logic has some merit, so if in doubt offer an option:
[image]
Load of the famous element classification test:
V0.77
[image]

No fuzzy logic:
[image]
Fuzzy 100:
[image]

Fuzzy 200:
[image]

The fact we don't have dots is a different issue, they are just not loaded for the time being as such.

jpirnay, 2022-09-27:
> You could argue there's one engrave with pure black in existence...

And rightly so, it's a slightly transparent black which by definition is not pure black. The complete opacity thing would open another discussion - my take: solid black != opaque black. If you want to have it: that's the reason we have fuzzy

Sophist-UK, 2022-09-27:
The reality is that there are many different ways that paths can be used - combined filled paths using even/odd, layered etc. - that it is likely that you can find two use cases which would ideally require conflicting classification algorithms. In other words an algorithm cannot get it right every time.

The good news is that the algorithm gets it right most times.

That said, I have not yet tested whether a test matrix example will get it right most of the time.

Finally, I counsel against an automatic "if in doubt offer an option" mentality - a proliferation of options to give the most minor of tweaks is not a good way to go IMO. But definitely for major things. And if you **are** going to give an option then:
1. Make sure that the default is going to be what most people will want.
2. Make sure that you give a decent explanation of what the option does - "Fuzzy color-logic - Ticked: Allow a certain colour distance for classification" does not really explain how this works and what impact it has. "Magic-wand - creates magic when you wave it" is clearly not sufficient information to allow you to be Harry Potter.

Sophist-UK, 2022-09-27:
Production burn successfully attempted, however...

* [ ] `Default actions` not executed (when I used the Big Green Button to run the burn).

tatarize, 2022-09-27:
Override rapid movement would require that jogs go at a weird speed. Since it basically forces that stuff into running at a different speed. The twitch is likely that it doesn't force c or whatever I did for twitchless. There's a flag in the speedcode to set that. Worth checking tho.

Sophist-UK, 2022-09-27:
Multiple issues - slow speed, twitches, failing to move, Also only goes horizontally or vertically not by fastest diagonal route!!!! Simply not working!

tatarize, 2022-09-28:
It will not go by the fastest diagonal route because it's not allowed. It' basically *only* in case your cutter isn't allowed to move at a rapid speed. Like if it will fail because it has a rotary and a minimum speed. Given that going at an angle isn't useful, you need to absolutely limit the speeds to the speeds you set. It does not travel faster because the point is that it should go slower. It's working fine.

Sophist-UK, 2022-09-28:
Really? What commands does this issue? Can't you can take the start and end points of the move and calculate the x/y speeds for a diagonal, then calculate the diagonal speed using Pythagoras, and then tell the m2-nano to move there at that speed?

**And it is not working fine - I found *multiple* issues with it. It is unreliable at best if not completely broken and unusable. If you are so ungrateful for my reporting genuine factual issues that you are going to this dismissive then I will stop bothering.**

Plus tooltip needs to have any explanation / constraints as above.

Sophist-UK, 2022-09-28:
P.S. I also think that the default "Minimum Jog Distance" for rapid moves needs to be smaller than the current default (if not zero to be like 0.7). Without adjusting this value, burns with many small travel moves are substantially slower.

I assume that there is some control code reason (switching from one mode to another) to avoid doing this for very small travel moves, however I think the default is 256 - which is c. 1/4" - is too large. Would (say) 50 be a better choice?

RuKeBo, 2022-09-28:
I mentioned this before but think it got lost, or maybe it's not considered an issue.  
If I set my units to "inch" (or anything other than "mm") in Preferences it will change the grid of the bed, the location of elements on the bed, and the size of selected elements on the bed to show in "inch" (or other units).  However, when I open the Properties for the element, the location (X: & Y:) and Size (Width: & Height:) are always shown in "mm".  I can enter values in these input boxes in other units, but they don't show up in the Preferred units and don't stick if you close and open the Properties again.  I'm thinking they should show in the chosen Preferred units but still allow for other units input if desired.

RuKeBo, 2022-09-28:
In the Properties of an element, you can enter a value in the input for the location (X & Y) or size (Width & Height) with no units and it will accept it but converts the value to something very small, and I am not sure of the conversion value.  It appears to be around 2,577.  If I enter "100", it will make the value "0.0388mm".  Is the use of no units meant to be something?  If so, what?  If not, should it not accept the value if you don't use units.  Personally, my default thinking on this is that, from above, the units (and value) displayed should be in the selected units from Preferences and if you don't enter units it should assume the selected units from Preferences.   
  
To be perfectly clear:  If I set the units to "inch", I would see the units in the Properties page in "inches" (and corresponding value in inches) and when I enter a value (number) as an input with no units, it assumes it is the the units selected in Preferences and same as was being displayed.  If I want to, for some reason, enter the value in different units, I can also do that, just like you can now.

tatarize, 2022-09-28:
* The suggestion to fix the white occlusion inside rasters. Give rasters an option to sort the references during cut planning. When we cutplan we sort the references into the same order as those objects occur in the elements. They will always render the way the elements renders solving any occlusion issues. What you see in scene is what you get in the rendered image. Unless you decheck that box.

jpirnay, 2022-09-28:
> 

There are a couple of different units you can use within MK, but all coordinates are stored in an 'universal minimal device unit' so if you wouldn't give any units that's what is used. That may be a bit confusing, admittedly. Will consider it

Sophist-UK, 2022-09-29:
> There are a couple of different units you can use within MK, but all coordinates are stored in an 'universal minimal device unit' so if you wouldn't give any units that's what is used. So to not provide a unit is actually providing one - the internal one.
> That may be a bit confusing, admittedly. We will consider it

This is not a requirement about how MK stores its units internally - that is a choice that is entirely down to developers. This is a requirement about having a **consistent** and **easily understood"" user interface that makes it **easy** for non-technical, non-development **users** to achieve what they want to do easily and quickly, without needing to ask questions of the developers, raise Github issues, refer to extensive documentation, poke around changing obscure preferences or do trial and error experiments. In this instance, what is needed in the UI is for the default units to be both shown and used by default wherever measurements are given, with the ability for the user to override those units with something different. The conversion of UoM to/from whatever units are used internally is a simple matter of coding.

So in Inkscape in Document Properties, you can define the default UoM for the document (in my case mm) but in the position and size boxes on the toolbar (or the size boxes for rectangle or ellipses) you can select a different UoM (say px or inches) if you so wish.

P.S. I am **not** saying that delivering a good UI is easy - it isn't, and Inkscape also has it's own UI issues (e.g. with their object tree). However once again it appears that I need to suggest that developers should actually listen to what users (@RuKeBo in this case) are saying about usability and not be so dismissive of this extremely useful feedback that they are getting about things not working well.

RuKeBo, 2022-09-29:
When I clear the Width or Height (in a certain way) of an object it sort of goes haywire.  This happened to me before but I could never reproduce it, it just seemed to happen randomly and maybe this is unrelated to that time, I can reproduce this.  If I add a Rectangle and then try and remove the value, clearing the input box, for Width or Height, it complains, changes the input box to red, and waits for me to enter something sensible.  If I change the Width to something large, like 22.0in, sometimes it moves the rectangle far off the bed to the right and does make the width 22in.  If I then put the cursor in the Width box at the start, before the "22" then press the Delete key two times, the width then becomes "0.0in" and then I cannot change the width further, I cannot change the width to anything.  The bounding box (if that is what is is called) is also detached from the rectangle (see image below).  I can also cause this by entering "0in" in the Width or Height. I can't seem to cause it if I try and use delete to delete the whole number and the Width or Height has a decimal value, it won't let me delete the decimal point or number past that to the right.  
  
The issue also occurs if I delete the decimal number alone.  If I have a Width or Height of say "2.5in" and put my cursor to the left of the "5", then press the delete key to delete it, the number goes to "0.0in" and everything I mentioned above happens (bounding box detaches, width is zero, can't select object directly, can't change the number any longer).

[image]

If I do this with a circle object, it seems to trash the bounding box, I can't see it and I can no longer select the circle.  It will also throw an error (below) when I try and close the Properties dialog.  When I go back into the properties (from the element in the tree) it does not show the Location or Size.  
    
[image]

MeerK40t crash log. Version: 0.8.0030 Beta22 git on Windows: Python 3.9.5: AMD64 - wxPython: 4.1.1 msw (phoenix) wxWidgets 3.1.5
Traceback (most recent call last):
  File "C:\Users\rborn\Documents\GitHub\meerk40t\meerk40t\gui\wxutils.py", line 537, in on_leave_field
    self._action_routine()
  File "C:\Users\rborn\Documents\GitHub\meerk40t\meerk40t\gui\propertypanels\attributes.py", line 496, in on_text_h_enter
    self.scale_it()
  File "C:\Users\rborn\Documents\GitHub\meerk40t\meerk40t\gui\propertypanels\attributes.py", line 466, in scale_it
    if bb[2] != bb[0]:
TypeError: 'NoneType' object is not subscriptable

<_io.TextIOWrapper name='MeerK40t-2022-09-29_10_38_09.txt' mode='w' encoding='cp1252'>

RuKeBo, 2022-09-29:
I'm not saying this is a problem but curious if it should do this.  If I put in a 2inx2in Rectangle at a location of X:1in & Y:1in then change the Height to 12in (by simply adding a "1" before the "2in", it will move the Rectangle to X:1in & Y:6in.  I'm just giving all the numbers in case there is some mathematical reason it moves as far as it does depending on the dimensions and/or location. If I changed the Height to "22in" (adding a "2" before the "2in"), it moved it to X:1in & Y:11in. There are times I can change the location and it will keep the object in place, not move it, just change the width or height.  I am having a hard time figuring out when it does each. My preference, or thinking, is that the object should stay where it is, but maybe there is a reason it doesn't.

RuKeBo, 2022-09-29:
In the "Object Dimensions" pane there is what appears to be a button called "Resize the object".  It has a tooltip and highlights if you hover over it. This button doesn't seem to do anything.  I am assuming that at one time you would change the Width and/or height values then had to press this button to enact the changes.  Currently, the Width and Height are changed on the fly.  I don't really have a preference at this time although it does appear you can enter some strange characters, like two decimal points by accident, and it will apparently make that zero but this (two decimal points) is not destructive like entering some other characters can be.   I'm just assuming that if the Width & Height change things on the fly, the button does not need to be displayed.

RuKeBo, 2022-09-29:
I'm not trying to get a head of you, just noting this so I don't forget later...Looks like the issue above with entering "0" has been fixed.  When I try to edit/change the decimal of the number, the entire value goes to zero.  This does not break it, you can just enter a number back in and it will update accordingly.  For instance:  If I have a Width or Height of "2.5in", put my cursor to the left of the "5", press the delete key, the value goes to "0.0in".  Interestingly, if I put the Lock on and do the same thing, the other locked value (Height if I am editing the Width and vice versa) changes to "0.0in", there is no update, and the value I am edited waits for me to put a number in.  If I have a rectangle of Width 3.5in and Height of 3.5in and remove the "5" in the Width, the Height changes to "0.0in" but when I type a "2" so that now the Width is "3.2in", the Height changes to "3.2in" and the object updates accordingly. Aside from the Height changing to "0.0in" in the process, this seems appropriate to me.  It holds on to the change until I enter the digit.  No idea how or why it does it when locked but it feels correct and maybe with it unlocked it should do the same.  To be clear, this does NOT break anything.  It's a little annoying since if you remove the decimal number, you need to type the whole thing over again since it zeros out.  You can put the new decimal number(s) in first then delete the old numbers, that will work, since it is never without a number to the right of the decimal point. Does same circle.

RuKeBo, 2022-09-29:
> * MK does apply all changes immediately _if you wish_, so any keystroke will lead to an update. If you don't need that immediate reaction feature, goto preferences and turn it off

My GUI Preferences does not look like yours.  I do not have "Misc." or "Scene"
[image]

tatarize, 2022-09-30:
> I assume that there is some control code reason (switching from one mode to another) to avoid doing this for very small travel moves, however I think the default is 256 - which is c. 1/4" - is too large. Would (say) 50 be a better choice?

Ironically there is... it was all those damned twitches. Which no longer exist.

closed 2022-10-19 by tatarize