meerk40t/meerk40t #1347: [Suggestion]: Multi Pane Configs / Views

will-janz, 2022-09-24:
### Summary Description

The customization of the UI is _fantastic_, but everything can be improved. I've found myself wanting a design view (something without machine controls and such), and a machine view (machine buttons, job buttons, etc.)

### Current functionality

Currently panes can be configured in one view.

### Future functionality

Either along the top, controlled by tabs (hard coded (Design, Machine, etc.) or user definable with an add button) or a large dropdown always present in the ribbon. Default views can be applied.

Switching between views would switch between saved pane layouts. If I have time I'll create some mockups.

### MeerK40t Version

v0.8.0027 Beta19

### MeerK40t Type

Executable

### Your Operating System

Windows

tatarize, 2022-09-24:
There has been a request or two for Design Mode. It seemed like it would be a bit extensive to make sure all the regular widgets hide and the design useful stuff shows up. It wouldn't be too hard to swap perspectives though, the aui thing for the panes lets you quickly set a different perspective without too much trouble. Having alternative sets of panes wouldn't be difficult at all. In fact, you can hide panes and view them again with `pane hide <pane>` and `pane show <pane>` swapping things isn't a problem. In fact, `reset panes` in the menu actually resets the panes to the default pane view which it saved during startup before applying the saved perspective.

tatarize, 2022-09-24:
That said, my current plan is to extend not the panes but the scene itself. There aren't that many design relevant things in the any of the panes. There's plenty of them and more can be easily added to do a bunch of other stuff but they aren't that many that are purely design relevant. But, within the scene it would be possible to add a few different viewing modes and have different tools and draw the scenes in different ways based on that selection.

The scene itself is wildly configurable and quite dynamic. If you want to see an example of this type `cyclocycloid` in console. Or try `tool ribbon` both of which are generally there to show off the scene and functionality therein. The utility of the scene expansion I sort of want to work on is that we could more easily set and view the different coordinate systems. Namely MeerK40t uses absolute coordinates. Where everything is relative to the bed and objects are placed in the bed. You tell it to move to a position and it'll go there within the bed. However, other laser programs use other coordinate systems. Namely, relative coordinates and origin point coords. Where we actually get our projects' bounding box and a mark placed in the corner of the there. Then the mark in the corner corresponds to the bed location of the project and it's set at an absolute location within the scene, either the current position or the origin position. It usually gets placed in nine different locations or so.

Well, there's some desire to get multiple origins to work. So you can flag 5 different locations, and tell the program to go to each of those five locations and then perform the same project relative to those five positions. To get this to look good and not be weird it would need to let you set this information and draw the project five times in those different locations. This isn't very hard from a scene drawing thing. You'd basically set your base coordinate system and then from there be able to run jobs within that coordinate system. In theory simulation could be built into this without too much issue. And in that system you could absolutely set a design mode. There's also some desire to have some quick drawing menus like pie-menus that could load up and give you the different tools you might need so it seems like a whole design mode would be best done along with that work.

---

Though just the pane stuff it's not that weird to want to save a pane configurations and load pane configurations and it's just a few lines of code, so regardless of the whole scene work code, I'll be sure to do `pane load <name>` and `pane save<name>`

tatarize, 2022-09-24:
https://github.com/meerk40t/meerk40t/pull/1346/commits/457d0daeaa600aa6c812ea004b12cb21a0a42a4a

Added to [0.8.x-bug-fixes](https://github.com/meerk40t/meerk40t/tree/0.8.x-bug-fixes)

Do `pane save design` and then `pane load design` and you'll be able to quickly change perspectives.

There's also a helpful command for `pane float -a <panel>` which will fully float a pane where it won't try to dock again. So `pane float -a laser` will float the laser pane and leave it floating like that.

Sophist-UK, 2022-09-24:
I have also previously suggested a UI improvement to match workflow stages:

* Design - focused on drawing - current scene plus Tools ribbon-bar for drawing tools - move Tools tab to first position and rename it Design, and have a separate choice of panes for this tab.
* Planning - focused on moving elements and operations - current scene and Home ribbon-bar - just move the Home tab and rename to "Planning", and have a separate choice of panes for this tab.
* Execution - New tab, change the scene to the one in the Simulation window, and have a separate choice of panes for this tab (including a new pane with the other controls on the simulation window).

We might want to make some panes only available on certain tabs.

Also allow multiple "projects" to be loaded simultaneously and switch between them. Thus you could be burning one project whilst designing or laying out the next one. 

If you like this idea please do a +1 to show your support.

tatarize, 2022-09-24:
The current plan would be less tabs and more somewhat specific mode selections.

[image]

See Blender prior art for how this tends to work. You get a mode change thing, and could go to drawing mode and see all the relevant tools, which would just be drawn in the scene. You'd be able to also do some things like making multiple-origin points viewable as such or give you a different view if you are set for relative view.

It wouldn't be so much as tabs for the pane but different views in the scene where you could select a different couple modes. Which would basically just repurpose the scene based on the workflow part. If you're in design then you get the design tools and can edit things etc. If you're in planning you'd get to see the different ways to fire off this project based on coord system and might give you several of the functions currently inside LaserPane for you to do job laying out. You'd then also have simulation which would basically be a lot like simulator, but mostly built into the scene directly. So time estimates in the scene, the device name (which would at that point be locked), ability to animate what it would try to burn etc. And execution which would in theory maybe let you see (if possible) some visualizations on job progress, this last one is not very well understood here.

The ease here is that with different aspects of the scene views we could just set the various aspects and, in theory, just not need the panes at all for most tasks. Now, obviously I wouldn't be making a bunch of move-to this location combo boxes, in scene to replace the stuff we already have, but the workflow would be that we'd want to actually be able to do all of our work. 

---

I am less inclined towards the multiple projects simultaneously bit. Though in that case the correct methodology would be to replace the scene with a notebook and actively allow different panes. I did some tests with this and it does sort of work. But, I had to use the panes to swap out the entire `elements` service. So you switch tabs and you got an entirely different `elements` object that swapped out. This meant the tree rebuilt for the other loaded project and scene drew differently as such. It wasn't terrible but it was a lot of somewhat weird things happening there. The coding infrastructure worked fairly well, and it had a lot of utility towards diagnosing coupling within the program. This would lend itself towards the whole multiple-projects methodologies. So "new" would actually launch a new tab. And you'd need to close the previous project. And it would work a bit like the whole multiple different documents loading in a text editor thing. It's a pretty starkly different architecture. And while I'm disinclined towards actually doing this, I would like to make sure MeerK40t's code is good enough to generally allow this.  --- If you really want to do two projects like this at the same time, it would be much more reasonable to launch two copies of meerk40t (the `-p <profile>` flag will let you not mess with the other settings). And you could already sort of run a project then clear out the scene and start messing with another project if you wanted. The stuff sent to the spooler will just remain there.

Sophist-UK, 2022-09-25:
I only have one laser, and I use Inkscape for drawing rather than MK, and I only occasionally need to have two projects open simultaneously, so this requirement is not something I personally need.

### BUT... 
* As you add new functionality for drawing and tweaking designs, there will be a greater need for supporting multiple projects
* Anyone with two or more lasers they want to run simultaneously may prefer multiple simultaneous projects to multiple simultaneous instances (i.e. so they can easily pick and choose which laser to send the current job to rather than have to choose the instance / laser to load the project into.
* If the vision is to be a better tool than Lightburn, this is probably needed to achieve that.

tatarize, 2022-09-26:
* I don't really think multiple projects open at the same time is going to be needed. I do think I should test things to make sure they are suitably decoupled and that actually does that quite well.
* The typical workflow that I've heard of is sending the same job to multiple lasers. People don't tend to send two jobs to to different lasers at the same time, and usually the time to send the job is minor and they can load, send, load, send. I'd need to see version of this workflow live.
* The vision is to be a very good tool, and a highly functional and hackable platform for lasering as well as sort of programming for lasers. That or being great. I think there are a few other fundamental changes that would be more useful before this which I believe may be somewhat of a lateral move. I don't think one project at a time is going to be an albatross around our neck for future expansion.

That said I do think some features like different coordinate systems are remarkably important and highly under appreciated by the current state of meerk40t.

Sophist-UK, 2022-09-26:
In the end neither of us really have an understanding of how power users will use MK. You probably need to find some users who have multiple lasers and see how they would like software to work (whether they currently use MK or Lightburn).

closed 2024-10-04 by jpirnay