meerk40t/meerk40t #1351: [Request] Tree context menu for element nodes to Unclassify emphasised nodes

Sophist-UK, 2022-09-27:
Rather than go to the (multiple) Ops and find and remove element references from non-hierarchical lists, it would be easier to select element nodes and right-click to unclassify and remove from operations.

jpirnay, 2022-09-27:
[image]

Sophist-UK, 2022-09-28:
I hadn't spotted that (perhaps because it is new)!! 😃

The language is inconsistent and IMO a little confusing (despite me being relatively knowledgeable about MK). In preferences we talk about classification, here we talk about assignment and references. And even I am unsure what "Inherit" means in this context?

I am not personally wedded (not that this matters since I am no longer a developer and was never Tat) to "Classify" / "Classification"; "Assign" / "Assignment" is equally good, or perhaps "Classification" is the name of automated "Assignment". It might be better to drop one and replace "Classification" with "Automated assignment".

So in preferences, it might be better to replace "Classify" with "Assign" in labels and tooltips, add a tooltip to the tab itself.

In the tree context menu I definitely think that "Remove all references from all ops" should be "Unassign from all ops".

And my guess about "Inherit stroke and classify similar" is that it means "Assign automatically based on Stroke colour"; and ditto for fill. Perhaps this description would be more consistent terminology. 😄

**Also,** the assignment menu needs to be added to groups, files and the Elements node.

jpirnay, 2022-09-28:
> Also, the assignment menu needs to be added to groups, files and the Elements node.
Added files and groups. Overlooked the elements idea

jpirnay, 2022-09-28:
My (personal) definition:
Classfication is the (automatic) process of assigning an element to an operation. That link between element and operation is called an assignment.

Sophist-UK, 2022-09-29:
Yes - that is my definition also, but if you use "Classification" to mean "Automatic assignment", then you need to explain this in a tooltip or text label so that everyone else can understand it too, and it may simply be easier to use "Automated assignment".

That said, there are also tree context menu items (on the Operations node) which talk about reclassifying, so there are probably more places than just Preferences where this is used.

jpirnay, 2022-09-30:
I will look into the different uses of the terms

closed 2022-11-09 by jpirnay