meerk40t/meerk40t #1355: [0.8Beta23]

jpirnay, 2022-09-30:
The latest BETA release of 0.8.0 is Beta 23. This issue is for reporting issues with this specific version of MeerK40t.

Please do all further testing on this release. Please do not report any further bugs with previous beta releases. Do give any other types of feedback, questions, concerns or comments.

RuKeBo, 2022-09-30:
Can you explain why the GUI Preferences panel I have does not match per the Beta22 Issues?  Just wondering if this is a problem or not.

RuKeBo, 2022-09-30:
I think things are pretty solid and I am trying to go through every option to test. This might be nit-picky but I think there are too many "Tools" labels.  I see the Tools menu, Tools option was renamed to "Laser-Tools "so that seems better, one less anyway.  In my GUI, the Tools menu option is right above the Tools ribbon tab and I sometimes click the Tools menu when I want to use the Tools ribbon.  And honestly, these Tools have nothing to do with each other, the Tools menu is not a duplicate of the Tools ribbon.  I think the Tools ribbon is correctly named, those are, by all accounts, Tools for drawing and object manipulation.  I do not have a solution for this and I expect this might evolve over time anyway, from what I hear, and be something different.  If the menu item makes sense to be called "Tools", maybe the Tool Ribbon could be named "Toolbox"?  Just an observation.

jpirnay, 2022-09-30:
You may be right here, that's a thing to change. Although as someone doing woodworking: you can never have enough tools ;-)

RuKeBo, 2022-09-30:
Hmmm, true.  I also found one under "Panes" , "Panes, Tools".

RuKeBo, 2022-10-01:
This doesn't seem to be new but when I "Define Custom Colors" then "Add to Custom Colors" in the Colors Preferences, it doesn't save it. It does add it temporarily and does use the color for the appropriate option, but when you go back in it is not there.

tatarize, 2022-10-02:
Its also different by os. We'd need to write our own dialog to really save.

closed 2022-10-09 by tatarize