meerk40t/meerk40t #1361: [0.8Beta24] Issues

Sophist-UK, 2022-10-02:
The latest BETA release of 0.8.0 is Beta 24. This issue is for reporting issues with this specific version of MeerK40t.

Please do all further testing on this release. Please do not report any further bugs with previous beta releases. Do give any other types of feedback, questions, concerns or comments.

Sophist-UK, 2022-10-02:
This comment left to summarise outstanding issues from previous betas.

Sophist-UK, 2022-10-02:
* [ ] Origin auto op still not working.

Sophist-UK, 2022-10-03:
* [ ] Cut Operation-passes > 1 do not simulate or burn more than one pass!!! I reported this in Beta 17 and it is still not fixed!!

Sophist-UK, 2022-10-03:
When I have the Cut op[ with 4 passes, it doesn't do this. I think I loaded it from a file with Ops saved from MK 0.7.

jpirnay, 2022-10-03:
Might be the same issue as in https://github.com/meerk40t/meerk40t/issues/1365. I could not validate the 'does not burn' part of it, as I am away from my laser

tatarize, 2022-10-04:
- [ ] Zdepth has some crash thing that is unhelpful.
- [ ] If you run a balor light job and you interrupt it with a full job, it deletes both. This is because it sorts then removes position 0, which is now the full job you sent. The original job is stopped and removes itself normally. This may relate to the weird need for try, except blocks inside the sync blocks inside spooler there. That stuff honestly can't remove except within that class and a single thread so how does it mess that up?

Sophist-UK, 2022-10-04:
> Might be the same issue as in #1365. I could not validate the 'does not burn' part of it, as I am away from my laser

1. #1365 happens when Operation-Passes = 1 and also happens in 0.7 where multiple Operation-Passes works fine. So I don't think it is the same issue at all. #1365 is **not** IMO a bug because it is correctly displaying the element stroke (it is just unhelpful to the user).
2. I did a burn and it only burned one pass. Definitely. So simulation and real burns have the same issue (not that surprising since the simulation runs the same cut planner).

Sophist-UK, 2022-10-08:
* [ ] Text kerning is not quite correct

    **Inkscape**
    [image]

    **MK 0.8.0b 24**
    [image]

[Slot sizer](https://user-images.githubusercontent.com/3001893/194705790-81184260-cec2-4c46-928b-eb656393de9b.svg)

Sophist-UK, 2022-10-09:
* [ ] Rapid jog is not working at all!!
    [image]

Sophist-UK, 2022-10-09:
> Can you please provide more information how you faced this issue, not able to reproduce with a simple test file, but maybe a more complex example might demonstrate it

I opened MK 0.8.0b24, file/new, went to design and created a red circle. Set op passes to 4. Clicked simulate, it did one pass.

tatarize, 2022-10-09:
The failure is for Merge Operations and not Merge Passes. If you have Merge Operations dechecked then it runs them all, if you have merge passes checked then it runs all of them. It's only No-Merge-Passes / Yes-Merge-Operations. Basically in that case I think we do one pass completely merged and the remaining passes as a different operation and that part isn't run correctly. Though that might be merging 1 pass when you were asked not to merge any passes.

I think that's the issue, that set of settings falls into a donut hole, it merges but not with passes. I'm not strictly sure what it should do there except that it shouldn't utterly fail. I tracked this bug before to it writing a 1 to the levels and then failed to finish the levels. I'm not really sure what merged-operations without merged-passes in the case that you had some operations that couldn't merge because they had passes would look like.

It might be correct to just filter off the operations that require multiple passes and not merge them at all. So exclude the merging of operations with multiple passes.

tatarize, 2022-10-09:
I'm pretty sure when I tried to tackle this before I just rewrote it to be a lot more obvious with `_should_merge` and simplifying the logic. But failed to simplify the logic. I think we should not merge objects with disharmonious pass levels. It's not an ideal solution but it will prevent things from messing up. Checking merge ops allows merging it doesn't require it.

tatarize, 2022-10-09:
Oh, yeah, the actual bug is a bit more complex than that. When it converts the op to `as_cutobjects` it passes the `op.settings` which is what it's required to do. But the code then sets the passes to equal 1 since that's what it was passed. But, that value sets the `op.settings` to 1 which then modifies the entire dictionary. The point is that the settings object *should* be the same between the operation and all cut operations it gave rise to because otherwise we can't tell if something changed as easily. If we have the same dictionary we both save memory and know for a fact that everything is the same.

tatarize, 2022-10-09:
> * [ ]  Text kerning is not quite correct

We have no access to kerning (currently) and as such cannot fix that.

tatarize, 2022-10-09:
https://github.com/meerk40t/meerk40t/pull/1369/commits/73a09e38075e2891b6f9203a58b005ec7cf818c1

Fixes this. Creates a new settings dictionary when something needs to modify it. Worked fine. First pass does everything. Second pass does the next things. etc. Merges the operations within the first pass and doesn't clip the rest of them weirdly.

closed 2022-10-19 by tatarize