meerk40t/meerk40t #1362: [0.7.8] Circles converted in Inkscape to combined paths burn weirdly

Sophist-UK, 2022-10-02:
A row of stitch holes (which started as circles in Inkscape and were combined into a path - without being separately converted in Inkscape to paths first) is burned as a series of 90degree segments burned in wrong order / wrong direction.

[image]

Here is the SVG file [Circles screwed up](https://user-images.githubusercontent.com/3001893/193468127-11397373-5ade-4e96-a623-c9994bd84071.svg)

tatarize, 2022-10-07:
[image]

Minimized out it's that there's a matrix which multiplies by -1, -1 for width and scale. I've seen a similar bug before but I thought I had fixed it. But, apparently not given this example.

tatarize, 2022-10-08:
meerk40t/svgelements#50

Somehow this is flipping the arc and sweeps flags, I'll add a test for it at the minimum.

tatarize, 2022-10-08:
```python
    def test_issue_mk_1362(self):
        circle = Circle(cx="-1.5", cy="-1.5", r="0.5")
        circle *= "matrix(0,-1,-1,0,0,0)"
        path = Path(circle)
        path_d = path.d()
        circle_d = circle.d()
        self.assertEqual(path_d, circle_d)
```
This test ultimately fails.

Expected :M 1.5,1 A 0.5,0.5 -90 0,1 1,1.5 A 0.5,0.5 -90 0,1 1.5,2 A 0.5,0.5 -90 0,1 2,1.5 A 0.5,0.5 -90 0,1 1.5,1 Z
Actual   :M 1.5,1 A 0.5,0.5 -90 0,1 2,1.5 A 0.5,0.5 -90 0,1 1.5,2 A 0.5,0.5 -90 0,1 1,1.5 A 0.5,0.5 -90 0,1 1.5,1 Z

Basically the path_d goes around clockwise and the real circle goes around counter clockwise, but with the correct arcs. So the we end up creating scooped arcs for all the parts of the circle, which then go even more haywire later when converted for some still undiagnosed reason.

tatarize, 2022-10-08:
Okay, fixed. Sort of. I think it could still go clockwise vs. counter-clockwise.

But, the rotated and flipped value, are such that the sweep should invert, but the criteria was wrong.

[image]

tatarize, 2022-10-15:
The bug was in svgelements so it's fixed for 0.7.8 too as is marked in this version. But, was a problem in 0.8.x since both use the same main code.

tatarize, 2022-10-18:
Fixed in main.

closed 2022-10-18 by tatarize