meerk40t/meerk40t #1365: Simulation scene burns not shown for vector ops for paths with fill and no stroke (manually assigned)

Sophist-UK, 2022-10-03:
I have some SVG files from a vendor (snowflakes) which are a single combined black-filled no-stroke path.

As you might imagine, these are (correctly) auto-classified as raster elements when they are loaded.

So I drag it to a Cut operation and look at the simulation and the burns are not shown (though when I try an execute the laser does turn on).

So I think we need a tweak to the simulation display so that all vector operations are shown even those with no-stroke or white-stroke or transparent stroke.

Sophist-UK, 2022-10-03:
Make an SVG with a path with black fill and no stroke.

Open it in MK. Manually move the element from a Raster op to a Cut op. Open Simulation. You don't see the path (invisible in the scene) but the path is burned if you animate the simulation scene or if you do a real burn. So the path is converted to burn cutcode it is just not shown in the scene.

tatarize, 2022-10-09:
Fixed in the latest bug checks.

closed 2022-10-24 by tatarize