meerk40t/meerk40t #1371: [Bug]:

wenchung01, 2022-10-08:
### Summary Description

Hi, I use MeerK40t : RuidaControl (Run my M2 Nano with Lightburn) and I succeeded. 
But I click (continuous jog ) and move, I can't stop moving!
My MeerK40t : console shows  Busy Error again and again! !

[image]

### MeerK40t Version

v0.7.8

### MeerK40t Type

Executable

### Your Operating System

Windows

tatarize, 2022-10-08:
I tested this and it seemed to work. 

https://user-images.githubusercontent.com/3302478/194724013-b059c4b2-a372-451f-9122-2c394da9c3d8.mp4

tatarize, 2022-10-08:
You could try `alias default` to reset the aliases. Also, you can see them in the console there. They do the same thing as the WASD keys for key down, etc.

wenchung01, 2022-10-10:
thank you !! I try

wenchung01, 2022-10-13:
I can't find the problem, he still doesn't work, and the crash ... >_<"
https://user-images.githubusercontent.com/115357635/195503251-181b0a30-c3c3-4896-9092-c62750b6ca53.mp4

wenchung01, 2022-10-13:
I can't find the problem, he still doesn't work, I got crash ... >_<"

https://user-images.githubusercontent.com/115357635/195507352-75ac90ea-ac81-4f8c-92b7-743a05719c0c.mp4

wenchung01, 2022-10-13:
I need to turn off the machine every time, I think....I can't click (continuous jog), only click one-to-one click.

https://user-images.githubusercontent.com/115357635/195509582-bd815ed1-cee7-4295-8488-95164bb6a891.mp4

tatarize, 2022-10-18:
I seem to recall there was an issue with something like this, where the alias was wrong (I checked this doesn't seem to be wrong). This might be fixed in 0.8.1 since it uses a different methodology to move it.

I believe it can be triggered if it loses a packet to do that. And might be better off using the move position directly in lightburn there. Since that's probably a better way of moving the laser position.

tatarize, 2022-10-18:
Yeah, this seems to be generally broken in 0.7.x pressing the button fairly easily seems to get stuck. It sometimes recovers but it takes longer than it should. This might be something weird with the timing since it might not be detecting the keypress correctly since it more or less locks up with the positional spamming there.

tatarize, 2022-10-18:
If you close down the console window it appears to work more consistently. It might be the rate at which it sends data that is largely responsible. It can't process that level of spam in the console window without occupying all of the time for the gui thread which then has no ability to process the key release.

tatarize, 2022-10-19:
https://github.com/meerk40t/meerk40t/commit/a67c273954ccf97605aaba83758209a6302d93b4 should hopefully fix that stuff.

closed 2024-10-04 by jpirnay