meerk40t/meerk40t #1372: [0.8beta25] Issues

Sophist-UK, 2022-10-09:
The latest BETA release of 0.8.0 is Beta 24. This issue is for reporting issues with this specific version of MeerK40t.

Please do all further testing on this release. Please do not report any further bugs with previous beta releases. Do give any other types of feedback, questions, concerns or comments.

Sophist-UK, 2022-10-09:
This comment left for developers to summarised fixed and outstanding issues.

Sophist-UK, 2022-10-10:
* [ ] Travel time estimates still way way way way way way way way out.
     [image]
 * [ ] Simulation playback at 5000% is still painfully slow. 
 * [ ] **Feature request:** I really think that playback "frames per second" needs to be variable depending on burn time i.e. so that playback is nearer real-time. 
 * [ ] **Feature request:** A playback clock showing elapsed time vs. total or remaining time would also be nice (as a future enhancement).

Sophist-UK, 2022-10-10:
* [ ] Position pane - tabbing between fields, sometimes existing values are selected, sometimes not.

Sophist-UK, 2022-10-11:
* [ ] Minor enhancement request - SVG file with ops and no elements - don't create the File node.

Sophist-UK, 2022-10-11:
Time estimate feedback: Cut-only job, estimate cut: 1:54:49, Travel: 1:01:36, history total time: 1:53:53.

So not sure why total time was almost 1m less than cut estimate. Early beeps probably accounted for about 20s of this, but travel time was probably 20s.

As an estimate the cut time is good enough for practical purposes, just unclear why it was as much as 1% out.

Sophist-UK, 2022-10-12:
* [ ] Crash on path Break-Apart:
    ```
    MeerK40t crash log. Version: 0.8.0033 Beta25 on Windows: Python 3.7.9: AMD64 - wxPython: 4.1.1 msw (phoenix) wxWidgets 3.1.5
    Traceback (most recent call last):
      File "meerk40t\gui\wxutils.py", line 150, in specific
      File "meerk40t\core\elements.py", line 8467, in inner
      File "meerk40t\core\elements.py", line 7975, in break_subpath_elem
      File "meerk40t\kernel\context.py", line 33, in __call__
      File "meerk40t\kernel\kernel.py", line 2097, in console
      File "meerk40t\kernel\kernel.py", line 2146, in _console_parse
      File "meerk40t\kernel\functions.py", line 279, in inner
      File "meerk40t\core\elements.py", line 2763, in element_subpath
      File "meerk40t\core\elements.py", line 9149, in classify
    AttributeError: 'Path' object has no attribute 'type'
    ```

jpirnay, 2022-10-12:
>  Position pane - tabbing between fields, sometimes existing values are selected, sometimes not.

That's a bit of a pickle, as the reduction to two decimals will often lead to the impression that there's a change compared to the internal representation. Needs some additional logic at central routines, not something to do lightly at the eleventh hour

Sophist-UK, 2022-10-13:
* [ ] `Use` elements - these are not loaded as I would expect, but instead are loaded inside a (new) group that does NOT exist in the SVG file - can we avoid the extra group node so that the tree matches the SVG file?

Sophist-UK, 2022-10-13:
### General feedback on production readiness and usability
I am now using this beta for production use (despite the bugs that were reported several betas ago). So a big thank you and well done to @tatarize and @jpirnay and @tiger12506 for getting this brand-new version to this point.

I find the support for Text and Use clones particularly useful, and hope to use the Wordlist functionality in due course. (I note that the wordlist window has a `How to use` tab which is excellent design - if this can be extended to other areas that would be great. I have some suggestions for text improvement on the Wordlist help tab which I will put into a PR - an exception because it is text and not code and so not really development.)

I haven't used the image raster functionality in production yet. I remain concerned by the lack of equivalent functionality to the 0.7 Raster Wizard i.e. easy preview in the Image Properties window and lack of a revert, but I will give feedback once I start to utilise this for production use.

Sophist-UK, 2022-10-13:
* [ ] Group inner burns (component functionality) broken when design contains rasters in more than one component because the functionality in 0.7 to make separate raster images is not working in 0.8.

Sophist-UK, 2022-10-13:
* [ ] [snowman laser7](https://user-images.githubusercontent.com/3001893/195609799-0407eee0-b214-4d02-9769-ff32f6aa6ad8.svg) In 0.7 the optimisation only takes a second or so (laser time 6:57, distance 23821.41) but in 0.8 it takes c. 45s (laser time 1:45:44, distance 912894.94)!!!!

tatarize, 2022-10-13:
>> elements - these are not loaded as I would expect, but instead are loaded inside a (new) group that does NOT exist in the SVG file - can we avoid the extra group node so that the tree matches the SVG file?

The objects within the use are not actually in the svg file at all. They are references to other parts and build as sub-elements of the use. These elements are actually within the use like a group with the use as a structural element. In theory we could skip the group within meerk40t, this comes from the interpretation of how the render tree works within the svg spec. Use is like a group of the referenced elements building that tree out.

tatarize, 2022-10-13:
> Group inner burns (component functionality) broken when design contains rasters in more than one component because the functionality in 0.7 to make separate raster images is not working in 0.8.

Porting it over was a problem and broke the import from 7 to 8 since there were files that were moved and the different parts of the algorithm wasn't encapsulated very well. So this was lost between 7 and 8. And I could probably rewrite it. Basically you'd really want to just submit a pixel closeness routine and build multiple rasters images if they do not overlap and their distances are greater than some value (though not strictly based on the raster. There's probably some heavier mechanisms to do clustering of individual raster points. So you could even divide images as such. Though it might require numpy to be fast enough to use which is actually a requirement of meerk40t now (at least generally).

Sophist-UK, 2022-10-13:
> The objects within the use are not actually in the svg file at all. They are references to other parts and build as sub-elements of the use. These elements are actually within the use like a group [...]

AFAIK, an SVG `Use` object is a one-to-one relationship with another object. Thus there is no need to use a parent object (like a Group node) to hold multiple references.

Sophist-UK, 2022-10-14:
* [ ] `Wordlist` Editor button should not IMO be on the `Settings` RibbonBar as words on the wordlist are part of the design / burn process rather than set-and-forget functionality.

Sophist-UK, 2022-10-14:
* [ ] Time estimates vary between simulation and operations:
    **Simulation**
    [image]
    **Tools/Operations/Operations-Information**
    [image]
    **Operation Properties**
    [image]

Sophist-UK, 2022-10-14:
* [ ] Special wordlist entries e.g. op_power should not have contents editable. Suggest a special Type is used for these entries.
* [ ] Wordlist `Counter` variables - should I be able to create multiple content values for these?

tatarize, 2022-10-14:
> AFAIK, an SVG `Use` object is a one-to-one relationship with another object. Thus there is no need to use a parent object (like a Group node) to hold multiple references.

It's sort of not. You can reference a particular object but there are attributes that are on the Use directly and some which are applied to the sub-object being referenced. It's supposed to be done like that. However, it does seem a bit pointless and while svgelements is following the spec. In theory, I am not. When the objects are resaved we don't actually use the Use there, it will be replaced. It's actually somewhat related to a fundamental flaw in svgelements that it treats the nodes like a render rather than actually applying a render there.

tatarize, 2022-10-14:
> [Extremely minor] - I get a flash of red when I overtype a position field with ".1" because the "." by itself is invalid. Can we put the validation on e.g. a 100ms-250ms wait since the last character was typed to avoid this?

I had forgotten about this we were debating it and never got around to it. Really that solution isn't needed rather `22.` should be considered a valid number.

Sophist-UK, 2022-10-14:
`22.` is a valid number in real life and by MK. `.1` is also a valid number. `.` by itself clearly isn't, but when you select the contents of the field and start to enter `.1` you get a flash of red after you type the `.` character and before you type the `1`.

jpirnay, 2022-10-14:
> Time estimates vary between simulation and operations:

Agree, this is a PIA... Simulation has to rely on the created cutcode and is computing the distances according to that representation, while the operation panel 'just' takes the geometric length of shapes and divides it by the speed defined in that very operation. And not all operations produce that value already, ie op hatch is showing a '0' when it comes to the estimate. Definitively something to revisit for the next version - up to tat to decide but I would put it in the v0.8.2 backlog (there are already additional features implemented that require more thorough testing)

Sophist-UK, 2022-10-14:
As I said, an estimate is just an estimate, but it seems to me that the only difference should be that Simulation can calculate the travel distances and times, whereas the Ops can only calculate laser time.

This 1% difference is not enough to need fixing in 0.8, but the other estimation issue - whereby travel time is massively over estimated - does need to be fixed.

jpirnay, 2022-10-14:
[image]

- Minor enhancement: Spool window - Option to clear the history list automatically on start-up.
_done_
- Spooler history - Not sure it really makes sense to log jobs which contain only a single (or only multiple) special-operations. These rather clutter the list of real jobs.
_Not yet implemented, but you can remove single entries if you find them irrelevant (context menu over history list)_

Sophist-UK, 2022-10-14:
So, I do a Home and then a half-dozen moves to check that  the bounds of the job are within the material, and then I run a 2 hr cut job. I do this 4 times over the day. For each of these I get 8 entries in the history, of which one is relevant (the actual burn) and then I have to delete 24 of the 32 entries by individually right-clicking and deleting???? Seriously????

tatarize, 2022-10-14:
>  Wordlist `op_device` - should this be `burn_device` i.e. because AFAIK you cannot send different Ops to different devices?
>  Wordlist - should `version` be  `mk_version` in case they have `version` in their CSV file? And `date`/`time` be `burn_date`/`burn_time` for similar reasons i.e. CSV contains `date` and `time` fields.

This might be better off just using <version> directly from their CSV if they have one. Since I don't see needing both but giving a bunch of things pre-emptive prefixes incase that word should be used, is a bit meh.

Sophist-UK, 2022-10-14:
Its a minor thing - not sure it will ever come up in real life. Do as you will with the point I raised.

tatarize, 2022-10-14:
Most of these are minor. The long text things with translations weirdly sprinkled throughout are getting a little bit of a change (though not doing that now, since it might break things). Not final on the API but certainly should work like this:

https://github.com/meerk40t/meerk40t/pull/1381/files

In theory any of those long text things could get put into something like that. It's basically just:
```python
english_minute_or_two = """
What a todo to die today at a minute or two to two.
A thing distinctly hard to to do, at a minute or two to two.
The dragon will come when he hears the drum, at a minute or two to two.
"""
```

And then calling `context.asset("minute_or_two")` will return that if our language is set to `English`. And just dump some stuff in help_assets or whatever. It should make  doing a PR from the web entirely reasonable to edit some of those large formatted text blocks or whatever. It should also manage to remove the big weird problematic assets from regular translations so it looks less daunting to a wouldbe translator.

tatarize, 2022-10-14:
[image]

Snowman laser thing suffered from the same issue as the double spiral. That got fixed so this was fixed. There was a component of calculating bbox called `viewport_transform` that wasn't deleted when sync'ing the svg causing it to scale the stroke wrongly and thus broke things. The time was generally done in making the snowman raster *huge* and it's not huge.

jpirnay, 2022-10-14:
> Wordlist indexes are NOT automatically increased. Should you be recommending use of {name++} as standard?

This is by design to allow the user to manually set the position. As you indicated {name++} does automatically increase the index but it's a bit fiddly to figure out what {name} is then referring to if you use the variable multiple times.
Taking your usecase example from the help you provided (which I like btw.):
```
+---------------------------------------+          +-----------------------------------------+
| This seat is reserved for {firstname} |          | This seat is reserved for {firstname#+1}|
| Dear {firstname}, we are so happy     |          | Dear {firstname#+1}, we are so happy    |
| to have you around...                 |          | to have you around...                   |
+---------------------------------------+          +-----------------------------------------+
```
would be my way to establish mutliple cards on one sheet.

> Wordlist Counter variables - should I be able to create multiple content values for these?

No, it will just use the one number taken from the first row, if you add #+1 to it, then the value plus one will be taken. This type has some usability challenges, so any feedback on the expected behaviour is welcome.

tatarize, 2022-10-24:
All open issues moved to their own issues.

closed 2022-10-24 by tatarize