meerk40t/meerk40t #1382: [0.8b26] Issues

jpirnay, 2022-10-15:
The latest BETA release of 0.8.0 is Beta 26. This issue is for reporting issues with this specific version of MeerK40t.

Please do all further testing on this release. Please do not report any further bugs with previous beta releases. Do give any other types of feedback, questions, concerns or comments.

jpirnay, 2022-10-15:
Placeholder to carry over unresolved issues from previous reports

jpirnay, 2022-10-15:
Placeholder to carry over previous improvement recommendations from previous reports

jpirnay, 2022-10-15:
# Functionality gaps compared to 0.7
This is an enumeration of functionalities that are present in 0.7x but are not available yet in 0.8
-[ ] Rotary support

Sophist-UK, 2022-10-15:
Whilst I am personally comfortable with a full 0.8.1 release that still has minor bugs and that is still missing functionality from 0.7, I am still unconvinced with @tatarize 's optimism that this is "Probably the last everybody's generally fine with it". That was said on Beta 17 and yet here we are 9 Betas later and still ironing out bugs.

I am also concerned that the list of outstanding issues from previous Betas may be woefully understated. An extremely quick test has identified at least one issue I have reported which is NOT currently in this list. When users like me spend our valuable time reporting bugs and issues we expect them to be taken seriously, and it is really pretty disrespectful for such reports to be ignored. Unfortunately disrespect for the efforts of others appears to be a hallmark of the leader of this project.

jpirnay, 2022-10-15:
If you read carefully through the above sections you will see that they are not deemed to be complete quite a lot of the previous information needs to be carried over. Nuff said

Sophist-UK, 2022-10-15:
Well - it is difficult to judge whether this beta is sufficiently bug free to be suitable for a full release without having a complete and definitive list of the reported issues that are still unfixed.

I can only make my assessments based on prior experience and judging from the title of Beta 17 which was (from memory) entitled something like "All known bugs fixed" when this was very far from the truth, the leader of this project has a habit of ignoring the issues reported (and if memory serves me of denying that they even exist on a few occasions).

Sophist-UK, 2022-10-15:
# Functional Gaps
* [ ] Raster Wizard preview before committing changes

tatarize, 2022-10-15:
> Beta 17 and yet here we are 9 Betas later and still ironing out bugs.

The definition of a bug has gone down a lot over those 9 betas. This isn't crashing stuff it's maybe this code is a bit needing of a tweak or something. It used to be trying to use this function broke stuff and died. Now it's this `i` is not dotted and looks like a `l`.

tatarize, 2022-10-15:
> An extremely quick test has identified at least one issue I have reported which is NOT currently in this list. When users like me spend our valuable time reporting bugs and issues we expect them to [...]

Sometimes this could happen if things got checked off wrongly or weren't fully understood to begin with.

tatarize, 2022-10-15:
* We should probably go through and highlight whatever is left and create their own issues. So issue for lack of rotary support, loss of step-repeat. Issue for image passthrough. We could also do issue for other board support (we actually do support a lot of boards but those would be case by case), issue for the current lack of ruidacontrol being fully trusted for this version, and some about the weirdness for how RasterWizard currently works.
* The Issues for github isn't a really a discussion forum but for these fix a bunch of bugs runs it's often worthwhile. There is a discussion board on github for discussions.

---
### I don't think any of the tree functional gaps sophist listed are actual issues. 

* Raster Wizard is *only* previewing changes before commit. The commit happens when you send it to the laser or do simulation. Everything before that is a preview. You can change the script any amount you want before that.
* We support plenty of boards, but that support is sort of predicated on squeaky wheel. We can totally add in features that are board specific or add in entire new drivers fairly easily.
* We do not need to emulate other boards the Lhyemulator is a sort of hidden feature and largely for beta testing. Jpirnay added in a sort of emulator for grbl to get it to track movements as the data is being sent.

PS. We should probably also raise an issue for grbl being misbuilt as a driver, it should actually be a holding driver since it has a small buffer and the sending and data creation can generally be done near time to allow tracking of data or whatever.

Sophist-UK, 2022-10-15:
> The definition of a bug has gone down a lot over those 9 betas. This isn't crashing stuff it's maybe this code is a bit needing of a tweak or something. It used to be trying to use this function [...]

Yes, the bugs are less catastrophic than before, and as a consequence the current beta is more usable for production for suitable designs without risking wasting your material, and the current beta is therefore closer to production quality than previous versions. Yet there are still several issues from #1329 that are not addressed and yet not mentioned above. And there have been 7 other Betas in-between, each of which has had issues reported many of which are not yet fixed.

But as a very experienced user, and the person who has reported more issues than everyone else combined, I don't expect the time I spent testing and documenting the issues to be disrespected like this.

> Sometimes this could happen if things got checked off wrongly or weren't fully understood to begin with.

You are the leader of this project and it is **your** responsibility to manage the list and ensure that the issues are checked off correctly. If you don't understand the issue, then you only need to ask for clarification, and doing this is also **your** responsibility. If you need more help from more volunteers then you should be more respectful of there contributions and not drive them away.

But the impact of this lack haphazard approach to tracking and fixing reported issues is that there are a raft of them that have been reported as many as 9 Beta Versions ago which are STILL not fixed.

> * The Issues for github isn't a really a discussion forum but for these fix a bunch of bugs runs it's often worthwhile. There is a discussion board on github for discussions.

This is a Github Issue for discussing issues with Beta 26 which **you** are claiming is ready for full release. Any discussion about whether it is indeed ready or not for full release is therefore entirely appropriate for inclusion here, particularly since you have a documented history of prematurely stating that the software is production ready when production users have reported problems that have not been fixed.

If you want to have a separate discussion thread on readiness for production, then create one and link to it in the introduction to this issue.

> I don't think any of the tree functional gaps sophist listed are actual issues.

I believe that I have been quite explicit (on several recent occasions) in stating that missing functionality is not automatically a reason to avoid releasing 0.8.1. It is the lack of clarity on status which **is** the issue. Each area of missing functionality needs to be considered on its merits as to whether it is essential or can wait, and then they either need to be fixed or to have a clear statement that they are not included in 0.8.1. 

Of the 3 issues, only the first is of personal relevance to me (see below). 

Grbl / Ruidacontrol are A) both of minority interest, and B) do not make use of ANY of the other enhancements in MK 0.8. So although there are e.g. quite a lot of people using ruida emulation, there is no real reason why users of this functionality cannot stick with 0.7.8 for the foreseeable future. (Indeed, ALL the remote functionality could be spun off into an entirely separate smaller executable without much impact because users don't need the rest of it on the remote server - I am not saying you should do this, just using this to show that having those users stick to 0.7.8 is not an issue.)

The issues for the remote server functionality is that **the project needs to make clear what the status is so that e.g. ruida emulation users don't switch to 0.8.1 and then find it doesn't work for them and they get ruined material**. 

You are of course entitled to your personal opinion as far as your own usage of MK, but making such generalised comments publicly is **disrespectful** to the users who report such things and who do believe they are issues for them. 

Anyone who knows **anything** about UI design knows that the U in UI needs to be respected and listened to (even to the extent in commercial projects of undertaking focus groups and usability labs and A/B testing etc.), and that the developer is probably the least likely person to be able to judge what 

> * Raster Wizard is _only_ previewing changes before commit. The commit happens when you send it to the laser or do simulation. Everything before that is a preview. You can change the script any amount you want before that.

I believe that the lack of image preview in the Image Properties window (to provide equivalent **zoomed** preview of the image as the Raster Wizard in 0.7) will significantly reduce the usability for any work I have which needs image preparation (because it means I need to zoom the scene in to enlarge the image before I change the Image Properties - and I need to make a note of the current values before making changes in case I need to reset it back to where they were before. This is a significant backward step in usability IMO.

> Step repeat / Rotary

Step repeat / Rotary are also functionality that I suspect is used only by a minority of users, but I suspect that those users who do use them may well do so on a regular basis. 

The absence of step repeat is mitigated to some extent by support of `Use` SVG nodes - with this I can create an SVG with step repeat equivalency pretty easily (and do more than MK because e.g. I can rotate o [... cut ...] ability to make minor changes without having to recreate the step repeat. So I don't see this being missing as being a significant issue so long as its exclusion is documented so that anyone who uses it knows they need to stick with 0.7 when they need that functionality.

The absence of Rotary may be more of an issue if e.g. they want WordList functionality and Rotary, I don't have a rotary so this is not an issue for me, and I don't really have any feel for the number of rotary users out there.

jpirnay, 2022-10-17:
> (from simulation comments) Feature request: A playback clock showing elapsed time vs. total or remaining time would also be nice (as a future enhancement).

They are already in place I think:
Simulation: 
[image]

During burn: either in spooler:
[image]

or in status bar:
[image]

Sophist-UK, 2022-10-17:
@jpirnay Jens - Apologies, yes, these extra boxes in the simulation (c.f. 0.7.8) are exactly what I was asking for. Personally I would remove the vertical bar between the current playback distance / time and the total distance / time and replace it with the word "of" because I think that would aid users in more easily realising what the two boxes are (though as soon as you click play it becomes clear).

Sophist-UK, 2022-10-17:
Whilst checking this out, I see that the travel estimate has been fixed. :-)

Sophist-UK, 2022-10-17:
* [ ] Raster operation node label would b benefit from DPI being shown - in 0.7 the step value was shown.

Sophist-UK, 2022-10-17:
* [ ] Inkscape labels on Use elements should be used to replace the label on the element that is cloned (e.g. group or individual element) rather than duplicating the original label.
* [ ] Changing DPI on images is screwed up - multiple issues I think including: labels not changing when using Raster Wizard or DPI right click or changing on OP and not ELEMENT nodes or vice versa, right click to set Raster Wizard or DPI on multiple selected images not working etc.

Sophist-UK, 2022-10-17:
* [ ] A **VERY** minor request - is there a monospace sans-serif font that can be used in WordList Help tab rather than Courier (which is Serif)?

jpirnay, 2022-10-17:
>  Image dithering is now disabled by default and doesn't get turned on when you use the Raster Wizard or set DPI.

I think it's there:
a) property window of image:
[image]
b) Right-Click on image in scene:
[image]

In which constellation does it become disabled?

Sophist-UK, 2022-10-17:
I loaded a file with several cloned groups with images, the images had dithering disabled in properties. The labels in Image op had incorrect DPI and size. I right clicked on an Image op node and selected DPI 500, and the DPI didn't change in the label. I opened the image properties and DPI was not 500 and closed it. I selected an images in the Images op and right clicked and clicked Raster Wizard and selected one of the scripts. The Image property window opened, dithering was still disabled (whereas in the 0.7 Raster Wizard it is enabled by default). I enabled dithering and selected 500 DPI and DPI did not change in the image op label. This functionality has completely changed in 0.8 and it is still buggy.

jpirnay, 2022-10-17:
[image]
That's my test file (wasn't able to up with something more elaborate) - I can set all kind of operations on regular / cloned images (cloned images aren't real clones anyway, they just get copied). Can you share your file please?

Sophist-UK, 2022-10-17:
[image]

jpirnay, 2022-10-18:
- [ ] Invalid treatment of transparency within an image. I still can dither, but there's definitely an issue with the treatment of the embedded picture. It uses some transparency information that is not properly dealt with within the image manipulation / preprocessing code. That may have wider implications. Easiest way to see: try to invert the image. I have established a fix, that requires some further testing before it may be applied across the board.
[image]

Sophist-UK, 2022-10-18:
I didn't actually find that issue - its a new one, though I cannot actually see what the problem is from the image you posted and probably would have been happy with the results anyway. However there have been similar issues in 0.6 and 0.7, and it seems that images with alpha channel are a bit of a minefield I guess.

tatarize, 2022-10-18:
The image inversion thing is a problem. There's also a view-invert bit somehow invert wasn't correctly processing the transparency and white such that it would invert things correctly. Clearly there was a bit that should have been black but got stripped weird, so the transparency to black wasn't seen correctly as a boundary that invert was required to preserve.

tatarize, 2022-10-24:
All open issues were moved to their own issues

closed 2022-10-24 by tatarize