meerk40t/meerk40t #1388: [0.8.1] Issues

Sophist-UK, 2022-10-17:
Version 0.8 has now reached full-release status with the formal release of version 0.8.1.

This issue report is provided for users of this software to report any bugs or issues found. Please be specific with your report and if feasible attach relevant information (screenshots, the file you used etc.) to it, so that the developer team can try to recreate the issue.

If you want to provide feedback on other subjects close to your heart please use appropriate channels:

* General discussion: https://github.com/meerk40t/meerk40t/discussions
* Feature requests: Please open a [new Github suggestion issue](https://github.com/meerk40t/meerk40t/issues/new?assignees=tatarize&labels=Type%3A+Enhancement&template=suggestion.yml&title=%5BSuggestion%5D%3A+] for your suggestion.

Sophist-UK, 2022-10-17:
This comment created for the development team to list all issues reported in previous betas that have not been resolved in 0.8.1.

tatarize, 2022-10-17:
0.8.1 issues should probably have any issues that were not resolved during the regular process either discussed in the discussion about the version.

https://github.com/meerk40t/meerk40t/discussions/1387

Or raised into their own issues. Some of the missing functionality needs their own issues just to discuss the various aspects of what is needed and involved. For example rotary is a pretty big topic, and might require (to do it right) some pretty noteworthy changes in scene as well as resolving various aspects of how splits should work if they don't take over the y-axis.

Sophist-UK, 2022-10-17:
Well, you already know my view that the image bugs are sufficient to consider this not ready for production. But in the end it is your decision.

IMO there are way more than a dozen issues remaining unresolved - and the fact that you are guessing is proof (if any were needed) that you haven't had the discipline to maintain a definitive list of open issues nor been willing to put the effort into trawling through the previous bug reports and create one. I expect to find at least another dozen or two when I start using 0.8.1. and I am damned if I am going to spend 10x the time to create a separate new issue for each one I find. (And if I did it would drive you crazy having a flood of (say) 20 individual issue emails arriving.)

So I will be reporting bugs I find in a combined issue, whether you like it or not - and on that basis if I find a bug that I previously reported that dropped ended down "the back of the sofa" then I may well include it again. I have already gone way further than any other user in reporting bugs - I have quite literally reported 10x more bugs in 0.8 than all other users combined, and after all the effort I have already put in if you now expect me to repeat myself and report the same bug a second time in an individual Github issue you are very much mistaken. You are, of course, at liberty to spend your own time splitting them out into separate issues if that is going to help you manage them better.

tatarize, 2022-10-17:
That's fine. It's mostly the missing feature things which aren't really bugs per se that need their own issues.

Sophist-UK, 2022-10-17:
The image stuff I reported yesterday are bugs not feature things. And without having kept a proper track or done a careful trawl through the bug reports and checked that they have been fixed, your statement that it is mostly "feature things" is **just a guess**.

MontyTheGrey, 2022-10-17:
> The image stuff I reported yesterday are bugs not feature things. And without having kept a proper track or done a careful trawl through the bug reports and checked that they have been fixed, your statement that it is mostly "feature things" is **just a guess**.

Please shut the f**k up, you're actually making it impossible for me to work out the actual bugs with this software and decide if it's worth using. "If you have any respect for developers and the development process", you'll go back and delete all your tantrums over the organisation/administration side of things and the features you'd like which aren't actually bugs because it's spamming an important tool developers rely on.... You're also being rude and abusive towards volunteers who owe you nothing, be grateful and appreciative instead of demanding they include every feature you want over making the software how they want.

Sophist-UK, 2022-10-18:
@MontyTheGrey What you don't understand is that I am actually following the culture of this project and the apparently expected behaviour as expressed by the owner who has made it completely clear to me in other places that this is how you should treat others in this project because this is how he has treated me. He has personally been rude and abusive and disrespectful to me, and he has allowed others to be abusive too and indeed then given them preferential treatment rather than acting to prevent it. He continues to disrespect me, and I am not going to put up with it in silence. If you don't like this culture, then take it up with him and try to persuade him that it makes this project not a nice place to be - because I made it clear to him a long time that he was making this a nasty place, and I was told that this is fine with him, so I am simply giving as good as I get.

Sophist-UK, 2022-10-18:
Since @tatarize has seen fit to comments, I think I need to clarify the situation. Just so we are clear, another developer was being abusive, I asked @tatarize to make the Discord channels a safe place and he responded not by preventing the abuser from being there but instead by A) creating a secret channel that everyone but me was invited to, and B) removing my admin privileges - he favoured the abuser and punished the abused. He said that this was because the abuser had (in the past) made useful contributions - which in itself is morally bankrupt as ethically abuse cannot not simply be excused by reference to other plus points, but even if you accept that premise he didn't apply it evenly because he took the abuser's contributions into account but not mine. This was simply further abusive behaviour by the leader of the project who should have been making it a safe space rather than making it even more unsafe. How is that honourable or justice or not simply completely screwed up? I left the development community because it was NOT a safe place, but tried still to be a helpful user.

Then a couple of weeks ago, Tat explained to me that, not only were his above actions entirely appropriate (which they clearly aren't) but proceeded to tell me just how much my code contributions were a waste of time (including cherry picking examples and ignoring others). Not only was he a glass half-empty person, but rather than an apology I got another ethically bankrupt self-righteous justification for why he was perfectly right and I was in the wrong. 

I will be happy to share the conversation log as evidence with anyone who wants to read it.

It has therefore been made abundantly clear on several occasions that respecting the efforts of others is NOT a cultural value for Tat, and I am simply now fitting into that culture by refusing to let Tatarize continue to disrespect the efforts I have made.

This is easy to fix. Tat can apologise and make things right. But he won't because he doesn't have the ethics nor humility to admit when he has made a major mistake.

Now we are definitely off topic here - because this has got nothing to do with bugs reported - but since Tat has seen fit to gaslight me here, I am going to respond to that.

tatarize, 2022-10-18:
Sorry about that. He's not a great welcoming squad. I've blocked him for 3 days or whatever.

tatarize, 2022-10-18:
As for the bugs, I've posted a couple issues on missing functionality. There's also a pretty good legacy branch that's been rather heavily tested and found to be rather stable. Also, K40 Whisperer is pretty great software too, and you don't really need to pick or choose between the two.

I'm planning on making some video guides. Explain things since bugs is much less of an issue than potentially a steeper learning curve.

MontyTheGrey, 2022-10-19:
@tatarize Thanks for the response and sorry for my language but I've never seen anyone publicly treat devs so badly and I mean that. He might have valid points sometimes but don't let that detriment your teams creative process and knowledge.

I'm currently using K40 whisperer on an old computer with Parsec but it's a bit janky. I'd like to run the old pc headless and control the K40 directly from my good pc on the other side of the house. The network and webcam functionality of MeerK40T seems perfect compared to the data loss I'd expect with VirtualHere. I'll make a post about it on Discord or the Discussion board once I work out what I'm doing

tatarize, 2022-10-20:
Eh, generally he strongly advocated for comeuppance for people talking trash about the project and annoying would-be new folks.  He has his faults but he more than pulls his weight.

--

Yeah, in theory the only issue with headless operation from across the house is that sometimes people are across the house and that's not exactly close enough to put out the literal fires.

Sophist-UK, 2022-10-23:
Yes - I did advocate for comeuppance for people giving personal ABUSE to other users. And I am self-aware enough to know that I wouldn't have stood for my own behaviour if I was in Tat's place, but then again I have sufficient emotional and ethical intelligence that I would not have created the situation in the first place, and if even I had I would have then have the self-awareness and humility to apologise and try to make it right, so it would never have escalated to this.  And as I said before, my disrespectful attitude is EXACTLY the attitude that Tat displayed to me only a week or two ago - so yet again dual standards operating.

Sophist-UK, 2022-10-23:
* [ ] Raster time estimate is way too low. Currently into a burn estimated @ 50 mins of which 49 mins is a single large raster - so far it has been running for 39.5 mins and has literally just come to the halfway point. It's a 270x175mm image rastering at 200mm/s DPI 500.

Sophist-UK, 2022-10-23:
* [ ] Crash on Sppoler/History/Delete History older than one week.
```
MeerK40t crash log. Version: 0.8.1000 on Windows: Python 3.7.9: AMD64 - wxPython: 4.1.1 msw (phoenix) wxWidgets 3.1.5
Traceback (most recent call last):
  File "meerk40t\gui\spoolerpanel.py", line 303, in check
  File "meerk40t\gui\spoolerpanel.py", line 279, in clear_history
TypeError: '<' not supported between instances of 'NoneType' and 'float'
```

tiger12506, 2022-10-23:
> Yes - I did advocate for comeuppance for people giving personal ABUSE to other users. And I am self-aware enough to know that I wouldn't have stood for my own behaviour if I was in Tat's place, but [...]

Tat tried to compromise with you over and over again. Wasted hours and days of his time doing so. He literally isolated the person abusing you so that you were 'safe' from his abuse. Nobody is going to apologize for doing this. It is extremely rude and disrespectful of you to have thrown that back in to his face. Like standing next to a bucket of water while a fire is raging, insisting that someone else splash out the fire, and then telling them they did it wrong after the fact. "You just can't help some people no matter how hard you try"

All you have done since then is be rude, ungrateful, disrespectful towards the dev who made it even possible for you to have this social justice warrior soapbox. Even now, he is saying kind things of you. Tat is literally a saint compared to you. If I were in charge, the first time you whined to me for several hours about how someone hurt your feelings I would have kicked you to the curb and handed you your pacifier. Personally, I don't find your level of testing and support worth it. Seems you're quite capable of dishing it out, but you can't take it yourself. The dual standard here is that you're allowed to be rude and abusive and still be seen in this public location. That your testing and feedback is more important than anyone else's. That your features should be implemented whether or not they mess up other people's usage or workflow just because you run some production items.
Go home, child, and learn to not be so entitled. Part of maturity is learning to accept this fact.

I can't possibly imagine what you want out of this whole situation. Do you want paiin banned regardless of future hardware support for meerk40t? Do you want Tat to apologize for giving you exactly what you asked for? Do you want everyone to just let you run everything? What possesses you to think that the future of MeerK40t owes you anything at this point? Do you want your name bigger in the About box? Do you just want to argue constantly for the rest of your life in outrage? What does it benefit you or anyone else?
Most importantly of all, do you recognize that this platform is about MeerK40t, and not you?

Many times it has been suggested that you could create your own fork of meerk40t and oversee it how you wish. If you ever do see fit to manage your own project, I hope your superior emotional and ethical intelligence serves you well.

Sophist-UK, 2022-10-23:
> he strongly advocated for comeuppance for people talking trash about the project and annoying would-be new folks. He has his faults but he more than pulls his weight.

Just to clarify this again. I advocated for comeuppance for a person who was making highly personal, nasty false comments which had no cause. That is abuse, and it makes the environment unsafe for everyone, not just the  individual being abused, and IMO should be dealt with promptly and robustly.

If someone had an opinion about the project that they were attempting to justify with valid facts, I would A) **listen and evaluate** whether there was some truth to what was being said, and B) try to persuade them that this was not the case. That is NOT abuse, it is a difference of opinion, however vocal. But if in the end having tried to understand their position and consider whether they actually had a point we still had serious and irreconcilable differences, I would ask them to leave.

(The third case is someone who has just turned up for a rant without any history, facts or justification, and who is not open to rational debate or persuasion. There is no point in listening, and they should be kicked.)

So these are **not**, as Tat is implying, equivalent in the slightest.

> He literally isolated the person abusing you so that you were 'safe' from his abuse.

**Tat literally did the absolute opposite to this - he isolated me and put everyone else in a group he kept secret from me. That is NOT the "comeuppance" he talked about above. He literally punished me and favoured the abuser.** And **this** is the crux of why I am so vocal. And he still has zero understanding of how inappropriate that was, and has never said a single word of apology about it, instead continuing to justify that as the correct behaviour and gaslighting me as the person in the wrong. He has never expressed any empathy for my having got upset about this, even if he doesn't agree with my view that this was highly unethical. 

> Even now, he is saying kind things of you. 

Yes - but he says one thing in public but something different to me in private. That is literally gaslighting.

You should see the transcript of the conversation I had with him on mIRC a couple of weeks ago. A more unapologetic, self-justifying, disrespectful and ungrateful set of comments you couldn't imagine. Not only did he still consider favouring the abuser and isolating the abused was the correct approach, but he literally told me that my code contributions were a waste of his time (essentially by being selective with the facts) and that whilst he valued my efforts in beta testing and (formerly) in promoting Meerk40t, the implications was that they didn't count because they were not activities that he would have done and which therefore freed up his time. In other words, he measures contributions only from the selfish perspective of his own time.

If anyone can dish it out but not take it, that would be Tat. He bluntly disrespects me, but then complains when I give as good as I get. Indeed in the past you (@tiger12506 ) also made it clear to me that your personal view was that people should stand up for themselves, and not expect Tat (as project leader) to deal with abusers, yet here you are complaining because I am standing up for myself.

Your comments about about how you see me as being abusive are as equally dismissive of the reasons behind my comments as Tat's. You clearly don't understand the reasons for me speaking out but instead are trivialising them as me whining and being abusive myself. Well...

* I don't believe that this project is all about me - it is Tat who measures people's worth by the amount of time they use up or save him personally.
* I have been clear on many occasions that this is great software, and that in the space of < 2 years from 0.6 to 0.,8 it has improved beyond belief and gone from strength to strength. I have also been clear on many occasions that I think that Tat is a talented programmer and he has created code that I could not have achieved - but being a talented programmer is not the measure of a person or a project leader. The majority of the issues I report get fixed. So I have every reason to contribute to this project and no reason to fork it. And, just like for 0.7, I have contributed **a lot** to the quality of 0.8 by beta testing.
* My contributions to the project are **not** my justification for speaking out, and that would be hypocritical of me if I did. My reasons for speaking out are to get justice for wrongs that need righting. It is Tat and you who keep going on about contributions as if this is the only measure for everything and right/wrong doesn't matter.
* If I wanted to fork the project and do stuff myself, I could have done that from the start - but actually I didn't because I knew that the best result would be from being part of a team / community and normally I **am** a team player. 

So where these comments about this being all about me and me wanting to fork the project have come from I have no idea, but the gist of them is they demonstrate that - yet again - a complete inability to listen or hear what the issue actually is.

Can you or Tat not see the difference between A) the past situation where one developer was being personally abusive to another developer, and B) these comments from someone who feels things are unjust, who has tried on several occasions to get justice in private, who has walked away from developing but continu [... cut ...]
But most of all, good leaders treat people as people (not just code creators), with opinions and emotions that are worthy of respect regardless of whether they agree or not - and good leaders listen and respond to defuse and make things right when they go off the rails rather than simply justify their own position.

So, at this point, having pointed out **yet again** that this is an issue of ethics, of the right for someone to feel aggrieved and  seek justice, and you can either choose not to listen and continue to characterise this as griping, or you can start to listen and try to defuse the situation and try to make things right.

tiger12506, 2022-10-23:
Quite simply, because he's tired of you, and so am I, and so is paiin, and so is jpirnay. This is not your soapbox. It really doesn't matter much if ethics are or are not being followed. This is about MeerK40t, and it's your desire to feel validated and right that is pushing conflict. Whether or not your pursuit of justice is righteous or not, it doesn't really matter anymore.

I don't really care what you've been a manager of, tbh. All it shows me is that you are capable of bullying and firing people that don't fit in your work ethics. To be honest, from the start I was all for your case, but by now I want to quit as your advocate and move on with my life.

You keep framing your argument as ethics, making a good team, for the better of meerk40t. But your very argument *isn't* good for meerk40t. You've already had one new user listen to you and make his own conclusions about this situation based on *your* behavior.

Tat has tried to be polite, I think he has all the best qualities of what it takes to manage people and get a team working well together.

Whereas *you* have managed to cause everyone on the team to dislike you. I don't want to burst your bubble of life-experience or anything, but a lot of people end up hating their managers. Perhaps you were one of those. Thank goodness you retired from that job.

"choose not to listena nd continue to characterise this as griping, or you can start to listen and try to defuse the situation and try to make things right"

And how do you suggest we make things right? Again. What do you want from this? Why would you continue? What *is your goal here*? I don't know how to make it any plainer. You need to think about what is good for meerk40t if you wish to continue helping it, and soapboxing on ethics and bringing up evidence from a cold case just isn't it.

Nobody is interested in apologizing to you, and we don't expect any apology from you either. So, either we all act like civilized grownups and move on, continuing to provide usefulness to meerk40t. Or you can leave, and the rest of us will breathe a sigh of relief, new users included.

Thank you for what you've done for MeerK40t so far, and the contributions you have made. Your name remains in the About box, rightly so.

Sophist-UK, 2022-10-24:
No - this is NOT about my wish to be validated or right. It is:

1. My belief that people should be accountable for their actions, especially when they mistreat people i.e. justice.
2. My belief that if it happened to me, it can happen to someone else in the future. The world is full of repeat examples because people made excuses. #metoo
3. My belief that all contributions by volunteers should be valued even if a minority of them take more time to review than it would to have coded them in the first place. I am not looking for effusive thanks for each and every little thing, but I do not expect my efforts to be called a waste of time (even if a minority of them needed as much review effort as the coding might have taken), and I expect such code quality issues to be called out during review as a learning opportunity (so next time the code will be better) and not accepted at the time and then trashed at a later date. 
4. My expectation that if there is going to be some form of much later critique - and there wasn't any need for the one Tat lobbed in recently - then it needs to be fair and factual and NOT an exercise in self-justification and disrespect by the leader.
5. My belief that it is by definition OK to follow the example set by the leaders. If Tat wants to treat people with disrespect, he shouldn't complain when they do the same to him.

As for my work ethics, I genuinely can't remember an occasion when I ever fired someone (though I do recall a supplier project manager who I asked to be removed once). The only occasion I can recall having to "fire" someone was when I was told by senior management to make someone redundant, and it was a painful experience because I do have empathy. I was careful about hiring, and that may be the reason I didn't need to fire anyone, but a belief in the positivity of people generally paid off (including unfiring people who I inherited during their notice periods who I found were valuable). As far as I know, no one who worked for me ever hated me - though I was certainly a bit pompous at times. (Lol - unbelievable I know!) So, yet again, you have made up stories about me based on zero real knowledge. As and when you have facts to use to disparage someone, please feel free to use them - until then please stop making things up to justify your own position and ignore someone elses.

Your statement about this being a cold case is also not true. Yes, there is a cold case in there, but there is a reason it has come up again - further unjustified disrespect. Tat couldn't let things lie - he had to rake over the embers and explain just why he was so righteous, and whilst doing that be dismissive about my code contributions, calling them a waste of time. To use an analogy, he ripped open a healing wound and poured salt into it.

As for making things right, actually it wouldn't be that difficult. A) Stop denying that mistakes were made. B) Stop believing that people shouldn't get upset if/when they feel they have been mistreated. C) Admit that creating a secret channel especially to keep the abuser onboard whilst isolating the abused was a mistake and apologise. D) Stop seeing the glass as half empty, and apologise for being so negative about the time and effort put in by a volunteer. In other words, be human and accept that you can make mistakes, and be humanist and treat people as people and value their contributions. What's so hard about that?

tiger12506, 2022-10-24:
No mistakes were made. You're just an asshole. Leave us alone. I'm done. I'm out of the conversation.

P.S.
I hope that you never compare the name calling paiin did to you face to face to someone who actually deserves to invoke #metoo. I know a few people who would do a lot worse to you than paiin ever could for even making the comparison. Get over yourself.

tatarize, 2022-10-24:
> I advocated for comeuppance for a person who was making highly personal, nasty false comments which had no cause. That is abuse, ,,,

Part of the issue here and why I could never successfully deescalate the situation. Upon retelling the wounds are deeper the attacks or more scathing and each time reconsidered get more deeply personal. While Paiin was at times rude, he wasn't making personal attacks. 

> So these are **not**, as Tat is implying, equivalent in the slightest.

The events as built up in your head compared to the events described actually are generally the same thing. The disanalogy is only found upon each retelling with each reconsideration creeping ever closer to Hitler.

> Tat literally did the absolute opposite to this - he isolated me and put everyone else in a group he kept secret from me. 

That Paiin was told to stayed in, as to not interact with you. That new users could not see, and would not be informed existed. The alternative was to start a different server altogether, or just communicate over email or whatever.  Would that also be excluding you if I didn't BCC you on any correspondence between paiin and I? 

> He literally punished me and favoured the abuser.

Putting paiin in a hole and telling him not to interact with you is not punishing you.

>And he still has zero understanding of how inappropriate that was,

This is accurate, I still do not understand how your interpretation makes any sense.

> He has never expressed any empathy for my having got upset about this, even if he doesn't agree with my view that this was highly unethical. 

I can't control your emotions but I absolutely did exactly that. I do not want valued people to get stuck in a loop where they feel everything was an attack on them. And said as much in our irc conversation. ```[03:45:48] <Tat> But, holding on to anger is like drinking poison and expecting the other person to die. And I certainly do not want that for you.``` -- That's literally expressing empathy at your reaction (which I cannot and do not control).

> You should see the transcript of the conversation I had with him on mIRC a couple of weeks ago. A more unapologetic, self-justifying, disrespectful and ungrateful set of comments you couldn't imagine. 

That's pleasantly ironic. Tiger could actually imagine such a thing, because I literally *had* told those exact things to Tiger. ```Also, in the spirit of honesty. Allow me to share my personal messages about the whole affair a day or so after you left. Keep in mind, I'm only sharing this with you to help you understand my position.``` -- Those personal messages were *with* Tiger. Not only did I post that conversation to the moderators channel, I was quoting my side of a conversation I already had with Tiger to give you a better understanding of what went wrong, there.

>and good leaders listen and respond to defuse and make things right when they go off the rails rather than simply justify their own position.

That's what keeping paiin away from you was attempting to do; It didn't help. You said you didn't want paiin around and when he so much as mentioned you even in passing you took it as a personal attack.

P.S. This topic should probably be closed 0.8.2b1 is out and drama doesn't look great or whatever.

tatarize, 2022-10-24:
All issues moved to #1422, #1421, #1420, #1419

closed 2022-10-24 by tatarize