meerk40t/meerk40t #1392: 0.8.x: Image Passthrough

tatarize, 2022-10-18:
The functionality used to generally exist, but due to the shifting of RasterWizard to be more integral to images that guarantee was lost.

Many programs allow for image pre-processing including meerK40t but with the expansion of services like Imag-r (http://imag-r.com) we need overt support for images being set in passthrough mode. The preprocessing of data is largely a heavily lossy operation usually ending with dithering the data down to 2-bit images. After this is done it is no longer possible to process the image further. However the problem arises if you try. Trying to process an already processed image can absolutely cause serious problems for the image and image quality.

Passthrough mode images should allow you to set a size (which is directly countered with DPI) or set a DPI which overtly requires the image be an exact size. Rotations that are not exactly 1/4th turn (or multiples thereof) or flips could not be permitted. And such a mode should provide a guarentee that we won't touch one pixel of the original image but send it entirely nakedly to our background driver in a 1:1 mode.

Passthrough should set by default if an image is loaded in 2-bit.

jpirnay, 2024-10-04:
Still relevant

MMMystery, 2026-10-28:
A small acceptance-test proposal for the pixel-preservation guarantee, separate from choosing the original/modified image in the current UI:

* Use an asymmetric black/white fixture, such as 9 x 7 pixels with one-pixel lines and an off-centre marker. Test both a 1-bit PNG and an 8-bit greyscale PNG whose only values are 0 and 255. A two-colour image is not necessarily stored with a 1-bit sample depth; automatic detection should inspect decoded values, not just PNG bit depth.
* With no transform, compare the decoded output pixels with the decoded input. Comparing PNG file hashes alone can fail even when the pixels are identical, because compression/metadata can change. Assert dimensions, polarity and every pixel; a second dither/contrast/invert step should not silently run in a strict passthrough path.
* For 90-degree rotations and flips, compare against the corresponding exact pixel permutation, including the asymmetric marker. For arbitrary angles or resizing, either reject strict passthrough or make the user explicitly leave that mode; resampling cannot preserve the original grid one-for-one.
* Keep physical placement separate from pixel preservation. A 1,000-pixel width at 254 DPI is 100 mm, and PNG pHYs can express that as 10,000 pixels/metre with unit 1. A physical-size/DPI edit should not silently resample a pre-dithered bitmap. Include a no-pHYs fixture too, so the import-size assumption is visible.

The metadata distinction is defined in the [W3C PNG pHYs specification](https://www.w3.org/TR/png-3/#11pHYs). I maintain RasterGo; our [free image-size/import checklist](https://rastergo.com/guides/laser-engraving-image-size/) illustrates the same file-side distinction. It is not a MeerK40t passthrough implementation or a machine-setting recommendation, and switching prep tools is not required. I have not run this test proposal against MeerK40t or a physical laser. AI-assisted wording, checked against the current image documentation and PNG specification.