meerk40t/meerk40t #1448: [0.8.1] Spooler - Queue estimate is twice actual estimate

Sophist-UK, 2022-10-28:
This screen shot shows the exact same job queued twice. As you can see the second job has twice the estimated time of the first job...
[image]

jpirnay, 2022-10-30:
The estimate is updated during execution and takes the amount of things done and the remaining tasks into account

Sophist-UK, 2022-10-30:
No - they should both say 11:07 as an estimated time - the Estimate for a burn job doesn't change until the item run time reaches the item estimate, and then the item estimate increases in step with the item run time and the total estimate increases with that.

This is a bug.

jpirnay, 2022-10-31:
Fixed

Sophist-UK, 2022-10-31:
So - it was a bug then?

jpirnay, 2022-10-31:
It certainly was, thanks for reporting

closed 2022-10-31 by Sophist-UK