meerk40t/meerk40t #1449: [0.8.1] I can delete the currently executing job in the spooler and it keeps running nevertheless.

Sophist-UK, 2022-10-28:
I think there should probably be separate context-menu items for:
* cancel currently burning job: which lets the instructions already sent to the laser complete, then puts the laser into a known state (and possibly pops up an interrupt box if there is already another job queued); and
* remove queued jobs

Also remove job on the current job doesn't give an entry in the job history despite that job having partially run.

tatarize, 2022-10-29:
If you hit *stop* mostly it will do this and kill that stuff. This is an estop. Cancels the currently burnings, breaks the running, and kills any followup jobs.

You probably should not be allowed to delete the current job. You should be required to cancel it. In which case it should probably stop at the first opportunity and move on to the next job in the queue.

If removed it didn't complete and thus isn't in the history but it should probably note the state in the history. Job run, made it up to this position. Then it was killed. There's also some chance you could maybe store enough information about that job that you could resume it. Though that might be a bit non-obvious but if you knew exactly where it stopped you might be able to resume as such.

Sophist-UK, 2022-10-29:
Exactly!

jpirnay, 2022-10-31:
That should be handled better now

closed 2023-03-13 by tatarize