meerk40t/meerk40t #1456: [0.8.1] Should run time keep increasing during interrupts?

Sophist-UK, 2022-10-30:
And if it should, then should the estimated time increase in step for the duration of the interrupt?

jpirnay, 2022-10-30:
Good point

jpirnay, 2022-12-13:
It's more complicated than anticipated, so no immediate resolution I am afraid, right now the runtime is calculated as the difference of current time vs starttime, which is independent of any window being open / watchdogs to look for interrupts etc.
A resolution would require a time keeping service.

Sophist-UK, 2022-12-13:
1. What I would do to avoid a "time service" is to hold an adjusted start time - which is initially set to the actual start time, but which is adjusted when an `interrupt` special op is completed by adding the time take for the interrupt op. You could then use this to calculate the elapsed time.

2. If you are unable to adjust the elapsed time to exclude interrupt time, the alternative is to add the interrupt time onto the estimated time, and then use the actual start time to measure the elapsed time.

tatarize, 2022-12-27:
Time the interrupts. Subtract that time from duration. *maybe*. You could want to know how long the job took including time you spent dealing with material etc.

closed 2024-10-04 by jpirnay